littlefs-project/littlefs #276: lfs_file_write takes a long time after the flash memory is full and I start erasing files

xxzp, 2019-08-25:
I am using LittleFS for a custom data logger that I have designed. It's been working very well with no issues so far until I tested the scenario of the memory becoming full. I am using a 64MB SPI NOR flash with a 4KB sector erase size, and the SPI clock running at 20MHz, part number MT25QL512ABB. 

I use all statically allocated buffers, with read, program and file buffers being 4KB in size. I have experimented with changing the lookahead buffer size from 4KB up to 16KB but this didn't make a difference to my problem. 

read_size = 1;
prog_size = 256;
block_size = 4096;
block_count = 16384;
cache_size = 4096;
lookahead_size = 16384; (tried 4096 up to 16384 with no difference)

I want to make the flash function so that when it becomes full, the oldest data is erased to make space for the new data. My current implementation creates a directory for a file, and then stores the data in "block files" which make up the overall file as below. The 0, 1, 2, 3 files under the directory are 1MB in size each. A new block is started whenever the size of the block reaches 1MB. The actual data packet is 17 bytes in size, and I write and sync every time I log a new data packet.

/FILE_NAME/0
/FILE_NAME/1
/FILE_NAME/2
...
...
...
/FILE_NAME/63 

When the flash is full, I delete the block file with the lowest number in the directory using lfs_remove, and then call lfs_file_sync. Until now, the time to write and sync a single data entry was < 100mS always. However after the flash becomes full, and I start doing what I described above, a lookahead buffer size of 4KB causes every 80-85th entry to take between 540 and 580mS to execute lfs_file_write. With a lookahead buffer size of 16KB, every 250-255th entry takes 540-580mS on lfs_file_write. 

I have timed the sector erase time and this is always between 20 and 30mS. 

What could be the cause of this long delay in lfs_write_file and is there anything I can try to decrease this time?

Thanks

joicetm, 2019-11-07:
i am also facing same issue, its slows down as the memory gets full.

keck-in-space, 2020-02-14:
Do you see this only after you erase? If the file system isn't full and you erase one of the files, do you still see this behavior? I'm just curious if it is really because the flash is full, or if its because you execute an erase.