littlefs-project/littlefs #285: How did you do the diagrams?

drudru, 2019-09-06:
Hi, I like the ascii diagrams in SPEC.md and DESIGN.md. How were those made?

geky, 2019-09-09:
Hi @drudru, thanks for the interest!

The ascii diagrams were created by hand with Vim's block edit mode (ctrl-v). It lets you edit a text file more as a grid.

There are some minor aesthetic things I've picked up on. You have lines `|-` and arrows `v<^>` and can make nice looking rounded corners with apostrophes and periods `'-.`. After that it's just experimenting to make it look nice.

I like ascii art because there is no concern about file hosting, and the images will always live with the text. Though they don't work for everything. I considered using ascii art for graphs, but the low-resolution made it difficult to see what was going on. Instead I generate graphs with [matplotlib](https://matplotlib.org/) on a separate branch: https://github.com/ARMmbed/littlefs/tree/gh-images.

drudru, 2019-09-10:
Awesome :+1:

closed 2019-09-10 by drudru