littlefs-project/littlefs #297: Python wrapper for littlefs: Permisson to use littlefs in name

jrast, 2019-09-29:
I'm currently working on a thin python wrapper for littlefs. The goal is to provide a simple, platform independent library which does not require any compilation steps on the system of the enduser. Therefore playing around with littlefs, testing if it would be a good fit or preparing/read/change flash memories on a computer would be a lot easier.

The current state of the implementation can be found here:
https://github.com/jrast/littlefs-python

I'm planing to release the library under the name "littlefs-python" and wanted to ask for permission to use this name before a initial release. The library will use the same BSD-3 Clause licence as littlefs itself. Is this ok or should I change the name?

geky, 2019-09-30:
That looks like a great project, thanks for sharing! Feel free to use the name, it would be more confusing without it : )

jrast, 2019-09-30:
Thanks! Hopefully the package will be of use for some people!

closed 2019-09-30 by jrast