littlefs-project/littlefs #300: LittleFS Explorer for Windows

bluscape, 2019-10-01:
Sorry for posting here. It is not an issue.
I've spent quite a bit of time developing a a file explorer for LittleFS under Windows.
If you think it is worthy, could you please add it to your list of related projects.

https://github.com/bluscape/LittleFS-Explorer-for-Windows

Thanks

fizzyade, 2019-10-08:
Useful, but no source. Is it going to be closed source or will you open it up at some point? (i know you have no obligation as littlefs is BSD licensed)

Didn't realise that delphi was still a thing either!  I use Qt commercially, but am currently developing a new project myself (github.com/fizzyade/sfpdoctor) and initially I was going to use Qt, but decided to use electron to develop the software, frameworks like electron and native-script hold an interest for me as they're both node based and useful for mobile development.

bluscape, 2019-10-09:
I will make the source available. I do mention it in my blog.

fizzyade, 2019-10-09:
> I will make the source available. I do mention it in my blog.

Awesome (btw I wasn't suggesting anything, only asking if you were going to open it up as its probably useful for quite a few people!  Nice looking app, is it going to be cross platform? (i have no idea about delphi in this respect).  

We manufacture radar equipment and I did a test build just a few weeks after littlefs became available, with little effort I had it running off it.  The downside was that without a method of manipulating the disk from windows I could foresee all sorts of calls from customers!

Its definitely on my roadmap though.

bluscape, 2019-10-10:
No problem. My intention is to make the source available. Most probably with the first stable release.

It should be possible to run it on Linux but I have not had a look at it. Should I bother if there is already a FUSE driver for Linux?

fizzyade, 2019-10-10:
I would say yes, its far simpler to use an app than it is install the fuse driver, as long as the app supports drag and drop to/from the device and general manipulation i'd say its useful.

I guess it might also be useful to provide command line options to perform various tasks without the GUI, I.e

fuse-app list devices
fuse-app copy <source> <target>to copy files to/from the device

And so forth, would allow easy configuration loading for a device.

bluscape, 2019-10-30:
Hi. I have a solution. Will post it tonight. Could you please post LFSE related questions on the LFSE page.
So please repost on LFSE so that other people can gain from this. Thanks

bluscape, 2019-10-30:
Have a look at "Mounting Binary Files" on my page: https://bluscape.blog/2019/10/01/littlefs-explorer-lfse-for-windows/comment-page-1/#comment-12

vincentfenet, 2024-05-31:
Could you please opensource your project ?