littlefs-project/littlefs #301: Root relocation when only reading drive?

e107steved, 2019-10-02:
Using lfs 2.1.2
I've got a flash drive which is only written to from a manual trigger.
I've set up a test to loop reading a small list of files flat out.
Every 3766 reads, I get a message of the following form:
```
lfs_debug:1656: Relocating 549 54A -> 54B 54A
lfs_debug:3879: Relocating root 54B 54A
```
(The numbers increment each time).
3766 doesn't appear to relate to drive size:
4096-byte blocks
1920 blocks total
16-byte read
256-byte write/cache

Is this expected behaviour?

geky, 2019-10-02:
No this shouldn't happen. Is this a tool or porting layer or is this coming from direct calls to `lfs_mount/lfs_dir_open/lfs_file_read`?

If it's through a separate tool it's possible they're updating the file's atime, which could cause this to happen.

e107steved, 2019-10-02:
It's through a relatively simple porting layer, which maps standard C-language file access calls . 
I've not counted the start and end bits, but the 3766 accesses before relocating root appears to be a fixed number - any stop and start is ignored. The system has an RTOS, and adding an extra load to the processor doesn't alter this count. It's around 35-40 seconds between relocations.
The two files I'm alternating between are both fairly small - 2677 and 4762 bytes.
I am storing a single date/time attribute value with each file (time of last write - set just before call to lfs_file_write()), and that does not change. There's no support in my code for use of atime.
Could an attempt to update the file's atime trigger this even if the time is ignored?

e107steved, 2019-10-02:
I've checked the porting layer, and also had a quick look at a representative newlib, and no obvious references to them setting atime

e107steved, 2019-10-07:
Done a couple more tests.
1. Looping round three files instead of two gives the same 3766 transactions between relocations (each transaction comprising open file, sequential read of whole file, close file). So count clearly related to total file system activity, rather than activity on a single file.
2. If I stop the test, reset the processor and restart the test, the sum of the transaction counts between relocations is still 3766 (maybe with 1 or 2 difference) - so its related to information stored in NVM rather than RAM

e107steved, 2019-10-07:
Two problems, I think.
The littleFs one is that if a file has attributes, the LFS_F_DIRTY flag is set regardless when reading the attributes during open (line 2412). I would expect that flag to be set only when something changes.
My problem is that the read-only flag is getting lost somewhere between higher levels and littleFs, hence exposing what I found.

e107steved, 2019-10-07:
Turned out that there was confusion on the file access flags in the higher levels - resulting in calling lfs_file_opencfg() with a flags value of zero.
So, two possible actions within LFS:
1. On file open, check that one of LFS_O_RDONLY, LFS_O_WRONLY, LFS_O_RDWR is set (I think an assert would be OK here)
2. With attributes, only set the LFS_F_DIRTY flag if the attributes are actually changed.

Please close once you've taken any action you deem appropriate.