littlefs-project/littlefs #304: lfs_file_open - doesn't work with static allocation

Leon11t, 2019-10-03:
Hi. I try to integrate littlefs with static memory allocation. I faced with a strange issue. I can't open file by "lfs_file_open" function. But all work with dynamic allocation. 

`
#define MAN_DATA_STORE_READ_SIZE               (256U)
#define MAN_DATA_STORE_WRITE_SIZE              (256U)
#define MAN_DATA_STORE_BLOCK_SIZE              (4096U)
#define MAN_DATA_STORE_BLOCK_COUNT         (8U)
#define MAN_DATA_STORE_CACHE_SIZE              (256U)
#define MAN_DATA_STORE_LOOKAHEAD_SIZE    (256U)
#define MAN_DATA_STORE_BLOCK_CYCLES          (1000U)

static uint8_t lfs_readBuff[MAN_DATA_STORE_READ_SIZE];
static uint8_t lfs_writeBuff[MAN_DATA_STORE_WRITE_SIZE];
static uint8_t lfs_lookaheadBuff[MAN_DATA_STORE_LOOKAHEAD_SIZE];
static uint8_t lfs_file_buf[MAN_DATA_STORE_READ_SIZE];

const struct lfs_config cfg = {
    // block device operations
    .read               = dataMan_ExtFlashRead,
    .prog               = dataMan_ExtFlashWrite,
    .erase              = dataMan_ExtFlashErase,
    .sync               = dataMan_ExtFlashSync,

    // block device configuration
    .read_size          = MAN_DATA_STORE_READ_SIZE,
    .prog_size          = MAN_DATA_STORE_WRITE_SIZE,
    .block_size         = FLASHSPI_SECTOR_SIZE,
    .block_count        = FLASHSPI_BLOCK_COUNT,
    .cache_size         = MAN_DATA_STORE_CACHE_SIZE,
    .lookahead_size     = MAN_DATA_STORE_LOOKAHEAD_SIZE,
    .block_cycles       = MAN_DATA_STORE_BLOCK_CYCLES,
    .read_buffer        = lfs_readBuff,
    .prog_buffer        = lfs_writeBuff,
    .lookahead_buffer   = lfs_lookaheadBuff,   
};
`

I found the temporary solution by using "lfs_file_opencfg" and addition  struct
`struct lfs_file_config f_cfg = {
    .buffer = lfs_file_buf,
};
`

Function call looks like
`lfs_file_open(&lfs, &file, "boot_count", LFS_O_RDWR | LFS_O_CREAT | LFS_O_APPEND);`

There is a log from FS mount and to file open

`
[main.c: 98: task_Main()] Flash Mount
lfs_trace:3574: lfs_mount(20001304, 0800d8b8 {.context=00000000, .read=08006e2d, .prog=08006ee1, .erase=08006dc9, .sync=08006ed9, .read_size=256, .prog_size=256, .block_size=4096, .block_count=8, .block_cycles=1000, .cache_size=256, .lookahead_size=256, .read_buffer=200013d0, .prog_buffer=200014d0, .lookahead_buffer=200015d0, .name_max=0, .file_max=0, .attr_max=0})

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x0
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1000
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1100
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1200
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1300
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1400
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1500
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1600
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1700
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1600
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1500
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1400
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1300
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1200
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1100
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1000
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1600
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1500
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1400
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1300
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1200
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1100
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1000
 Size 256
lfs_trace:3703: lfs_mount -> 0

[main.c: 102: task_Main()] Flash Open
lfs_trace:2466: lfs_file_open(20001304, 2000137c, "boot_count", 903)
lfs_trace:2313: lfs_file_opencfg(20001304, 2000137c, "boot_count", 903, 0800d984 {.buffer=00000000, .attrs=00000000, .attr_count=0})

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x0
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1000
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1100
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1200
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1300
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1400
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1500
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1600
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1700
 Size 256

[flashspi.c: 561: drvFlashSpi_ReadByAddress()] READ BY ADDRESS 
 Address 0x1600
 Size 256
lfs_trace:2475: lfs_file_close(20001304, 2000137c)
lfs_trace:2662: lfs_file_sync(20001304, 2000137c)
lfs_trace:2713: lfs_file_sync -> 0
lfs_trace:2494: lfs_file_close -> 0
lfs_trace:2460: lfs_file_opencfg -> -12
lfs_trace:2470: lfs_file_open -> -12

[man_data_store_init.c: 128: flashOpen()] lfs_trace:3058: lfs_file_size(20001304, 2000137c)

[main.c: 170: assert_failed()] Wrong parameters value: file src\general\Middlewares\Third_Party\littlefs\lfs.c on line 3059
`

Please confirm if there is a bug in FS or my wrong configuration??

Ralim, 2019-10-04:
Hey,
I'm using Littlefs in a static configuration.
This is the command i'm using to open the file

```
        static uint8_t uploadFileName[65]; 
	static lfs_file_t uploadFile;
	static struct lfs_file_config uploadFileConfig;
	memset(&uploadFileConfig, 0, sizeof(struct lfs_file_config));
	uploadFileConfig.buffer = lfs_read_file_buffer;  // use the static buffer
	uploadFileConfig.attr_count = 0;

	int res = lfs_file_opencfg(&lfs, &uploadFile, (char*) uploadFileName, LFS_O_RDONLY, &uploadFileConfig);
```

Remember each file needs a buffer + the three for Littlefs itself. (cache,read,write)

marekr, 2019-10-21:
Yea this is a strange break in api.

zxc503, 2019-11-04:
// Size of block caches. Each cache buffers a portion of a block in RAM.
// The littlefs needs a read cache, a program cache, and **one additional
// cache per file.** Larger caches can improve performance by storing more
// data and reducing the number of disk accesses. Must be a multiple of
// the read and program sizes, and a factor of the block size.

in struct lfs_config, it just allocate read cache and program cache,so i think it should allocate cache for each file

Ralim, 2019-11-04:
You have to manually provide the cache using the lfs_file_config struct, otherwise it will try and malloc its own cache (or fail with error codes).

marekr, 2019-11-04:
@zxc503 Yea the documentation does not match the function :P

arduzilla, 2019-11-05:
@Ralim but you you do not have to use the `lfs_file_open_cfg` function. I just use `lfs_file_open` It opens the file. But in my case when I create more than 200-300 files from time to time it fails miserably and the filesystem is corrupted. Only format recovers from it. A bit disappointing  for the "fail safe"  filesystem. Even fatFS behaves  better.

marekr, 2019-11-05:
@arduzilla wrong. You aboslutely must provide it a buffer via lfs_file_opencfg or else it uses malloc. You have to explicitly disable building with malloc (by default it does).

geky, 2020-01-27:
> Yea this is a strange break in api for static allocation :/

I'm open to other suggestions. Unfortunately littlefs is sort of straddling multiple types of systems here. Some have a full blown RTOS, malloc access is assumed, and it's expected to be able to have multiple independent filesystem running.

I figured the file config struct was an ok compromise to allows static allocations while also allowing forward-compatibility to add other file configuration in the future.

> You have to manually provide the cache using the lfs_file_config struct, otherwise it will try and malloc its own cache (or fail with error codes).

If you define LFS_NO_MALLOC, this will force those memory allocations to return LFS_ERR_NOMEM instead of still calling malloc.

Or if you only ever have one file open you can make lfs_malloc return a single buffer (not full malloc) and keep the simpler `lfs_file_open` API.

Igor-Misic, 2020-11-29:
> Hey,
> I'm using Littlefs in a static configuration.
> This is the command i'm using to open the file
>
> ```
>         static uint8_t uploadFileName[65];
> 	static lfs_file_t uploadFile;
> 	static struct [...]

Thanks for this, it helped. 
Just one addition. if anyone decides to use static memory for file each file, just make sure u use the same buffer size as you for catch buffer. 

`.cache_size = 256,`
then
` static uint8_t uploadFileName[256]; `

It took me some time to figure out why is my program randomly crashing.

geky, 2020-12-08:
Ah yeah, you're not the first person to be confused by the relationship of `read_buffer/prog_buffer` and `cache_size`. I'm considering changing the name to a more general `buffer_size` (in https://github.com/littlefs-project/littlefs/pull/491) to hopefully help with that.