littlefs-project/littlefs #305: user_provided_block_device_<> functions are not self-explaining

Protonerd, 2019-10-09:
I just now started to implement littleFS on an Adafruit Feather BLE using a Nordic nRF52 chip. In the example I see a config filled out like this:

const struct lfs_config cfg = {
    .read  = user_provided_block_device_read,
    .prog  = user_provided_block_device_prog,
    .erase = user_provided_block_device_erase,
    .sync  = user_provided_block_device_sync,
…. and so forth.

Unfortunately I can't seem to find any reference as to what function calls to substitute the user_provided_block_device_<>'s. I tried with the functions like lfs_file_read etc. from the lfs.c but it does not work, gives me an error: "sorry, unimplemented: non-trivial designated initializers not supported"

I'm using Arduino IDE. Is there a better example explaining how to configure lfs?

e107steved, 2019-10-10:
Look at the declaration for struct lfs_config in lfs.h, which is reasonably self-explanatory.
There are also some more detailed explanations in the responses to some of the issues raised.

Protonerd, 2019-10-10:
Obviously if searching the net up and down had delivered an answer I would not have asked. I red the linked issue but I got no wiser... can someone link in an example code for Arduino? Is the lib compatible with the compiler of Arduino at all?

Protonerd, 2019-10-21:
Thanks a lo sslupsky, that one looks exactly like what I’ve been looking for. I will sure give it a try!!!