littlefs-project/littlefs #307: Question: Atomic update of multiple files

thomas-ot, 2019-10-10:
Hi,

Does littlefs support atomic update of multiple files? 
If not, what would be the best approach to achieve that with littlefs?

Kind regards,
Thomas

geky, 2019-11-14:
Hi @thomas-ot, littlefs does not support atomically updating multiple files.

You may be able to get what you're looking for with a directory rename, however this would atomically update all files in the directory which may not be what you're looking for.

What exactly are you looking to achieve? Most filesystem consider files as separate atomic units. You may be able to group all of the data you are trying to update into a single file.

thomas-ot, 2019-11-15:
Hi @geky,

I need to make updates to multiple files resilient to power loss. 
It is not an option to group the data into a single file.

geky, 2019-11-27:
littlefs doesn't provide an operation like that. It only handles atomic updates for individual files or directories.

One option would by to use a lock file. Create the lock file before updating the files, and remove the lock file when you are done. If a power-cycle occurs you can detect the lock file and try to fix any out of date files.