littlefs-project/littlefs #310: lfs_file_write() return when it's no more space available

kuoyaoming93, 2019-10-22:
Hi!

I'm using **littleFS v2.1.1** for storing data in a 2Gb SLC NAND Flash Memory (Winbond W25M02GWZEIT). It consist of **2 dies of 1Gb**.  

I would like to know if it's possible to check the return value of **lfs_file_write()** in order to detect when it's no more space available. Thus I could switch to another die.

kuoyaoming93, 2019-10-22:
@geky 
Could the function **lfs_file_write** return **LFS_ERR_NOSPC**?
In this case, my problem is solved :D

geky, 2019-11-14:
Hi @kuoyaoming93, sorry about the late response. Yep littlefs will return LFS_ERR_NOSPC if it runs out of space : )

Note it can be returned from both **lfs_fs_write** and **lfs_fs_close**, since close also needs to write out metadata.

As a sidenote, when the storage device approaches the end of its life and the all blocks are dead, littlefs will return LFS_ERR_NOSPC. The filesystem models blocks wearing out as a filesystem shrinking so that code written to handle space issues should also work at the end of the devices life.

clinton-r, 2022-02-28:
Hi @geky,

So if lfs_file_write() returns a nonnegative value would that value always be equal to the 'size' parameter we passed to it?  If not, what would a returned nonnegative value != size mean?

Also, when the file system does get full, will it always still be possible to erase files?

Thanks