littlefs-project/littlefs #312: Can't get the filesystem to work properly, error returned by `lfs_dir_commitcrc`

Balint134, 2019-10-25:
Hi, feel free to close this if it is not appropriate but I'm struggling to get littlefs to work properly.
I think it must be a minor configuration issue, but I'm kinda lost in debugging right now. First of all, to reduce potential issues I've disabled all RTOS tasks in my project to reduce the chance other peripheral's and high freq. interrupts are interfering with littlefs. I'm trying to use a M25P16 NOR chip which has 32 sectors and total capacity of 16MB. [Datasheet here](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&ved=2ahUKEwiM9JzZ-bflAhWIvp4KHU-vBwcQFjAAegQIAhAC&url=https%3A%2F%2Fwww.micron.com%2F-%2Fmedia%2Fclient%2Fglobal%2Fdocuments%2Fproducts%2Fdata-sheet%2Fnor-flash%2Fserial-nor%2Fm25p%2Fm25p16.pdf&usg=AOvVaw2DG7owXowcDGXVa8Vr8Wik)
And my littlefs configuration accordingly:
```
const lfs_config cfg = {
            .context = &driver,

            .read  = &fsRead,
            .prog  = &fsProg,
            .erase = &fsErase,
            .sync  = &fsSync,

            .read_size = 64,
            .prog_size = 64,
            .block_size = 65536,
            .block_count = 32,
            .block_cycles = 500,
            .cache_size = 64,
            .lookahead_size = 64,
            .read_buffer = readBuff,
            .prog_buffer = progBuff,
            .lookahead_buffer = lookaheadBuff
        };
```

Testing the filesystem with the following code I can observe two things:
```
        // All the volatile and err reassignment are for debugging only
        volatile int err = lfs_mount(&lfs, &cfg);
        if (err) {
            err = lfs_format(&lfs, &cfg);
            err = lfs_mount(&lfs, &cfg);
        }

        lfs_file_t file;
        lfs_file_config filecfg{ .buffer = fileBuff };

        uint8_t testRead[10];
        err = lfs_file_opencfg(&lfs, &file, "test_file", LFS_O_RDWR | LFS_O_CREAT, &filecfg);
        err = lfs_file_read(&lfs, &file, testRead, 10);
```
First, setting the file name to `test_file` fails with a LFS_ERR_NAMETOOLONG error which is obviously not true as it is clearly less than 256. Piggybacking on this error it seems another error is being masked to this in `lfs_file_opencfg` which is the result of an error coming back from `lfs_dir_commitcrc`.
Interestingly changing the filename to something shorter e.g. `tf` won't fail and seemingly I can read and even write to the file although when calling `lfs_file_close` I get an LFS_ERR_CORRUPT which seems to be coming from the same `lfs_dir_commitcrc` method.

I am totally clueless at this point, I have verified (as best as I can) that my device driver is working properly. I've done so by writing 1024k data and immediately reading it back and comparing them, I've repeated this test using various random addresses.

My question is, are there any special instructions to follow when implementing the device read/prog operations? Here's my `fsRead` function (`fsProg` is pretty similar)

```
static int fsRead(const struct lfs_config *c, lfs_block_t block, lfs_off_t off, void *buffer, lfs_size_t size) {
        auto dev = reinterpret_cast<driver::M25P16 *>(c->context);
        auto data = reinterpret_cast<uint8_t *>(buffer);
        auto buff = core::io::Buffer(size, data);

        const uint32_t addr = driver::M25P16::addr_first +
                              block *
                              (driver::M25P16::page_size * driver::M25P16::num_pages / driver::M25P16::num_sectors) +
                              off;

        dev->read(addr, buff);

        return 0;
    }
``` 

My `fsSync` function is an empty function returning 0;

Balint134, 2019-10-25:
so, it has turned out that my address bytes calculation were in different endians 🤦‍♂ Sorry for polluting the project, although opening the ticket ultimately helped finding the issue :)

closed 2019-10-25 by Balint134

walterhanger, 2019-10-28:
@Balint134 Hey I think i am experiencing the same problem as you. Can you clarify what you mean by "address bytes calculation" ?

MasoudShah, 2020-09-07:
Hello @Balint134 . I have faced the same issue as you did. The error occurs in lfs_dir_commitcrc function and eventually it returns LFS_ERR_NAMETOOLONG error, but this happens for the first time I want to create a file. After that when I mount the filesystem again, I can open the file and read or write into it. What do you mean exactly by different endians in address bytes calculation? Maybe that would help, solving my problem.

Balint134, 2020-09-08:
Hi @MasoudShah I'm using a M26P16 Flash chip and the addressing of that chip is 3 bytes in little endian format. In my case I accidentally sent the address as big endian which resulted in this issue.

MasoudShah, 2020-09-09:
Thank you for your guide @Balint134 . I didn't have issue with my addressing system but adding delay between some functions, solved my problem. Maybe it's the matter of synchronizing.