littlefs-project/littlefs #316: Seed bad block table

sslupsky, 2019-10-28:
I am working with a Toshiba TC58CVG2 spi nand flash chip.  Toshiba indicates that a new device can have some bad blocks.  Toshiba indicates these blocks are marked at the beginning of the block and recommends that the user scan all blocks to find the initial bad blocks before using the chip.  The device doesn't manage bad blocks so it is left to the user the manage the bad blocks.

My question is, how can I pass the initial bad block information along to LittleFS?  I cannot find a way to tell LittleFS about the existing bad blocks.

geky, 2019-11-14:
We don't have a great solution for this currently. LittleFS doesn't currently track free blocks so unfortunately the only option currently is to handle it on the block device side.

If you know which blocks are bad during erase, returning LFS_ERR_CORRUPT from the block device erase operation will inform LittleFS to find a new block.

However if the bad blocks are marked with a magic byte pattern it's possible LittleFS may happen to write the magic pattern by accident. Causing valid blocks to be marked as bad.

One option that requires more work is reserving the first block of storage to record which blocks are bad, and use that to tell LittleFS which blocks are bad during erases.

This is something we realize needs to be added to the filesystem, but unfortunately don't have a built in solution yet.

stsymbaliuk, 2021-02-25:
@geky 
I use MT29F1G08ABAEAWP NAND flash witch have spare areas. Manufacturer marks a first spare area of each bad block with zeros. I've added the first block spare area check and return LFS_ERR_CORRUPT from erase function. As LittleFS does not white to spare area so we can write any magic byte pattern (or zeroes as the manufacturer do) to mark new bad blocks.

My question is: when a new bad block appears during lifetime how can we mark it bad? 

For example: block `0xa` is bad from the factory. Block `0x9` is a new bad block from which data are read with error.
I've got the same debug messages `Bad block at` for both blocks. But block which are not marked as bad will be erased again.

For block `0xa` (LFS_ERR_CORRUPT returned from erase function):
```
D:/Work/littlefs_test/Src/littlefs/lfs.c:1786:debug: Bad block at 0xa
```
For block `0x9` (LittleFS reads data but data is corrupted):
```
D:/Work/littlefs_test/Src/littlefs/lfs.c:1786:debug: Bad block at 0x9
```
Code line which show debug message:
https://github.com/littlefs-project/littlefs/blob/1863dc7883d82bd6ca79faa164b65341064d1c16/lfs.c#L1790

Maybe there is some callback to set? Where is the best place to put a new bad block marking function call?

marcosgabrielserrano, 2024-11-13:
@stsymbaliuk Did you ever find/make a solution to this problem?