littlefs-project/littlefs #319: lfs_mount may return LFS_ERR_INVAL when suddenly reboot.

mttbx, 2019-11-05:
I test lfs with a chip of sd card. No error return, and everything are fine (LFS_ERR_OK). But when I suddenly reboot, sometimes lfs_mount may return  LFS_ERR_INVAL.

lfs config:
```
#define SD_DEFAULT_BLOCK_SIZE 512
#define LFT_DEFAULT_MULTI_OF_BLOCK_SIZE 8
const struct lfs_config cfg = {
    // block device operations
    .read  = lfs_port_read,
    .prog  = lfs_port_prog,
    .erase = lfs_port_erase,
    .sync  = lfs_port_sync,

    // block device configuration
    .read_size = SD_DEFAULT_BLOCK_SIZE,
    .prog_size = SD_DEFAULT_BLOCK_SIZE,
    .block_size = SD_DEFAULT_BLOCK_SIZE * LFT_DEFAULT_MULTI_OF_BLOCK_SIZE,
    .block_count = 512,
    .cache_size = SD_DEFAULT_BLOCK_SIZE,
    .lookahead_size = SD_DEFAULT_BLOCK_SIZE,
    .block_cycles = 500,
};
```

test program:
  ```
int err = lfs_mount(&lfs, &cfg);
  if (err) {
    lfs_format(&lfs, &cfg);
    lfs_mount(&lfs, &cfg);
  }
  while (1)
  {
    uint32_t t1, t2;
    int res;
    t1 = HAL_GetTick();
    res = lfs_file_open(&lfs, &file, "boot_count", LFS_O_RDWR | LFS_O_CREAT);
    if (res < LFS_ERR_OK) while (1);
    res = lfs_file_read(&lfs, &file, &boot_count, sizeof(boot_count));
    if (res < LFS_ERR_OK) while (1);

    // update boot count
    boot_count += 1;
    res = lfs_file_rewind(&lfs, &file);
    if (res < LFS_ERR_OK) while (1);
    res = lfs_file_write(&lfs, &file, &boot_count, sizeof(boot_count));
    if (res < LFS_ERR_OK) while (1);

    // remember the storage is not updated until the file is closed successfully
    res = lfs_file_close(&lfs, &file);
    if (res < LFS_ERR_OK) while (1);
    t2 = HAL_GetTick();
    printf_small("boot_count: %d, time: %d\n", boot_count, t2-t1);

    HAL_Delay(1000);
  }

```

code in lfs_mount:
```
in function lfs_mount

    // found superblock?
    if (lfs_pair_isnull(lfs->root)) {
        err = LFS_ERR_INVAL;           <------------- when reboot, code enter here
        goto cleanup;
    }
```

mttbx, 2019-11-05:
SD card normally rw with block, so I set read_size / write_size to 512
When I need to rw my sd card I calculate how may blocks should I rw in SD card.
block address of sd card :
 (block * LFT_DEFAULT_MULTI_OF_BLOCK_SIZE + off / SD_DEFAULT_BLOCK_SIZE),
size of rw (in block size):
size / SD_DEFAULT_BLOCK_SIZE

mttbx, 2019-11-05:
I also find that lfs_mount may return LFS_ERR_NOENT when I suddenly reboot

code in lfs_mount:
```
in function lfs_mount
        // fetch next block in tail list
        lfs_stag_t tag = lfs_dir_fetchmatch(lfs, &dir, dir.tail,
                LFS_MKTAG(0x7ff, 0x3ff, 0),
                LFS_MKTAG(LFS_TYPE_SUPERBLOCK, 0, 8),
                NULL,
                lfs_dir_find_match, &(struct lfs_dir_find_match){
                    lfs, "littlefs", 8});
        if (tag < 0) {
            err = tag;                      <---------------------when reboot, code enter here
            goto cleanup;
        }
```

e107steved, 2019-11-06:
How are you rebooting?
If you don't cycle the power, possibly the SD card is left in a state which means that initial operations fail, unless your startup code recovers properly

mttbx, 2019-11-07:
@e107steved I just cut the power, and then reboot it. How should I recover? Can you give me some advice ?

e107steved, 2019-11-13:
I was thinking more of a situation where you push the reset button, say - it could leave the memory in an unusual state where the first few transactions don't work as intended. You could get a similar situation if you just turn the power off momentarily; capacitors may not discharge fully, again resulting in an unusual state.

There is often a sequence of commands you can send which will guarantee the device is properly reset; I don't know what it is for SD cards, so suggest you look at what other sources do - there's likely to be a driver in the mbed library, for example

geky, 2019-11-14:
From what I can tell your configuration looks good.

On top of @e107steved suggestion, another thing it could be is an issue with the SD card's caches. Does the SD card have a sync command you could send it in lfs_port_sync?

mttbx, 2019-11-15:
Thank you guys. Yes there's a sequence of commands I should send to sd card so that the device can properly reset. I solve this problem to add a big capacitors so that when the power is off I can send these commands. I don't find sync command, I do find a file called JESD 84-B51. It suggest that I must send a sequence of commands to close it before the power off when writing.

closed 2019-11-15 by mttbx