littlefs-project/littlefs #323: Why the cost is n + n * s / ( d + 1 ) ?

mttbx, 2019-11-06:
Hi, I just read the design.md. And I don't know why the cost is n + n * s / ( d + 1 ) ? Can you give some explain? @geky

geky, 2019-11-14:
Right, it is quite a bit of a jump. [(context for others)](https://github.com/ARMmbed/littlefs/blob/master/DESIGN.md#metadata-pairs).

First thing to note is we assume all the entries are the same size (![n](https://latex.codecogs.com/svg.latex?n)). This isn't true in practice, but it helps with the math.

So if we look at what garbage collection looks like, we take a set of entries and remove those that are outdated.

```
.----------------.    .----------------.
|       A        | => |       A''      |
|----------------|    |----------------|
|       B        |    |       B'       |
|----------------|    |----------------|
|       C        |    |       C        |
|----------------|    |----------------|
|       A'       |    |       |        |
|----------------|    |       v        |
|       A''      |    |                |
|----------------|    |                |
|       B'       |    |                |
'----------------'    '----------------'
```

We can call the entries that change dynamic, and the entries that do change static. We use these terms just because they are more common in writing about filesystems.

In the above example we have A, A', A'', B, B', and C. That gives us 3 static entries (![s](https://latex.codecogs.com/svg.latex?s)) that go unchanged during our garbage collection, and 3 dynamic entries (![d](https://latex.codecogs.com/svg.latex?d)) we can remove during garbage collection.

Now lets look at how much work garbage collection takes. The exact cost depends on the underlying algorithm, but we know at minimum we will need to copy over all the static entries. Each entry is ![n](https://latex.codecogs.com/svg.latex?n) bytes, so we can say garbage collection costs roughly ![n*s](https://latex.codecogs.com/svg.latex?n%5Ccdot%20s).

The interesting part comes when we look at the _amortized_ cost, that is the cost per update. The important thing here is that garbage collection doesn't happen every update (hopefully), it only happens with a block is full. If we start from an empty block, this would give us the size number of updates before garbage collection. But unfortunately we don't get to start with an empty block usually, because those static entries stick around. We only get the number of entries cleaned during the previous garbage collection. Once we've filled the cleaned entries we need to garbage collect again.

Note that the number of cleaned entries is ![d](https://latex.codecogs.com/svg.latex?d). We need to do garbage collection work ![n*s](https://latex.codecogs.com/svg.latex?n%5Ccdot%20s) every ![d](https://latex.codecogs.com/svg.latex?d) updates.

Ah, but garbage collection itself can update an entry that never lands on the previous disk. This sort of ephemeral update gets us to ![d+1](https://latex.codecogs.com/svg.latex?d&plus;1) updates. It's a small improvement but important because it means we can make progress even if the block is completely full.

```
.----------------.    .----------------.
|       A        | => |       A''      |
|----------------|    |----------------|
|       B        |    |       B''      |
|----------------|    |----------------|
|       C        |    |       C        |
|----------------|    |----------------|
|       A'       |    |       |        |
|----------------|    |       v        |
|       A''      |    |                |
|----------------|    |                |
|       B'       |    |                |
'----------------'    '----------------'
 ................
:       B''      : <- update that triggers garbage collection
 ''''''''''''''''
```

So we do ![n*s](https://latex.codecogs.com/svg.latex?n%5Ccdot%20s) work every ![d+1](https://latex.codecogs.com/svg.latex?d&plus;1) updates.

Ah, but we can't forget that there is also work to update the entry itself, even if there is no garbage collection. That's where the extra constant ![n](https://latex.codecogs.com/svg.latex?n) work comes from.

So every update we do ![n](https://latex.codecogs.com/svg.latex?n) work, and every ![d+1](https://latex.codecogs.com/svg.latex?d&plus;1) updates we do ![n*s](https://latex.codecogs.com/svg.latex?n%5Ccdot%20s) work. This gives us our final formula for amortized work:

![n+n*s/(d+1)](https://latex.codecogs.com/svg.latex?cost%20%3D%20n%20&plus;%20n%20%5Cfrac%7Bs%7D%7Bd&plus;1%7D)

---

Hopefully that helps a bit. There is quite a bit of handwaving because we don't need an _exact_ formula, just something to work with to get a better understanding of how the performance behaves as the number of entries scales.

mttbx, 2019-11-15:
Good explaination, thank you! I understand this formula now.

closed 2019-11-15 by mttbx