littlefs-project/littlefs #324: How to migrate from v1.x to 2.x?

svenifax, 2019-11-06:
Hi,

I would like to upgrade to version 2.x. The system is running v1.x. How can I upgrade?

My steps would be:
- Check existing file system version
- if this existing version is 1.x the call lfs_migrate https://github.com/ARMmbed/littlefs/pull/85/commits/4ad09d6c4ec95c126909b905900237cdc829c6b0

But, how do I know the version number of the already existing file system? I couldn't found any helper function. Should I try to mount with the 2.x version?

If the upgrade fails, is it possible to use the 1.x version to mount, that means can I use 1.x and 2.x version of the library at the same time? Is there a configurable prefix or anything like that?

Best regards,
Sven

e107steved, 2019-11-13:
Have a look at the comments with the releases (https://github.com/ARMmbed/littlefs/pull/85/commits/4ad09d6c4ec95c126909b905900237cdc829c6b0) and also https://github.com/ARMmbed/littlefs/pull/85/commits/4ad09d6c4ec95c126909b905900237cdc829c6b0 . This might help.
You might have to experiment, but basically a V1 file system looks as if it won't mount using V2 code.

There are some discussions about adding prefixes to the API entry points so that you could support V1 and V2 together; not sure of the state of that

geky, 2019-11-14:
The **Migration** section of the v2 release notes probably has the best info:
https://github.com/ARMmbed/littlefs/releases/tag/v2.0.0

@e107steved is correct the best way to figure out which filesystem is on disk is to mount with the version you want to check. The mount should fail with no side-effects if it isn't the correct version.

We would actually need to mount with a specific version before we could tell you which version the filesystem is, so a helper function wouldn't be very helpful.

This is one way to make sure you have v2 mounted:

``` c
/// try mounting
int err = lfs2_mount(&lfs, &cfg);
if (err) {
    err = lfs2_migrate(&lfs, &cfg);
    if (err) {
        // could not migrate, lfs1 fallback here
        // note we also get here if there is no lfs on disk at all
    }
}

// v2 should be mounted at this point
```

---

There is a prefixed version now 👍 This will let you compile both v1 and v2 into the same binary.

The scripts/prefix.py script lets you assign arbitrary prefixes. Or you can just use the [v2-prefix](https://github.com/ARMmbed/littlefs/tree/v2-prefix) branch. The branch is auto-updated in such a way that you can `git pull origin v2-prefix` just as you would with master.