littlefs-project/littlefs #331: README: fix incorrect description

gmpy, 2019-11-15:
commit 5aa508bc0963f4c6589636c9b105f73d59ccf6fb
Author: liaoweixiong <liaoweixiong@allwinnertech.com>
Date:   Fri Nov 15 18:53:53 2019 +0800

    README: fix incorrect description
    
    In my point of view, file updates will commit to filesystem only when
    sync or close. There is a extra word 'no' here.
    
    Fixes: bdff4bc59eb6 ("Updated DESIGN.md to reflect v2 changes")
    Signed-off-by: liaoweixiong <liaoweixiong@allwinnertech.com>

diff --git a/README.md b/README.md
index 8017983..41d5890 100644
--- a/README.md
+++ b/README.md
@@ -110,7 +110,7 @@ directory functions, with the deviation that the allocation of filesystem
 structures must be provided by the user.
 
 All POSIX operations, such as remove and rename, are atomic, even in event
-of power-loss. Additionally, no file updates are not actually committed to
+of power-loss. Additionally, file updates are not actually committed to
 the filesystem until sync or close is called on the file.
 
 ## Other notes

geky, 2019-11-26:
Hi @gmpy, thanks for this. I've turned this into a PR here: https://github.com/ARMmbed/littlefs/pull/338 and will merge it when I can.

geky, 2019-12-19:
I believe this is resolved, feel free to reopen if otherwise.

closed 2019-12-19 by geky