littlefs-project/littlefs #333: how to achieve the "XOR linked list"

Johnxjj, 2019-11-20:


geky, 2019-11-20:
Hi @Johnxjj is this what you're looking for?
https://en.wikipedia.org/wiki/XOR_linked_list

XOR linked lists are a data-structure trick that is nice to know about.

Johnxjj, 2019-11-20:
[image]
In the SPEC.md, I saw the content on this picture, but I don't understand it very well. I guess it may be related to the XOR list, but the discovery is different. Can you explain in more detail why this XOR processing is done here? What would happen if I didn't use XOR? What part of the space is saved?

Sorry, I can't access the link above. Maybe I need to over the wall.

geky, 2019-11-26:
Hi @johnxjj, sure thing. Is this accessible over there? https://stackoverflow.com/questions/16138998/how-exactly-does-a-xor-linked-list-work. It may help to make sense what is going on.

---

The trick we used with XOR is that it is symmetrical. This means if we XOR two things together:
```
z = x xor y
```
We can then use knowledge of one of the variables to extract the other:
```
x = z xor y
y = z xor x
```

Keeping this trick in mind, the problem we are wanting to solve is to store variable-length metadata in a metadata block so we can read it in two different ways:
1. In commit order - This is how we "fetch" the metadata block. We need to read "in-order" to be able to fall back to a previous commit if we find a bad CRC.
2. In reverse-commit order - We do this when we look up a piece of metadata. Once we find a piece of metadata we know it is the most recent since we are scanning in reverse-commit order.

To mark what each piece of metadata is, we use a 32-bit tag which contains some identifying information and a size.

However the tricky bit is that the metadata is variably sized. This means we can't store a single tag for each piece of metadata. When we scan, either forward or backward, we won't know how far the next tag is.

```
.-------------------. |
|  revision count   | |
|-------------------| |
|  tag A length 4   | v Tag B is 4 bytes away
|-------------------| :
|       data A      | :
|                   | :
|-------------------| :
|  tag B length 5   | v We can scan in one direction
|-------------------|
|       data B      |
|                   |
|                   |
|-------------------| 
|  tag C length 2   | ^ How far is tag B?
|-------------------| |
|       data C      | |
'-------------------' |
```

One solution is to put tags on both ends of the metadata. This is done often for dynamic memory heaps for example.

```
.-------------------. |
|  revision count   | |
|-------------------| |
|  tag A length 4   | v Tag B is 4 bytes away
|-------------------| :
|       data A      | :
|                   | :
|-------------------| :
|  tag A length 4   | :
|-------------------| :
|  tag B length 5   | v We can scan in both directions
|-------------------|
|       data B      |
|                   |
|                   |
|-------------------|
|  tag B length 5   | ^
|-------------------| :
|  tag C length 2   | :
|-------------------| :
|       data C      | :
|-------------------| :
|  tag C length 2   | ^ Tag B is 2 bytes away
'-------------------' |
```

However, this adds the cost of needing to store two tags. Considering that some metadata, such as directory pointers, are only ~8 bytes, and tags are 4 bytes, this adds an amount of overhead for each piece of metadata.

Fortunately, we can use the XOR trick to be a bit clever. When we're iterating through the metadata-pairs, we know the previous tag, it is effectively redundant information. If we XOR both the next and previous tags together, we can use the symmetry of the XOR trick to extract the next tag if we know the previous tag.

This effectively shoves the two tags together and cuts the overhead in half.

Here is an example with XORing the lengths together, though in practice we also XOR the tag's identifying information together:

```
.-------------------. |
|  revision count   | |
|-------------------| |
|tag A length 0x4=4 | v Tag B is 4x0=4 bytes away
|-------------------| :
|       data A      | :
|                   | :
|-------------------| :
|tag A length 4x5=1 | v Tag C is 1x4=5 bytes away
|-------------------|
|       data B      |
|                   |
|                   |
|-------------------|
|tag B length 5x2=7 | ^ Tag A is 7x2=5 bytes away
|-------------------| :
|       data C      | :
|-------------------| :
|tag C length 2x0=2 | ^ Tag B is 2x0=2 bytes away
'-------------------' |
```

The main downside is that this technique is a bit complicated and can be difficult to follow. But hopefully this helps explain it a bit.

I mention the XOR-linked list, because it is a relatively common use of the XOR trick, though I haven't seen anyone apply the same technique to linear memory.

Though similar, the XOR trick isn't useful for dynamic memory heaps because you also need random access to the tags. It only works here because we never need to access a tag randomly in the middle of a metadata pair (we wouldn't know if it is outdated anyway).

Johnxjj, 2019-11-28:
@geky Thank you very much for your reply, it helped me a lot！

closed 2019-11-28 by Johnxjj