littlefs-project/littlefs #335: Directory content wipped when N files are created

pokitoz, 2019-11-22:
Hello!

I am using littlefs on a STM32L4 with the QSPI Flash memory.
I tried the example given to create files (which works fine) and wanted to try creating multiple files to remove them later. However, it seems that the files are deleted (or at least the directory structure cannot keep track of them).

The code is:

```
for (int i = 0; i < 30; i++)
  {
    lfs_file_t fd;
    int result = 0;
    char t[] = "/a";
    t[1] = t[1] + i;
    result = lfs_file_open(&fs, &fd, t, LFS_O_RDWR | LFS_O_CREAT);
    printf("open %d", result);

    result = lfs_file_write(&fs, &fd, &i, sizeof(i));
    printf("write %d", result);

    result = lfs_file_close(&fs, &fd);
    printf("close %d", result);

    sListFiles();
}
```

Where sListFiles is implemented as follow:
```
void sListFiles(void)
{
    lfs_dir_t dir;
    lfs_file_t fd;
    int resultFileSystem = 0;
    struct lfs_info info;

    printf("List file");
    resultFileSystem = lfs_dir_open(&gLittlefsFs, &dir, "/");
    if (LFS_ERR_OK != resultFileSystem) {
      return;
    }

    int statusRead = lfs_dir_read(&gLittlefsFs, &dir, &info);
    while (0 < statusRead)
    {
      if ((LFS_TYPE_REG == info.type) || (LFS_TYPE_DIR == info.type))
      {
        printf("- %s", info.name);
      }
      statusRead = lfs_dir_read(&gLittlefsFs, &dir, &info);
    }

    lfs_dir_close(&gLittlefsFs, &dir);
}
```

I am getting the following output, where I am expecting having all the files listed one by one. The file `h` disappears and then it seems that the files are wiped out.
Any idea where the issue could come from ?

```
 open 0
 write 4
 close 0
 List file
 - .
 - ..
 - a
 open 0
 write 4
 close 0
 List file
 - .
 - ..
 - a
 - b
 open 0
 write 4
 close 0
 List file
 - .
 - ..
 - a
 - b
 - c
 open 0
 write 4
 close 0
 List file
 - .
 - ..
 - a
 - b
 - c
 - d
 open 0
 write 4
 close 0
 List file
 - .
 - ..
 - a
 - b
 - c
 - d
 - e
 open 0
 write 4
 close 0
 List file
 - .
 - ..
 - a
 - b
 - c
 - d
 - e
 - f
 open 0
 write 4
 close 0
 List file
 - .
 - ..
 - a
 - b
 - c
 - d
 - e
 - f
 - g
 open 0
 write 4
 close 0
 List file
 - .
 - ..
 open 0
 write 4
 close 0
 List file
 - .
 - ..
 - i
 open 0
 write 4
 close 0
 List file
 - .
 - ..
 - i
 - j
 open 0
 write 4
 close 0
 List file
 - .
 - ..
 - i
 - j
 - k
 open 0
 write 4
 close 0
 List file
 - .
 - ..
 - i
 - j
 - k
 - l
 open 0
 write 4
 close 0
 List file
 - .
 - ..
 - i
 - j
 - k
 - l
 - m
 open 0
 write 4
 close 0
 List file
 - .
 - ..
 - i
 - j
 - k
 - l
 - m
 - n
 open 0
 write 4
 close 0
 List file
 - .
 - ..
 - i
 - j
 - k
 - l
 - m
 - n
 - o

...
```

geky, 2019-11-26:
I think this may be the same bug fixed in https://github.com/ARMmbed/littlefs/pull/336. Does https://github.com/ARMmbed/littlefs/pull/336 solve the issue?

pokitoz, 2019-11-28:
Unfortunatly, it did not fix the issue. I enabled the logs if that helps:

```
[Nov 28 2019|18:52:17] Bootup STM32L4
lfs_trace:3595: lfs_mount(0x20011640, 0x8025648 {.context=0x0, .read=0x80108d5,                                                              .prog=0x8010905, .erase=0x8010935, .sync=0x8010959, .read_size=256, .prog_size=2                                                             56, .block_size=4096, .block_count=2048, .block_cycles=100, .cache_size=512, .lo                                                             okahead_size=256, .read_buffer=0x200116b8, .prog_buffer=0x0, .lookahead_buffer=0                                                             x0, .name_max=0, .file_max=0, .attr_max=0})
lfs_trace:3719: lfs_unmount(0x20011640)
lfs_trace:3721: lfs_unmount -> 0
lfs_trace:3714: lfs_mount -> -22
Format filesystem because of error -22
lfs_trace:3514: lfs_format(0x20011640, 0x8025648 {.context=0x0, .read=0x80108d5,                                                              .prog=0x8010905, .erase=0x8010935, .sync=0x8010959, .read_size=256, .prog_size=                                                             256, .block_size=4096, .block_count=2048, .block_cycles=100, .cache_size=512, .l                                                             ookahead_size=256, .read_buffer=0x200116b8, .prog_buffer=0x0, .lookahead_buffer=                                                             0x0, .name_max=0, .file_max=0, .attr_max=0})
lfs_trace:3575: lfs_format -> 0
lfs_trace:3595: lfs_mount(0x20011640, 0x8025648 {.context=0x0, .read=0x80108d5,                                                              .prog=0x8010905, .erase=0x8010935, .sync=0x8010959, .read_size=256, .prog_size=2                                                             56, .block_size=4096, .block_count=2048, .block_cycles=100, .cache_size=512, .lo                                                             okahead_size=256, .read_buffer=0x200116b8, .prog_buffer=0x0, .lookahead_buffer=0                                                             x0, .name_max=0, .file_max=0, .attr_max=0})
lfs_trace:3709: lfs_mount -> 0
lfs_trace:2473: lfs_file_open(0x20011640, 0x20017f94, "/a", 103)
lfs_trace:2322: lfs_file_opencfg(0x20011640, 0x20017f94, "/a", 103, 0x80255ac {.                                                             buffer=0x0, .attrs=0x0, .attr_count=0})
lfs_trace:2459: lfs_file_opencfg -> 0
lfs_trace:2476: lfs_file_open -> 0
open 0
lfs_trace:2816: lfs_file_write(0x20011640, 0x20017f94, 0x20017fec, 4)
lfs_trace:2940: lfs_file_write -> 4
write 4
lfs_trace:2481: lfs_file_close(0x20011640, 0x20017f94)
lfs_trace:2668: lfs_file_sync(0x20011640, 0x20017f94)
lfs_trace:2719: lfs_file_sync -> 0
lfs_trace:2500: lfs_file_close -> 0
close 0
List file
lfs_trace:1942: lfs_dir_open(0x20011640, 0x20017f44, "/")
lfs_trace:1988: lfs_dir_open -> 0
lfs_trace:2008: lfs_dir_read(0x20011640, 0x20017f44, 0x20017de8)
lfs_trace:2016: lfs_dir_read -> 1
- .
lfs_trace:2008: lfs_dir_read(0x20011640, 0x20017f44, 0x20017de8)
lfs_trace:2022: lfs_dir_read -> 1
- ..
lfs_trace:2008: lfs_dir_read(0x20011640, 0x20017f44, 0x20017de8)
lfs_trace:2055: lfs_dir_read -> 1
- a
lfs_trace:2008: lfs_dir_read(0x20011640, 0x20017f44, 0x20017de8)
lfs_trace:2029: lfs_dir_read -> 0
lfs_trace:1993: lfs_dir_close(0x20011640, 0x20017f44)
lfs_trace:2002: lfs_dir_close -> 0
lfs_trace:2473: lfs_file_open(0x20011640, 0x20017f94, "/b", 103)
lfs_trace:2322: lfs_file_opencfg(0x20011640, 0x20017f94, "/b", 103, 0x80255ac {.                                                             buffer=0x0, .attrs=0x0, .attr_count=0})
lfs_trace:2459: lfs_file_opencfg -> 0
lfs_trace:2476: lfs_file_open -> 0
open 0
lfs_trace:2816: lfs_file_write(0x20011640, 0x20017f94, 0x20017fec, 4)
lfs_trace:2940: lfs_file_write -> 4
write 4
lfs_trace:2481: lfs_file_close(0x20011640, 0x20017f94)
lfs_trace:2668: lfs_file_sync(0x20011640, 0x20017f94)
lfs_trace:2719: lfs_file_sync -> 0
lfs_trace:2500: lfs_file_close -> 0
close 0
List file
lfs_trace:1942: lfs_dir_open(0x20011640, 0x20017f44, "/")
lfs_trace:1988: lfs_dir_open -> 0
lfs_trace:2008: lfs_dir_read(0x20011640, 0x20017f44, 0x20017de8)
lfs_trace:2016: lfs_dir_read -> 1
- .
lfs_trace:2008: lfs_dir_read(0x20011640, 0x20017f44, 0x20017de8)
lfs_trace:2022: lfs_dir_read -> 1
- ..
lfs_trace:2008: lfs_dir_read(0x20011640, 0x20017f44, 0x20017de8)
lfs_trace:2055: lfs_dir_read -> 1
- a
lfs_trace:2008: lfs_dir_read(0x20011640, 0x20017f44, 0x20017de8)
lfs_trace:2055: lfs_dir_read -> 1
- b
[... 211 lines cut ...]
lfs_trace:2459: lfs_file_opencfg -> 0
lfs_trace:2476: lfs_file_open -> 0
open 0
lfs_trace:2816: lfs_file_write(0x20011640, 0x20017f94, 0x20017fec, 4)
lfs_trace:2940: lfs_file_write -> 4
write 4
lfs_trace:2481: lfs_file_close(0x20011640, 0x20017f94)
lfs_trace:2668: lfs_file_sync(0x20011640, 0x20017f94)
lfs_trace:2719: lfs_file_sync -> 0
lfs_trace:2500: lfs_file_close -> 0
close 0
List file
lfs_trace:1942: lfs_dir_open(0x20011640, 0x20017f44, "/")
lfs_trace:1988: lfs_dir_open -> 0
lfs_trace:2008: lfs_dir_read(0x20011640, 0x20017f44, 0x20017de8)
lfs_trace:2016: lfs_dir_read -> 1
- .
lfs_trace:2008: lfs_dir_read(0x20011640, 0x20017f44, 0x20017de8)
lfs_trace:2022: lfs_dir_read -> 1
- ..
lfs_trace:2008: lfs_dir_read(0x20011640, 0x20017f44, 0x20017de8)
lfs_trace:2029: lfs_dir_read -> 0
lfs_trace:1993: lfs_dir_close(0x20011640, 0x20017f44)
lfs_trace:2002: lfs_dir_close -> 0
```

geky, 2019-12-01:
That log actually does help quite a bit 👍 It has the geometry of the device so I was able to plug your reproducible example into emubd and test locally.

If you write this to main.c and run `make && ./lfs` should should be run the example locally as well in case it helps:

``` c
#include "lfs.h"
#include "emubd/lfs_emubd.h"

// lfs declarations
lfs_t lfs;
lfs_emubd_t bd;

uint8_t read_buffer[512];

const struct lfs_config cfg = {
    .context = &bd,
    .read  = &lfs_emubd_read,
    .prog  = &lfs_emubd_prog,
    .erase = &lfs_emubd_erase,
    .sync  = &lfs_emubd_sync,

    .read_size      = 256,
    .prog_size      = 256,
    .block_size     = 4096,
    .block_count    = 2048,
    .block_cycles   = 100,
    .cache_size     = 512,
    .lookahead_size = 256,
};

// list files
void sListFiles(void) {
    lfs_dir_t dir;
    lfs_file_t fd;
    int resultFileSystem = 0;
    struct lfs_info info;

    printf("List file\n");
    resultFileSystem = lfs_dir_open(&lfs, &dir, "/");
    if (LFS_ERR_OK != resultFileSystem) {
      return;
    }

    int statusRead = lfs_dir_read(&lfs, &dir, &info);
    while (0 < statusRead)
    {
      if ((LFS_TYPE_REG == info.type) || (LFS_TYPE_DIR == info.type))
      {
        printf("- %s\n", info.name);
      }
      statusRead = lfs_dir_read(&lfs, &dir, &info);
    }

    lfs_dir_close(&lfs, &dir);
}

// Entry point
int main(void) {
    lfs_emubd_create(&cfg, "blocks");

    lfs_format(&lfs, &cfg);
    lfs_mount(&lfs, &cfg);
    for (int i = 0; i < 30; i++) {
        lfs_file_t fd;
        int result = 0;
        char t[] = "/a";
        t[1] = t[1] + i;
        result = lfs_file_open(&lfs, &fd, t, LFS_O_RDWR | LFS_O_CREAT);
        printf("open %d\n", result);

        result = lfs_file_write(&lfs, &fd, &i, sizeof(i));
        printf("write %d\n", result);

        result = lfs_file_close(&lfs, &fd);
        printf("close %d\n", result);

        sListFiles();
    }
    lfs_unmount(&lfs);

    lfs_emubd_destroy(&cfg);
}
```

That being said, running it locally didn't reproduce the failure. There are some differences running locally, such as the block device being initialized to zero, but this suggests it may be a configuration issue.

I noticed you provided a `read_buffer` but not a `prog_buffer`. `prog_buffer=NULL` will still call malloc. I don't know if this is intended.

I also noticed `read_size=256` but `cache_size=512`. The naming here is unfortunate, but `read_buffer` must be at least `cache_size` bytes large. So I would double check that the size of `read_buffer` is 512.

pokitoz, 2019-12-02:
As you suggested, I checked the size of the buffers. I updated the config to be:

    ```
    uint8_t lfs_read_buf[4096];
    uint8_t lfs_prog_buf[512];
    uint8_t lfs_lookahead_buf[512];

    .block_size = 4096,
    .block_count = 2048,
    .cache_size = 4096,
    .block_cycles = 100,

    .read_buffer = lfs_read_buf,
    .read_size = sizeof(lfs_read_buf),

    .prog_buffer = lfs_prog_buf,
    .prog_size = sizeof(lfs_prog_buf),

    .lookahead_buffer = lfs_lookahead_buf,
    .lookahead_size = sizeof(lfs_lookahead_buf),
```

And it seems that my read buffer and my cache buffer was too small before (512 bytes updated to 4096) as the block size is 4096. I guess an assert is checking this but I disabled the asserts 🤕 .
Thanks for your support!

closed 2019-12-02 by pokitoz

geky, 2019-12-02:
You may not need to make your read buffer 4096 bytes. I think the confusion is that `sizeof(prog_buf) != prog_size`. They are (unfortunately) not related in that way. Instead `sizeof(prog_buf) == cache_size`.

So this configuration may also work and use less RAM?

``` c
uint8_t lfs_read_buf[512];
uint8_t lfs_prog_buf[512];
uint8_t lfs_lookahead_buf[512];

.read_size = 512,
.prog_size = 512,
.block_size = 4096,
.block_count = 2048,
.block_cycles = 100,

.cache_size = min(sizeof(lfs_prog_buf), sizeof(lfs_read_buf)),
.read_buffer = lfs_read_buf,
.prog_buffer = lfs_prog_buf,

.lookahead_buffer = lfs_lookahead_buf,
.lookahead_size = sizeof(lfs_lookahead_buf),
```