littlefs-project/littlefs #341: slow speed the first time write and every 128KB write

summer20100514, 2019-11-28:
hello, I use littlefs on SD card, I have increased the block_size to 1MBytes, but I still found low speed write when I do the tests.
1. write several big file using littlefs-fuse on ubuntu. every file is about 20-80M big. total file size is about 240M
2. on my board mount littlefs, and create a new file and write 1KBytes data every loop till to 1MBytes. open/write/close the file every loop.
I found that the first time create/write needs about 80s to complete. What's worse, the fllowing 1KBytes write slows down to around 70s to complete after approx. 128KBytes data has been written.
here is my configurations:
```
    .read_size = 512,
    .prog_size = 512,
    .block_size = 512*2048,
    .block_count = 400,
    .cache_size = 1024,
    .lookahead_size = 64,
    .block_cycles = 500,
```
thank you if you can help me point out what's going on here.

Johnxjj, 2019-11-28:
I guess the result is this: each time you reopen the file and write 1KB, there will be a process of moving blocks, which will cause you to write 1KB for the first time, 2KB for the second time, ... 1024KB for the 1024th time, so Actually wrote about 512M of data. This is related to the wear leveling mechanism of littlefs.

summer20100514, 2019-11-28:
@Johnxjj @geky here is the log file
[1MBytes seppd test.txt](https://github.com/ARMmbed/littlefs/files/3900436/1MBytes.seppd.test.txt)

Johnxjj, 2019-11-28:
It looks like the superblock migration has happened. You should print out the specific write block (which block) and erase block (which block) so that you can analyze where the time is spent.

geky, 2019-12-01:
Hi @summer20100514, thanks for creating an issue.

This look related https://github.com/ARMmbed/littlefs/issues/75, but it looks like you've already gone for the workaround of increasing the block size, so I suspect you may now be running into https://github.com/ARMmbed/littlefs/issues/214? So it may be worth trying to scale the block size down a bit and see if it improves performance.

Though honestly, littlefs just doesn't scale to larger sizes well. It's an area I have some ideas on, but I can't make any promises about when I'll be able to make progress.

summer20100514, 2019-12-02:
@geky hopefully one day the performance issue will be solved. what I am thinking about is that more RAM consumption is acceptable compared to the unacceptable longlonglong latency