littlefs-project/littlefs #345: emudb example?

joel-felcana, 2019-12-04:
Hi,

I would like to test my implementation of a wrapper for littleFS without having to go through the ordeal of testing on the HW. I've seen `emubd` and how it's used in tests. I've tried to run those tests in Ubuntu over WSL (18.04, gcc 7.4, latest python 2 and 3), but `make test` fails. I've tried other commits but they fail too. This is the output in 2.1.4

```
./tests/test_format.sh
: invalid optionmat.sh: line 2: set: -
set: usage: set [-abefhkmnptuvxBCHP] [-o option-name] [--] [arg ...]
: invalid signal specification: trap: DEBUG
./tests/test_format.sh: line 5: $'\r': command not found
=== Formatting tests ===
./tests/test_format.sh: line 8: $'\r': command not found
--- Basic formatting ---
/usr/bin/ld: i386 architecture of input file `lfs_util.o' is incompatible with i386:x86-64 output
lfs.o: In function `lfs_dir_fetchmatch':
lfs.c:(.text+0x8be): undefined reference to `lfs_crc'
lfs.c:(.text+0x933): undefined reference to `lfs_crc'
lfs.c:(.text+0xaba): undefined reference to `lfs_crc'
lfs.o: In function `lfs_dir_commitprog':
lfs.c:(.text+0x1dee): undefined reference to `lfs_crc'
lfs.o: In function `lfs_dir_commitcrc':
lfs.c:(.text+0x2055): undefined reference to `lfs_crc'
lfs.o:lfs.c:(.text+0x219c): more undefined references to `lfs_crc' follow
collect2: error: ld returned 1 exit status
Makefile:72: recipe for target 'lfs' failed
make[1]: *** [lfs] Error 1
./tests/test_format.sh: line 13: $'\r': command not found
--- Basic mounting ---
test.c: In function ‘main’:
test.c:94:41: error: expected expression before ‘>’ token
     test_assert(lfs_format(&lfs, &cfg) => 0;
                                         ^
test.c:21:55: note: in definition of macro ‘test_assert’
         test_assert(__FILE__, __LINE__, #v " => " #e, v, e)
                                                       ^
test.c:21:9: error: too few arguments to function ‘test_assert’
         test_assert(__FILE__, __LINE__, #v " => " #e, v, e)
         ^
test.c:94:5: note: in expansion of macro ‘test_assert’
     test_assert(lfs_format(&lfs, &cfg) => 0;
     ^~~~~~~~~~~
test.c:10:13: note: declared here
 static void test_assert(const char *file, unsigned line,
             ^~~~~~~~~~~
Makefile:78: recipe for target 'test.o' failed
make[1]: *** [test.o] Error 1
./tests/test_format.sh: line 21: $'\r': command not found
--- Invalid superblocks ---
ln: failed to create symbolic link 'blocks/0'$'\r': No such file or directory
ln: failed to create symbolic link 'blocks/1'$'\r': No such file or directory
/usr/bin/ld: i386 architecture of input file `lfs_util.o' is incompatible with i386:x86-64 output
lfs.o: In function `lfs_dir_fetchmatch':
lfs.c:(.text+0x8be): undefined reference to `lfs_crc'
lfs.c:(.text+0x933): undefined reference to `lfs_crc'
lfs.c:(.text+0xaba): undefined reference to `lfs_crc'
lfs.o: In function `lfs_dir_commitprog':
lfs.c:(.text+0x1dee): undefined reference to `lfs_crc'
lfs.o: In function `lfs_dir_commitcrc':
lfs.c:(.text+0x2055): undefined reference to `lfs_crc'
lfs.o:lfs.c:(.text+0x219c): more undefined references to `lfs_crc' follow
collect2: error: ld returned 1 exit status
Makefile:72: recipe for target 'lfs' failed
make[1]: *** [lfs] Error 1
rm: cannot remove 'blocks/0': No such file or directory
rm: cannot remove 'blocks/1'$'\r': No such file or directory
./tests/test_format.sh: line 29: $'\r': command not found
--- Invalid mount ---
/usr/bin/ld: i386 architecture of input file `lfs_util.o' is incompatible with i386:x86-64 output
lfs.o: In function `lfs_dir_fetchmatch':
lfs.c:(.text+0x8be): undefined reference to `lfs_crc'
lfs.c:(.text+0x933): undefined reference to `lfs_crc'
lfs.c:(.text+0xaba): undefined reference to `lfs_crc'
lfs.o: In function `lfs_dir_commitprog':
lfs.c:(.text+0x1dee): undefined reference to `lfs_crc'
lfs.o: In function `lfs_dir_commitcrc':
lfs.c:(.text+0x2055): undefined reference to `lfs_crc'
lfs.o:lfs.c:(.text+0x219c): more undefined references to `lfs_crc' follow
collect2: error: ld returned 1 exit status
Makefile:72: recipe for target 'lfs' failed
make[1]: *** [lfs] Error 1
./tests/test_format.sh: line 34: $'\r': command not found
--- Expanding superblock ---
test.c: In function ‘main’:
test.c:94:41: error: expected expression before ‘>’ token
     test_assert(lfs_format(&lfs, &cfg) => 0;
                                         ^
test.c:21:55: note: in definition of macro ‘test_assert’
         test_assert(__FILE__, __LINE__, #v " => " #e, v, e)
                                                       ^
test.c:21:9: error: too few arguments to function ‘test_assert’
         test_assert(__FILE__, __LINE__, #v " => " #e, v, e)
         ^
test.c:94:5: note: in expansion of macro ‘test_assert’
     test_assert(lfs_format(&lfs, &cfg) => 0;
     ^~~~~~~~~~~
test.c:10:13: note: declared here
 static void test_assert(const char *file, unsigned line,
             ^~~~~~~~~~~
Makefile:78: recipe for target 'test.o' failed
make[1]: *** [test.o] Error 1
test.c: In function ‘main’:
test.c:94:40: error: expected expression before ‘>’ token
     test_assert(lfs_mount(&lfs, &cfg) => 0;
                                        ^
test.c:21:55: note: in definition of macro ‘test_assert’
         test_assert(__FILE__, __LINE__, #v " => " #e, v, e)
                                                       ^
test.c:21:9: error: too few arguments to function ‘test_assert’
         test_assert(__FILE__, __LINE__, #v " => " #e, v, e)
         ^
test.c:94:5: note: in expansion of macro ‘test_assert’
     test_assert(lfs_mount(&lfs, &cfg) => 0;
     ^~~~~~~~~~~
test.c:10:13: note: declared here
 static void test_assert(const char *file, unsigned line,
             ^~~~~~~~~~~
Makefile:78: recipe for target 'test.o' failed
make[1]: *** [test.o] Error 1
./tests/test_format.sh: line 50: $'\r': command not found
: No such file or directorye 51: scripts/results.py
Makefile:66: recipe for target 'test_format' failed
make: *** [test_format] Error 127
```

Is there a guide on how to use emubd, or is it intended to be used via the test system? If so, what are the requirements to run the tests?

Thanks!

geky, 2019-12-05:
Hello! I would highly suggest local-testing wherever possible, so glad to see you're looking into it.

Interesting, well, the tests shouldn't break, but I haven't ran them on WSL. The tests do use non-portable bash commands without restraint, so I can't say I'm that surprised.

I also have no idea why you're seeing this error:
```bash
i386 architecture of input file `lfs_util.o' is incompatible with i386:x86-64 output
```

One thing I'm looking at is reworking the testing infrastructure to run entirely in python, so that may have better chance of working in WSL, though it's not ready yet.

In the meantime, you should at least be able to use the emubd without the testing infrastructure. Something like this should work. emubd picks up the geometry to use from the lfs_config structure.
``` c
#include "lfs.h"                                           
#include "emubd/lfs_emubd.h"                               
                                                           
// lfs declarations                                        
lfs_t lfs;                                                 
lfs_emubd_t bd;                                            
                                                           
uint8_t read_buffer[512];                                  
                                                           
const struct lfs_config cfg = {                            
    .context = &bd,                                        
    .read  = &lfs_emubd_read,                              
    .prog  = &lfs_emubd_prog,                              
    .erase = &lfs_emubd_erase,                             
    .sync  = &lfs_emubd_sync,                              
                                                           
    .read_size      = 256,                                 
    .prog_size      = 256,                                 
    .block_size     = 4096,                                
    .block_count    = 2048,                                
    .block_cycles   = 100,                                 
    .cache_size     = 512,                                 
    .lookahead_size = 256,                                 
};                                                         
                                                                                                             
// entry point                                                         
int main(void) {                                                       
    int err = lfs_emubd_create(&cfg, "blocks");                                  
    if (err) {
        return err;
    }
                                                                       
    err = lfs_format(&lfs, &cfg);         
    if (err) {
        return err;
    }                                   
    err = lfs_mount(&lfs, &cfg);                          
    if (err) {
        return err;
    }                   
  
    // code here    

    err = lfs_unmount(&lfs);                                                  
    if (err) {
        return err;
    }
                                                                       
    lfs_emubd_destroy(&cfg);                                           
}                                                                       
```

joel-felcana, 2019-12-06:
Fantastic! I'll give it a spin and see how far I can get.

closed 2019-12-06 by joel-felcana

geky, 2020-01-27:
Heads up: I've been reworking the testing framework to move away from the shell scripts for testing. This may be more useful for you.

https://github.com/ARMmbed/littlefs/pull/372

At the very least the tests are now stored in regular toml files, so all you really need is a toml parser to get access to the code of the tests themselves.