littlefs-project/littlefs #346: Updating attributes on an open file

joel-felcana, 2019-12-04:
Hi,

Just a quick note as I have to run out of the office (will add more info tomorrow). What's the way to change the value of an attrib on a file that is open?

I tried with `lfs_setattr()` and it will show as that value while the file is open, but as soon as I close it, the attribute returns to the old value. On the other hand, modifying the file config struct works, but only if I have something to write to the file (a file sync doesn't modify the value if there is no data in the buffer).

So, it looks like:

1) if the file is closed, using lfs_setattr() works. If the file is open, the value doesn't get stored when the file is closed.
2) If the file is open, modifying the file config struct works, but only if I have something to write to the file.

Am I doing something wrong or is that the intended behaviour?

joel-felcana, 2019-12-05:
I researched the code a bit this morning.

First, about 2). `lfs.h` says that an attrib will be written in a file on sync or close as long as the file is in write mode. This is because `close` calls `sync`. I checked `lfs_file_sync()`, and it only writes to disk if the file is marked as dirty.
In the case of 2), obviously, writing values in the config struct doesn't label the file as dirty, so that's why it requires me to write at least 1B to the file for it to work. If I Iabel the file as dirty myself, syncing with 0B in the buffer work as expected (unless syncing with 0B in the buffer have other ramifications that I haven't seen).

I'll have lunch and carry on working on 1) :D

geky, 2019-12-05:
> obviously, writing values in the config struct doesn't label the file as dirty, so that's why it requires me to write at least 1B to the file for it to work. If I Iabel the file as dirty myself, syncing with 0B in the buffer work as expected

Oh hmm, that is interesting. And I'm not sure how we could fix this other than writing the file to disk even if not dirty...

I guess we could compare against the old value to see if it changed. Actually that may be a good fix.

---

For number 1, the issue is that lfs_setattr doesn't change the attribute loaded as a part of lfs_file_open. So lfs_setattr writes an attribute, and the lfs_file_sync overwrites the attribute with the old value.

It's come up before that this is confusing, but I'm not yet sure how to fix it. I could, as a part of lfs_setattr, march through open files and update attributes to keep them in sync (there is an internal linked-list). But this raises a similar question for lfs_file_truncate. In lfs_file_truncate's case, we don't touch other open copies of the same file, because a file opened multiple times acts as multiple snapshots of the file. I'm not sure if lfs_file_truncate's behavior should change, or if we should make the two behave inconsistently...

joel-felcana, 2019-12-06:
That explains it, thanks. I see that is not an easy thing to fix, but at least I have an understanding of what's going on.

For the time being, in my wrapper, I can check that if the file is open or not, and then use lfs_setattr or write to the cfg struct, and then mark as dirty and do a sync. That cover all cases, as long as syncing a file with no write data pending doesn't mess with the file somehow.

nickray, 2019-12-11:
Over in Rust land, we have both `sync_all` and `sync_data`: https://doc.rust-lang.org/std/fs/struct.File.html. Maybe your `sync` could stay like that `sync_data`, and you'd add a `sync_all` which writes even if the file is not dirty?

geky, 2019-12-18:
That's not a bad idea. Though it would be nice to avoid adding an API if possible.