littlefs-project/littlefs #347: Damage fs. Probably in case of power failure. Dumps

AVykhovanets, 2019-12-06:
Hi. I am using littlefs 1.7.2 along with Macronix MX25L3233F memory. We know about approximately 30 crashes littlefs  on 6,000 devices per month. I have some dumps from faulty devices. Damage is likely to occur at the time of power failure. On appliances with constant power, no damage was observed.

`File="dump_maketv1.1.bin"
NDEF: FsInit
NLFS: read=3,prog=0,erase=0,sync=0
ILFS: Dir="/"
ILFS: Dir="/flash"
ELFS: lfs_dir_fetch.494: Corrupted dir pair at 5 4
ILFS: Dir="/eeprom"
ELFS: lfs_dir_fetch.494: Corrupted dir pair at 7 6
ILFS: File="/boot_count"
ILFS: Dir="/bbox"
ELFS: lfs_dir_fetch.494: Corrupted dir pair at 10 9
NLFS: read=25,prog=0,erase=0,sync=0

File="dump_19021037.bin"
NDEF: FsInit
NLFS: read=3,prog=0,erase=0,sync=0
ILFS: Dir="/"
ILFS: Dir="/flash"
ILFS: File="/flash/update.bin"
ILFS: Dir="/eeprom"
ILFS: File="/eeprom/system.cfg"
ILFS: File="/boot_count"
ILFS: Dir="/bbox"
ILFS: File="/bbox/bbox0000.bin"
CLFS: lfs_ctz_find.1144: !LFS_ASSERT
ILFS: File="/bbox/bbox0001.bin"
ILFS: File="/bbox/bbox0002.bin"
ILFS: File="/bbox/bbox0003.bin"
ILFS: File="/bbox/bbox0004.bin"
CLFS: lfs_ctz_find.1144: !LFS_ASSERT
ILFS: File="/bbox/bbox0005.bin"
CLFS: lfs_ctz_find.1144: !LFS_ASSERT
ILFS: File="/bbox/bbox0006.bin"
ILFS: File="/bbox/bbox0007.bin"
CLFS: lfs_ctz_find.1144: !LFS_ASSERT
ILFS: File="/bbox/bbox0008.bin"
CLFS: lfs_ctz_find.1144: !LFS_ASSERT
ILFS: File="/bbox/bbox0009.bin"
ILFS: File="/bbox/bbox0010.bin"
CLFS: lfs_ctz_find.1144: !LFS_ASSERT
ILFS: File="/bbox/bbox0011.bin"
CLFS: lfs_ctz_find.1144: !LFS_ASSERT
ILFS: File="/bbox/bbox0012.bin"
ILFS: File="/bbox/bbox0013.bin"
ILFS: File="/bbox/bbox0014.bin"
ILFS: File="/bbox/bbox0015.bin"
CLFS: lfs_ctz_find.1144: !LFS_ASSERT
ILFS: File="/bbox/bbox0016.bin"
CLFS: lfs_ctz_find.1144: !LFS_ASSERT
ILFS: File="/bbox/bbox0017.bin"
CLFS: lfs_ctz_find.1144: !LFS_ASSERT
ILFS: File="/bbox/bbox0018.bin"
CLFS: lfs_ctz_find.1144: !LFS_ASSERT
ILFS: File="/bbox/bbox0019.bin"
CLFS: lfs_ctz_find.1144: !LFS_ASSERT
ILFS: File="/bbox/bbox0020.bin"
ILFS: File="/bbox/bbox0021.bin"
ILFS: File="/bbox/bbox0022.bin"
NLFS: read=343,prog=0,erase=0,sync=0

File="dump_19020422.bin"
NDEF: FsInit
NLFS: read=3,prog=0,erase=0,sync=0
ILFS: Dir="/"
ILFS: Dir="/flash"
ELFS: lfs_dir_fetch.494: Corrupted dir pair at 5 4
ILFS: Dir="/eeprom"
ELFS: lfs_dir_fetch.494: Corrupted dir pair at 7 6
ILFS: File="/boot_count"
ILFS: Dir="/bbox"
ILFS: File="/bbox/b`"
ELFS: lfs_dir_fetch.494: Corrupted dir pair at 426 425
ILFS: File="/bbox/bbox0011.bih"
ELFS: lfs_dir_fetch.494: Corrupted dir pair at 426 425
NLFS: read=120,prog=0,erase=0,sync=0

IDEF: File="ODB_II.bin"
NDEF: FsInit
NLFS: read=3,prog=0,erase=0,sync=0
ILFS: Dir="/"
ILFS: Dir="/flash"
ILFS: Dir="/eeprom"
ILFS: File="/eeprom/system.cfg"
ILFS: File="/eeprom/power.cfg"
ILFS: File="/eeprom/btconf.cfg"
ILFS: File="/eeprom/modem.cfg"
ILFS: File="/eeprom/gnss.cfg"
ILFS: File="/eeprom/accconf.cfg"
ILFS: File="/eeprom/status.cfg"
ILFS: File="/eeprom/odm.cfg"
ILFS: File="/eeprom/leds.cfg"
ILFS: File="/eeprom/rsconf.cfg"
ILFS: File="/eeprom/ioconf.cfg"
ILFS: File="/eeprom/setup.cfg"
ILFS: File="/eeprom/gtum.cfg"
ILFS: File="/boot_count"
ILFS: Dir="/bbox"
ELFS: lfs_dir_fetch.494: Corrupted dir pair at 31 19
NLFS: read=135,prog=0,erase=0,sync=0

Сhip tested. No errors detected
File="0034.bin"
NDEF: FsInit
NLFS: read=3,prog=0,erase=0,sync=0
ILFS: Dir="/"
ILFS: Dir="/flash"
ILFS: File="/flash/"
ILFS: File="/flash/bbox0000.bin"
ILFS: Dir="/eeprom"
ELFS: lfs_dir_fetch.494: Corrupted dir pair at 7 6
ILFS: Dir="/bbox"
ILFS: File="/bbox/"
ILFS: File="/bbox/bbox0000.bin"
NLFS: read=133,prog=0,erase=0,sync=0

Сhip tested. No errors detected
File="0030.bin"
NDEF: FsInit
NLFS: read=3,prog=0,erase=0,sync=0
ILFS: Dir="/"
ILFS: Dir="/flash"
ELFS: lfs_dir_fetch.494: Corrupted dir pair at 5 4
ILFS: Dir="/eeprom"
ELFS: lfs_dir_fetch.494: Corrupted dir pair at 7 6
ILFS: Dir="/bbox"
ELFS: lfs_dir_fetch.494: Corrupted dir pair at 10 9
NLFS: read=23,prog=0,erase=0,sync=0

Сhip tested. No errors detected
File="0020.bin"
NDEF: FsInit
NLFS: read=3,prog=0,erase=0,sync=0
ILFS: Dir="/"
ILFS: Dir="/flash"
ILFS: File="/flash/bbox0000.bin"
ILFS: Dir="/eeprom"
ELFS: lfs_dir_fetch.494: Corrupted dir pair at 7 6
ILFS: Dir="/bbox"
ILFS: File="/bbox/bbox0000.bin"
NLFS: read=38,prog=0,erase=0,sync=0

Сhip tested. No errors detected
File="0008.bin"
NDEF: FsInit
NLFS: read=3,prog=0,erase=0,sync=0
ILFS: Dir="/"
ILFS: Dir="/flash"
ELFS: lfs_dir_fetch.494: Corrupted dir pair at 5 4
ILFS: Dir="/eeprom"
ELFS: lfs_dir_fetch.494: Corrupted dir pair at 7 6
ILFS: Dir="/bbox"
ELFS: lfs_dir_fetch.494: Corrupted dir pair at 10 9
NLFS: read=23,prog=0,erase=0,sync=0

Сhip tested. No errors detected
IDEF: File="0006.bin"
NDEF: FsInit
NLFS: read=3,prog=0,erase=0,sync=0
ILFS: Dir="/"
ILFS: Dir="/flash"
ELFS: lfs_dir_fetch.494: Corrupted dir pair at 5 4
ILFS: Dir="/eeprom"
ELFS: lfs_dir_fetch.494: Corrupted dir pair at 7 6
ILFS: Dir="/bbox"
ELFS: lfs_dir_fetch.494: Corrupted dir pair at 30 10
NLFS: read=23,prog=0,erase=0,sync=0

Сhip tested. No errors detected
File="0004.bin"
NDEF: FsInit
NLFS: read=3,prog=0,erase=0,sync=0
ILFS: Dir="/"
ILFS: Dir="/flash"
ELFS: lfs_dir_fetch.494: Corrupted dir pair at 5 4
ILFS: Dir="/eeprom"
ELFS: lfs_dir_fetch.494: Corrupted dir pair at 7 6
ILFS: Dir="/bbox"
ELFS: lfs_dir_fetch.494: Corrupted dir pair at 10 9
NLFS: read=23,prog=0,erase=0,sync=0

Bonus. Damaged memory chip. Cannot read valid data.
File="0024.bin"
NDEF: FsInit
NLFS: read=3,prog=0,erase=0,sync=0
ILFS: Dir="/"
ILFS: Dir="/flash"
ILFS: Dir="/flash/flash"
ILFS: Dir="/flash/flash/flash"
ILFS: Dir="/flash/flash/flash/flash"
ILFS: Dir="/flash/flash/flash/flash/flash"
ILFS: Dir="/flash/flash/flash/flash/flash/flash"
ILFS: Dir="/flash/flash/flash/flash/flash/flash/flash"
ILFS: Dir="/flash/flash/flash/flash/flash/flash/flash/flash"
ILFS: Dir="/flash/flash/flash/flash/flash/flash/flash/flash/flash"
ILFS: Dir="/flash/flash/flash/flash/flash/flash/flash/flash/flash/flash"
ILFS: Dir="/flash/flash/flash/flash/flash/flash/flash/flash/flash/flash/flash"
ILFS: Dir="/flash/flash/flash/flash/flash/flash/flash/flash/flash/flash/flash/flash"
ILFS: Dir="/flash/flash/flash/flash/flash/flash/flash/flash/flash/flash/flash/flash/flash"
ILFS: Dir="/flash/flash/flash/flash/flash/flash/flash/flash/flash/flash/flash/flash/flash/flash"
ILFS: Dir="/flash/flash/flash/flash/flash/flash/flash/flash/flash/flash/flash/flash/flash/flash/flash"
...
OUT OF MEMORY`

`#define LFS_NAME_MAX 31`

`#define LFS_ASSERT(test) \
  do { \
    if (!(test)) \
    { \
        SYSLOG_C("%s.%d: !LFS_ASSERT", __func__, __LINE__); \
        return -1; \
    } \
  } while (0)`

`#define LFS_DEBUG(fmt, ...) \
    SYSLOG_D("%s.%d: " fmt , __func__, __LINE__, __VA_ARGS__)
#define LFS_WARN(fmt, ...) \
    SYSLOG_W("%s.%d: " fmt , __func__, __LINE__, __VA_ARGS__)
#define LFS_ERROR(fmt, ...) \
    SYSLOG_E("%s.%d: " fmt , __func__, __LINE__, __VA_ARGS__)`

`static const struct lfs_config cfg =
{
  .read_size = 256,
  .prog_size = 256,
  .block_size = 4096,
  .block_count = 1024,
  .lookahead = 128,
};`

`static int ReadAllFilesInDir(lfs_t *llfs, const char *path)
{
  lfs_dir_t dir;
  struct lfs_info info;

  check(path, ERRPARAM);
  check(strlen(path) > 0, ERRPARAM);
  SYSLOG_I("Dir=\"%s\"", path);
  if (lfs_dir_open(llfs, &dir, path) == 0)
  {
    while (lfs_dir_read(llfs, &dir, &info) > 0)
    {
      size_t full_size;
      char *full_path;

      full_size = strlen(path) + 1 + strlen(info.name) + 1;
      full_path = heap_malloc(full_size);
      check(full_path, ERRALLOC);
      if (path[strlen(path) - 1] == '/')
      {
        snprintf(full_path, full_size, "%s%s", path, info.name);
      }
      else
      {
        snprintf(full_path, full_size, "%s/%s", path, info.name);
      }

      if (info.type == LFS_TYPE_REG) // Файл
      {
        void *buff;

        SYSLOG_I("File=\"%s\"", full_path);
        buff = heap_malloc(info.size);

        if (buff)
        {
          lfs_file_t file; int ret;

          ret = lfs_file_open(llfs, &file, full_path, LFS_O_RDONLY);
          if (ret == LFS_ERR_OK)
          {
            lfs_file_read(llfs, &file, buff, info.size);
            lfs_file_close(llfs, &file);
          }
          heap_free(buff);
        }
      }
      if ((info.type == LFS_TYPE_DIR) && (info.name[0] != '.'))
      {
        ReadAllFilesInDir(llfs, full_path);
      }
      heap_free(full_path);
    }
    lfs_dir_close(llfs, &dir);
  }
  return 0;
ERRPARAM:
  return -1;
ERRALLOC:
  lfs_dir_close(llfs, &dir);
  return -1;
}

int fs_test_read(void)
{
  int err;
  lfs_t llfs;
  struct lfs_config lcfg;

  lcfg = cfg;
  wSyslogSetStr("LFS", "I");
  _lock(0);
  err = lfs_mount(&llfs, &lcfg);
  if (!err)
  {
    err = ReadAllFilesInDir(&llfs, "/");
    lfs_unmount(&llfs);
  }
  _unlock(0);
  return err;
}`

[dumps.zip](https://github.com/ARMmbed/littlefs/files/3931247/dumps.zip)

I hope this information is helpful.

davidefer, 2019-12-12:
Hi
just to understand for me.
Do you constantly write data in the FS files and then, when the power failure occurs, you get the problems - or you don't write data at all and you just loose the FS content?

AVykhovanets, 2020-01-10:
Hi. I regularly write to the directory /bbox. Recording occurs  blocks of a fixed size. The block size depends on the configuration. The maximum file size is 8kB, the maximum number of files in a directory /bbox is 64. After all the files are recorded, overwriting of the oldest one begins. Also in the process of working in blocks  are overwritten individual bytes - flags. The directory /eeprom is used to store settings. In fact, it is written only at the stage of product setup, and then only read. At each startup also overwritten the file /boot_count

darren1713, 2023-03-08:
Hi @AVykhovanets did you resolve this issue or find a solution to your littlefs corruption issue you were experiencing?