littlefs-project/littlefs #348: Repeatedly power on and repeatedly erase the A and B blocks of data.

Johnxjj, 2019-12-09:
I encountered such a situation, I will write a 4K file data when I power on, and then in the first 10 power-on, I will write the file in different blocks, but later power-on will be in A, B two Write files in blocks cyclically, which will cause the life of these two blocks to drop dramatically.
I am guessing that the CRC check value is duplicated.

geky, 2019-12-17:
If I understand the concern correctly, this is expected. littlefs doesn't aggressively move metadata-pairs as this creates a wear-amplification problem for CoW filesystems (all the writes gets pushed to the root, which gets exhausted quickly).

Instead, littlefs cycles two blocks for metadata-pairs until `lfs->cfg->block_cycles` as passed, and then the blocks are moved. You can set `lfs->cfg->block_cycles` to a small number if you want more aggressive wear-leveling at a performance cost.

Johnxjj, 2020-01-14:
hi @geky , sorry to reply so late.I don't mean blocks of metadata pairs, but to blocks of data that store files.

geky, 2020-01-28:
No worries, I'm also late. If the files are smaller than the cache_size then they will be stored inside the metadata pairs. I've been calling these "inline files".

littlefs will bounce these between two blocks, but will move them after lfs->cfg->block_cycles erases.

You can decrease lfs->cfg->block_cycles to have more aggressive wear-leveling, but this will create more writes since the tree needs to be updated every time a pair is relocated.

closed 2023-11-09 by Johnxjj