littlefs-project/littlefs #350: Standard read-only attribute?

nickray, 2019-12-11:
Would be a breaking change, but would it make sense to reserve some of the attribute types for future standard usage?
A first candidate would be setting a "read-only" flag on a file, so it can't accidentally be overwritten. Or a "can't read" flag, so a file can't accidentally be read. I guess this is global read/write _permissions_, which currently are given by default.

e107steved, 2019-12-12:
I agree - I've made a similar suggestion in the past.
It wouldn't necessarily break backward compatibility; I wouldn't expect littleFs to check attribute numbers at run-time; they would be a set of #defines (or an enum) which could be used or ignored (or even overridden). The main point is that if there is a defined range of reserved attributes, people know to steer clear of them.
File time/date is one candidate for a reserved attribute.
I'd call your suggestion "permissions". Rather than have littleFs implement a scheme which doesn't suit everyone (and which would probably take for ever given the length of @geky's TODO list and his available time), how about a simple 'check permissions' callback on file open? Then the applications programmer can implement a scheme appropriate to the use. Usernames/passwords could be thread-local, or passed via a programmer-defined extension of the lfs_file_config structure.

It is, of course, perfectly possible to implement your own permissions scheme totally outside littleFs, by intercepting file open calls. That could be more flexible.

nickray, 2019-12-12:
Thinking about it more, I'd definitely like to see read and write permissions as part of the core API (not in attributes), checked at run-time. So if a file does not have the write permission set, can only open read-only. If a file does not have the read permission set, can't even open. To circumvent, need to explicitly set the corresponding permission. Bonus: would fit in spare bytes between 104 and 128 :grin:. This might just be my liking, so for now I'll add this in my [Rust wrapper](https://github.com/nickray/littlefs2) via attribute 0.

Regarding attribute reservations: Yes that sounds best, "deprecate" use of a chosen range for a while, if necessary enforce it later. I do wonder if people in practice use the low or the high range or all over the place :thinking:

e107steved, 2019-12-16:
I like the idea of using spare bytes in the directory entry. Apart from any other considerations, it's one less memory read on file open. (And more of those bytes could be used for file time/date....). However I don't think it's as simple as a read flag and a write flag. Who can change those flags? An administrator? So then you're into multi-level permissions. And whatever was implemented into littleFs, a percentage of users would want something different! Hence the suggestion of a callback - if NULL, everything is allowed.  If implemented, it returns a set of flags indicating what is allowed.

The other point is that use of attributes is generic - the code is present, tested and universal. Using spare bytes in the directory would need new code - no idea how much - and would represent a code path that got limited use. There would also need to be an API which updated the directory flags.

For now, I'd be inclined to leave everything as-is until all the more obscure possible bugs have been looked at; then in the medium term maybe implement some directory-based variables (whose use could probably be made transparent to the application)

geky, 2019-12-18:
This isn't really documented, but there is actually enough space to have 9-bit ids for attributes. That could give you 8-bits of "user attributes" and 8-bits of "system attributes.

Though in hindsight I probably should have suggested staying in 7-bits to be more conservative. I'm not sure which is better.

> It is, of course, perfectly possible to implement your own permissions scheme totally outside littleFs, by intercepting file open calls. That could be more flexible.

> However I don't think it's as simple as a read flag and a write flag. Who can change those flags? An administrator? So then you're into multi-level permissions. And whatever was implemented into littleFs, a percentage of users would want something different!

This aligns with my thinking. There are too many system-level details that get involved when providing time/permissions/owner attributes, and the custom attribute system is an attempt to sidestep the issue.

This repo provides the core of littlefs that can be used directly on a baremetal system, but if you're integrating it into an RTOS/OS, you really should be wrapping up littlefs in your own layer so that it follows the semantics your system is expecting. Yeah this is pushing the problem somewhere else, but I think the RTOS/OS is a much better place to answer these system-level questions.

Common attributes that are _portable_ across the number of RTOS/OSs out there is not something I can coordinate myself. At least not right now.

Even then, it should be a hard rule that any standard attributes should be optional and ignorable for smaller devices that can't support them.

> I like the idea of using spare bytes in the directory entry. Apart from any other considerations, it's one less memory read on file open. (And more of those bytes could be used for file time/date....)

Custom attributes are already stored alongside the directory entry. They can even be loaded in sync with lfs_file_open by using the attrs field of the lfs_file_config struct.
https://github.com/ARMmbed/littlefs/blob/master/lfs.h#L280-L291

> I do wonder if people in practice use the low or the high range or all over the place 🤔

This would be a very interesting psychology experiment : )