littlefs-project/littlefs #352: [QUESTION] LFS v1.6 re-entrancy

henrygab, 2019-12-12:
At least some of these should be simple to answer.

### Q1 -- LFS not multi-thread safe?

Based on two prior comments by @geky ([Comment 1](https://github.com/ARMmbed/littlefs/issues/124#issuecomment-457442048_), [Comment 2](https://github.com/ARMmbed/littlefs/issues/12#issuecomment-359612016_)), it appears LFS is not safe to use concurrently from multiple threads.

**_Is this true of both v1.6 and v2.x?_**

### Q2 -- if no update/write operations.... (e.g., mount a volume read-only)

It was mentioned in another thread that multiple files could READ from a file, so long as there are no writers to that same file.

**_Is this true for either/both v1.6 and v2.x?_**

**_Are concurrent metadata operations to other files / directory entries allowed?_**

(i.e., if three threads are reading from /foo/bar/baz.txt, can a fourth thread concurrently add a file /foo/bar/omega.txt?)

### Q3 -- updates to files and/or metadata...

Is it correct to say that LFS is safe for multi-threaded use with the following constraints?

* LFS may only have a single one of the following operations occur at any time
  * any metadata update operation (e.g., file size, add file, remove file)
  * any operation that relies upon metadata (e.g., enumerate directory, open file)
* LFS may have any number of readers of an already-opened file, If-and-only-if LFS synchronizes metadata operations that would affect that file with the already-opened handle to the file

Is it correct to say that LFS does not have any synchronization between metadata operations and existing opened files?

### Q4 -- what is preferred / official recommendation...

Let's say LFS will be used in FreeRTOS, that multiple tasks (which we have no control over) will write files.  What function entry points need to be synchronized?  Is it safe to use a non-recursive mutex, or is it necessary to use a recursive mutex?

## Thanks!

Thanks for help understanding the restrictions and capabilities of LFS!

geky, 2019-12-18:
Hi @henrygab, hopefully these answer some questions.

None of the behavior around multithreading has changed v1 -> v2.

A quick note. littlefs is not by itself thread-safe, but it also doesn't use any threads. So all that's needed to make littlefs thread-safe is wrapping each function in a non-recursive mutex.

---

> Based on two prior comments by @geky (Comment 1, Comment 2), it appears LFS is not safe to use concurrently from multiple threads.

Not as is, no. If you're want to concurrently access the same filesystem (lfs_t struct) from two threads you will need to provide synchronization. This can be by wrapping all of the littlefs APIs with mutexes, or by provided mutexes at a higher-level depending on how you integrate littlefs into your system.

> It was mentioned in another thread that multiple files could READ from a file, so long as there are no writers to that same file.

Just like the other APIs, lfs_file_read is not thread safe by itself. It could modify both the lfs_t and lfs_file_t structs and needs external synchronization.

That being said, if you add a mutex that synchronizes every littlefs function, you can modify two open files simultaneously. The lfs_file_read calls may interleave depending on the timing of the threads, but this is no different than calling lfs_file_read on multiple open files in a single thread.

Now, when we open the _same_ file with multiple lfs_file_t structs, things get a bit complicated.

The way littlefs handles this is by pretending each file has its own isolated "copy" of the file. This works because under the hood, littlefs uses a copy-on-write (CoW) data structure. When a file is written two, this creates a CoW "fork" of the file, and the old file is not released while there are any references from the disk or other open files.

What this means is that there can me multiple readers of a file even if there are other writers. Open files contain a snapshot of the file as it was when lfs_file_open is called. Note this does have a storage cost and you may run out of disk space if you have two many open forks of old revisions.

You can even have multiple writers on the same file. However littlefs will just use the most recently synced file data without updating other files. This can creates semantic questions about when data/metadata in RAM should be updated (https://github.com/ARMmbed/littlefs/issues/346).

Should file writing actually update other open file handles to the same file? Not sure.

> Are concurrent metadata operations to other files / directory entries allowed?

Ah yes, this is fine. As long as the API is synchronized.

> Is it correct to say that LFS is safe for multi-threaded use with the following constraints?
>
>    LFS may only have a single one of the following operations occur at any time
>        any metadata [...]

If you synchronize the API of littlefs, the disk operations should also be synchronized. If you don't, you should not call the API of littlefs from different threads with the same lfs_t struct. This could corrupt the shared state and anything could happen.

> Let's say LFS will be used in FreeRTOS, that multiple tasks (which we have no control over) will write files. What function entry points need to be synchronized? Is it safe to use a non-recursive mutex, or is it necessary to use a recursive mutex?

All of the public littlefs API.

Because littlefs usually doesn't go back out and call wrappers of its public API, you shouldn't _need_ a recursive mutex, unless you are calling other littlefs functions from lfs_fs_traverse the block device callbacks, neither of which I would suggest.

---

Wrapping the API is tedious, depending on how much of the littlefs API you want to expose. Which is why it's been suggested to add a lfs_mutex hook similar to lfs_malloc to make it easier to integrate with existing RTOS codebases: https://github.com/ARMmbed/littlefs/pull/317

This is where recursive mutexes may be an intial requirements, because, internally, the littlefs functions do call each other, though this should probably be restructured.

henrygab, 2019-12-20:
I have so much to write, but succinctly:
Thank you, @geky , for the thorough answer.

> [CoW description]
> Now, when we open the same file with multiple lfs_file_t structs, things get a bit complicated.
…
> each file has its own isolated "copy" of the file … copy-on-write (CoW) data structure. … Open files contain a snapshot of the file as it was when lfs_file_open is called.

OK, this is VERY helpful.  It means that, if two threads open a file handle (each getting a distinct lfs_file_t), then neither one would ever see an update made by the other, unless it closes/re-opens the file.  That was non-obvious to me until your answer, but I think it's really important.

Yes, wrapping the LFS API is, indeed, more than a trivial change.  Is there a minimum C version that LFS is intended to compile under?  (C99?)  I didn't see it called out in the design or other readme's....

henrygab, 2019-12-27:
I'll open a new question on the minimum C version for the code...
Thanks for the great discussion and help understanding how LFS works!

closed 2019-12-27 by henrygab