littlefs-project/littlefs #355: lfs: littlefs/lfs.c:75: lfs_bd_read: Assertion `block < lfs->cfg->block_count' failed.

remfan77, 2019-12-19:
I'm Paolo and I'm testing how littlefs is strong on power failure.
My board is an ARM with Linux with eMMC.
I'm using fuse as wrapper of littlefs, so littlefs is seen as an usual linux filesystem.
Before begin the test I use the following commands to initialize partition.
```
dd if=/dev/zero of=/dev/mmcblk1p2
lfs /dev/mmcblk1p2 --format
```
This below is the script that is executed at each boot:
```
lfs /dev/mmcblk1p2 /data
while true
do
    for i in $(seq 1 20)
    do
        echo -n "$i " >> /data/$i
    done
done
```
I think all writes are serialized, each write is done when old one is finished.
When this test will pass, I will test adding a mutex of each littlefs function.

During these writes the system after 3 seconds is rebooted writing a bit in an ARM register.
The reboot is not clean and writes can be interrupted at any time. In this way I can test how littlefs is strong on power failure.
After some hours I can reproduce this bug
```
lfs: littlefs/lfs.c:75: lfs_bd_read: Assertion `block < lfs->cfg->block_count' failed.
```
The mount is done correctly.
I can see the files with **ls -l** command.
I can read files with **cat /data/N** command.
But if I append data to any files the fuse mounter is killed because **Assertion `block < lfs->cfg->block_count' failed**

I can do any test to help to undertand the problem.
I hope all this is not my error.
Best regards,
Paolo

[image_mmcblk1p2.zip](https://github.com/ARMmbed/littlefs/files/3982365/image_mmcblk1p2.zip)

remfan77, 2019-12-19:
I'm running the same test **without** reboot the ARM board.
The test continues running without problems.
I use fuse to mount littlefs and I was not sure 100% about serialization between fuse and littlefs.
So I can say that the curruption is due to to continuous restarts.

geky, 2019-12-19:
Hi @remfan77, thanks for opening an issue. This looks related to https://github.com/ARMmbed/littlefs/issues/295.

Currently I'm working on revamping the test framework to better test these cases and then will work on hunting them down.

remfan77, 2019-12-19:
After some investigation I'm trying a small patch.
The open syscall in linux environment need the **O_SYNC**.
With this option all calls are syncronous, that is the linux funtion write return **after** data are written.
Now I have two board running and they are running for 30 minutes.  It is already a success.
They will run for the night. Tomorrow I will write if the test continue running.

remfan77, 2019-12-20:
Hi @geky ,
the test worked all the night on two board and this morning continue to work corectly.
This seems to say that the linux eMMC driver in some way reorders writes, but it is only an idea that has to be verified. And it seems very strange to me. But the test seems to say this. I will continue to test. I will write more than one file ... I will keep you updated.

remfan77, 2019-12-20:
Hi @geky ,
I have a question.
I'm using an eMMC, that is a flash memory with FTL (Flash Translation Layer) inside.
For this type of flash we do not need wear-level.
In lfs_fuse.c there is
```// actual fuse functions
void lfs_fuse_defaults(struct lfs_config *config) {
    // default to 512 erase cycles, arbitrary value
    if (!config->block_cycles) {
        config->block_cycles = 512;
    }
```
I do not understand what this variable **config->block_cycles** mean.
Does it contain the max writes of a block ?
Using eMMC what is the right value to set ?
Thanks for support

e107steved, 2019-12-20:
Occasionally reading the docs gives better answers than reading the code!
If you're using LFS2, set block_cycles = -1 to disable wear levelling.
(Think it was value zero for LFS1)

remfan77, 2019-12-20:
Thanks @e107steved 
The version of littlefs included in littlefs-fuse states:
```
    // Number of erase cycles before we should move data to another block.
    // May be zero, in which case no block-level wear-leveling is performed.
    uint32_t block_cycles;
```

The version in branch master states:
```
    // Number of erase cycles before littlefs evicts metadata logs and moves.
    // the metadata to another block. Suggested values are in the
    // range 100-1000, with large values having better performance at the cost
    // of less consistent wear distribution
    //
    // Set to -1 to disable block-level wear-leveling.
    int32_t block_cycles;
```
I use the fuse version, so I should use 0 to disable wear-level.
For now I'm using standard value, that is 512. I will try also with 0 later.

remfan77, 2019-12-23:
Now for some days I cannot test. When I will return to work I will try using a real restart cutting power (not a software reset). If all will be OK I will close this Issue.

eastmoutain, 2019-12-31:
@remfan77 Though this assertion is caused by powerloss, the assertion location is same as [343](https://github.com/ARMmbed/littlefs/issues/343)

remfan77, 2020-01-10:
After 5000 cycle of powerOn / powerOff my ARM continues to run and filesystem is mounted.
I only added the O_SYNC flag to linux open syscalls of mmc device driver.

geky, 2020-01-27:
Hi @remfan77 thanks keeping this issue up to date. Sorry about the absence. I'm not sure the issue you are seeing is related to #343/#372 since I think you're right it's related to eMMC caching.

You're right eMMC caches block writes and this can be a problem if you don't explicitly sync the block device. Though I don't have experience with accessing eMMC on Linux.

I'm glad you were able to get it working. Did you add O_SYNC to the block device open call here?
https://github.com/ARMmbed/littlefs-fuse/blob/48cbae72a7dd02ca1c275917bd40ef27c9cc1819/lfs_fuse_bd.c#L26

Ahh, I thought hooking fsync would be sufficient ([here](https://github.com/ARMmbed/littlefs-fuse/blob/master/lfs_fuse_bd.c#L118)), but it looks like I was wrong since fsync only synchronizes with the OS unless you also have the O_SYNC flag?

If I understand how the O_SYNC flag works I think you're right it's needed.

---

> Using eMMC what is the right value to set ? is it 0 (because I do not need wear-level) ?

Yes, disabling wear leveling is the best option (0 for v2.0, -1 for v2.1). Having wear-leveling on isn't the end of the world, but it adds write cycles that aren't needed if the underlying block device wear levels.

> I do not understand well what this variable config->block_cycles mean.

This variable indicates how many erase cycles to allow on a metadata block before relocating the data in the block. Relocations enable wear-leveling, but involve extra writes to update the tree with their new position.

So fewer block cycles gives you more aggressive wear-leveling but at the cost of more overall writes. A value between 100-1000 seems to be a good tradeoff, though more benchmarking is needed here.

> If you're using LFS2, set block_cycles = -1 to disable wear leveling.

This changed in v2.1 since it was too easy for users to have wear-leveling disabled.

Currently littlefs-fuse is behind and needs to be updated.

remfan77, 2020-02-06:
Excuse @geky  I read only now.
I switched to an other project, so for now I cannot continue to test littlefs.
I added the O_SYNC at the device driver level of mmc (/dev/mmcblk0).
The O_SYNC tells that at the return of write syscall all data must be physically written. Maybe if the data is not written (cached), the controller *could* change the order of writes or something similar and this can damage the filesystem. With O_SYNC I tried thousands of power cut-off during writes without any problem. Without O_SYNC after some power cut-off I had a damaged filesystem. Thanks for your work.