littlefs-project/littlefs #357: copy paste example from readme

texrg, 2019-12-21:
and error

```
gcc -Os -I. -std=c99 -Wall -pedantic -Wextra -Wjump-misses-init -Wundef -Wno-missing-field-initializers  moje.c -o moje lfs_util.o lfs.o lfs.a 
moje.c:10:14: error: ‘user_provided_block_device_read’ undeclared here (not in a function)
     .read  = user_provided_block_device_read,
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
moje.c:11:14: error: ‘user_provided_block_device_prog’ undeclared here (not in a function)
     .prog  = user_provided_block_device_prog,
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
moje.c:12:14: error: ‘user_provided_block_device_erase’ undeclared here (not in a function)
     .erase = user_provided_block_device_erase,
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
moje.c:13:14: error: ‘user_provided_block_device_sync’ undeclared here (not in a function)
     .sync  = user_provided_block_device_sync,
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

```

remfan77, 2019-12-23:
Those function must be written for your block device.

visakh-png, 2020-01-06:
Please write your own read,write and erase function wrappers and map to the configuration file.