littlefs-project/littlefs #361: using LittleFs on W25N01GV 1Gb NAND Flash with FreeRTOS

visakh-png, 2020-01-03:
Hi ,

I am using Winbond QSPI Nand flash with W25N01GV 1 G-bit(128 MB) memory with only block erase(128 KB). Is there any size limitation for using higher size/Block-size in LittleFs?.

For my flash memory (with a page read/program and block erase interface) , does it requires a flash translation layer?. In my first look and up on looking the configuration structures I felt like LittleFs is handling these translation.

If it not requires the translation layer, Can I use it straight in my program which runs in FreeRTOS*?. With minimal mapping, i.e. only configuring "lfs_config".

If anybody have experience in porting the same and you have a how to doc. please share, if it is for STM and for FreeRTOS, really wonderful :-)  ....

Regards,
Visakh SV

visakh-png, 2020-01-03:
Hi,
I have ported littlefs to my application and written wrapper functions for it to attach to the config structure (lfs_config). This wrapper function is tested individually by writing a test application  that writes a 100byte buffer to all the 1024 blocks (each block 128 KB) start addresses(first column) and read back and compared. All is Success. 

But when I started to test with the example like in the littlefs readme.md mount error and format error has returned and finally hangs at fileopen(the next instruction). I can paste the test code here and config here, if someone have a clue please respond..... I expects a solution from @geky @geky 
// configuration of the filesystem is provided by this struct
const struct lfs_config cfg = {
    // block device operations
    .read  = QSPIFlashRead,
    .prog  = QSPIflashProgram,
    .erase = QSPIflashErase,
    .sync  = QSPIflashSync,

    // block device configuration
    .read_size 	= 16,
    .prog_size 	= 16,
    .block_size = 0x20000, //128KB
    .block_count = 0x400, //1024 blocks
    .lookahead_size = 0x800, // less idea , but given size of a page
	.cache_size = 16,
	.block_cycles = 500,
};

------------------------------Test code -------------------------

uint32_t FlashTest(void) {
    // mount the filesystem

	uint32_t err = lfs_mount(&lfs, &cfg);

    // reformat if we can't mount the filesystem
    // this should only happen on the first boot
    if (err) {
    	err= lfs_format(&lfs, &cfg);
    	{
    		CPrintf("ERROR:  Format  \n\r");
    	}
        lfs_mount(&lfs, &cfg);
        {
				CPrintf("ERROR:  Mount  \n\r");
	    }
    }
    // read current count
    uint32_t boot_count = 0;
    err= lfs_file_open(&lfs, &file, "boot_count", LFS_O_RDWR | LFS_O_CREAT);
    if(err)
    {
   		CPrintf("ERROR:  File Open  \n\r");
   	}
    err= lfs_file_read(&lfs, &file, &boot_count, sizeof(boot_count));
    if(err)
	{
		CPrintf("ERROR:  File Read  \n\r");
	}

    // update boot count
    boot_count += 1;
    err= lfs_file_rewind(&lfs, &file);
    if(err)
	{
		CPrintf("ERROR:  File Rewind  \n\r");
	}

    err= lfs_file_write(&lfs, &file, &boot_count, sizeof(boot_count));

    if(err)
	{
		CPrintf("ERROR:  File Write  \n\r");
	}

    // remember the storage is not updated until the file is closed successfully
    err= lfs_file_close(&lfs, &file);

    if(err)
	{
		CPrintf("ERROR:  File Close  \n\r");
	}

    // release any resources we were using
    lfs_unmount(&lfs);

    // print the boot count
    CPrintf("boot_count: %d\n", boot_count);

}

eastmoutain, 2020-01-07:
But when I started to test with the example like in the littlefs readme.md mount error and format error has returned and finally hangs at fileopen(the next instruction).

what's the returned error code? 

I suggest setting the read/prog/cache buffer size 128/256

visakh-png, 2020-01-07:
@eastmoutain . Hi thank you for the reply. My issues were solved, after correcting configuration and mapping correct FreeRTOS calls. I am able to write and read back using littleFs API's now. However I couldn't make it work with static buffers. I am using FreeRTOS port alloc functions for allocating read/write sizes.

mershaywu77, 2020-01-13:
@visakh-png 
Thank you for your initial investigation. I have the same plan to evaluate whether the littleFS can be used in my product which is a ARM Cortext A9 + FreeRTOS + eMMC, what is you feeling? Is is feasible /stable for now or it requires considerable effort to use it on a industry product?

gmomchilov, 2020-03-15:
> @eastmoutain . Hi thank you for the reply. My issues were solved, after correcting configuration and mapping correct FreeRTOS calls. I am able to write and read back using littleFs API's now. [...]

@visakh-png could you, please, share your configuration? I am having similar issues as you.

visakh-png, 2020-05-04:
@mershaywu77  Sorry for replying late.  I have found issues for opening a large files(>300K). So i have stopped it a while and using my own memory management. But it will be very easier , if it is in place of memory management with hardcore addresses. If i check it some time i will post it here, the progress
..

mershaywu77, 2020-05-04:
@visakh-png , thank you for reponse. I tried it on ZC702 evk and compared it to several commercial version of fail-safe file systems and a safeFAT, the littleFS has the least performance. It may not meet my expections for the fail safe file system.

visakh-png, 2020-05-04:
@mershaywu77 ohhh. That is something bad news. Do you only feel a performance glitch or any threat to the fail-safe behavior like that I have faced. Do you have any open source to suggest supports eMMC NAND flash.

mershaywu77, 2020-05-04:
@visakh-png, no issue found for sail safe part, only cons is the read/write performance is as good as commercial one. Sorry I did not found any open source one which can have the miliar performance as commercial solutions.

visakh-png, 2020-05-04:
@mershaywu77 . Okay. Thanks for the information. :-)

visakh-png, 2020-05-04:
@gmomchilov , may I know some  details of the system you are using.

gmomchilov, 2020-05-04:
@visakh-png I am also using FreeRTOS and W25N01GV. I was wondering what your LittleFS lfs_config variable looks like.

I am having issues using the ECC of the chip with LittleFS and thought I could have misconfigured LittleFS in some way.

visakh-png, 2020-05-04:
@gmomchilov
cfg.context= hqspi;
	cfg.read  = QSPIFlashRead;
	cfg.prog  = QSPIflashProgram;
	cfg.erase = QSPIflashErase;
	cfg.sync  = QSPIflashSync;

	 // block device configuration
	cfg.read_size 		= W25N01GV_PAGE_SIZE;
	cfg.prog_size 		= W25N01GV_PAGE_SIZE;
	cfg.block_size 		= W25N01GV_PAGE_SIZE;//*W25N01_PAGE_BLOCK_COUNT;
	cfg.block_count 	= W25N01GV_PAGE_COUNT;
	cfg.lookahead_size 	= W25N01GV_PAGE_SIZE;
	cfg.cache_size 		= W25N01GV_PAGE_SIZE;
	cfg.block_cycles 	= BLOCK_CYCLES_TO_READ;

where page size is 2K....

visakh-png, 2020-05-04:
@gmomchilov gmomchilov . ECC Error not making issues in bare write and with Little FS, i think for me

gmomchilov, 2020-05-04:
@visakh-png thank you. 

Can you explain why your block size is equal to the page size? Don't you run into the problem that LittleFS erases the whole block (128K) deleting all of your data in your block even though you wanted to earse 2K only? 

And why do you use the lookahead size to 2K?

visakh-png, 2020-05-05:
@gmomchilov Hi I have observed that W25N01GV only erases 2KB from the page address. So if I am giving 128KB (0x20000) as the block size, then file system will assume that on a single erase 128KB(64 pages) will be erased, but in actual only 2KB get erase and it will result in a write failure of your trial after first 2KB from the page address.

ByteGalaxy, 2020-05-28:
@visakh-png : If you are using freeRTOS, you should also map malloc and free functions provided by freeRTOS in lfs_utils.h. This might fix your problem.

visakh-png, 2020-05-29:
@ByteGalaxy  I am using portalloc and vportfree of CMSIS Free RTOS instead of malloc and free. This is not current issue. For large files littlefs is not useful

Visit-Learn, 2020-07-15:
@visakh-png 

Could I know how do you choose BLOCK_CYCLES_TO_READ value on W25N01GV?
Thank you so much.

Igor-Misic, 2020-11-28:
> @gmomchilov Hi I have observed that W25N01GV only erases 2KB from the page address. So if I am giving 128KB (0x20000) as the block size, then file system will assume that on a single erase 128KB(64 [...]

You were probably checking at the wrong address if all pages are erased. I am also using W25N01GV and the block erase command deletes 128KB (all 64 pages).

jimarcher, 2021-03-21:
Hi all, I'm trying to use LittleFS with the W25N01Gbit like you ll are, and having various problems. Have any of you been able to resolve these issues? If not, which commercial file system did you go to? I appreciate any suggestions, I need to get a product out the door and this has been a substantial blocker. Thank you.

mershaywu77, 2021-03-21:
@jimarcher , we finally go to HCC safeFAT, hope it can help.

Igor-Misic, 2021-03-21:
@jimarcher it works for me. Maybe here you will find some helpful informations https://github.com/littlefs-project/littlefs/issues/11#issuecomment-785851320

jimarcher, 2021-03-22:
Thanks @Igor-Misic I'll check that out.

jimarcher, 2021-03-23:
> @jimarcher , we finally go to HCC safeFAT, hope it can help.

@mershaywu77   Thank you, if I can't make Little FS work I'll check out safeFAT, but it does seem I should be able to make it work.

simon88, 2024-09-09:
Hi all,
I'm trying to implement littlefs with an `W25N01G`but I'm stuck with format operation. Is there anyone who has succesfully configure w25 with littlefs ?

eastmoutain, 2024-09-09:
Sorry to hear that, I haven’t tried it on w25.