littlefs-project/littlefs #366: Infinite loop in lfs_fs_traverse after lfs_dir_read of sub directory

ArkEngr, 2020-01-14:
I created a user interface that let me make and read directories and make and read files on a PC.  However when reading a sub directory with one file, I got in an infinite loop in the lfs_fs_traverse function when it was called from lfs_fs_size after the lfs_dir_read of the sub directory path.  This was found to be caused by  lfs_t structure element f->next equal to the current f  which caused the for loop in lfs_fs_traverse:
    // iterate over any open files
    for (lfs_file_t *f = (lfs_file_t*)lfs->mlist; f; f = f->next) {
to never exit.
there seems to be a couple of f = f->next  in for loops that may have the same problem.

My source code:
              case 'l' :  /* fl [<path>] - Directory listing */
                    while(*ptr == ' ')
                    {
                      ptr++;
                    }

                    printf("ptr size %d\n", strlen(ptr));

                    res = lfs_dir_open(&LFSfs, &Dir, ptr);

                    if(res)
                    {
                      put_rc(res);
                      break;
                    }

                    p11 = s1 = s2 = 0;

                    for(;;)
                    {
                      res = lfs_dir_read(&LFSfs, &Dir, &Finfo);

                      if((res < LFS_ERR_OK) || !Finfo.name[0])
                      {
                        break;
                      }

                      if(Finfo.type == LFS_TYPE_DIR)
                      {
                        s2++;
                      }
                      else
                      {
                        s1++;
                        p11 += Finfo.size;
                      }

                      xprintf("%c %15,llu  %-12s  %s\n",
                                (Finfo.type & LFS_TYPE_DIR) ? 'D' : '-',
                                Finfo.size, Finfo.name,
                                "");
                    }

                    printf("%4u File(s),%15llu bytes total\n%4u Dir(s)", s1, p11, s2);

                    ss1 = lfs_fs_size(&LFSfs);  // infinite loop when path is subdirectory
                    if(ss1 >= 0)
                    {
                      printf(", %15d blocks used\n", ss1);
                    }
                    else
                    {
                      put_rc(ss1);
                    }
                    break;

geky, 2020-01-27:
Interesting, thanks for reporting this.

It shouldn't happen but may be a bug relating to closing files/dirs? Do you call lfs_file/dir_close for every  open call?

ArkEngr, 2020-01-29:
If you close the open for, it does not happen.

rickymohk, 2025-05-12:
Opening the same file twice without closing first will cause this infinite loop. Should the second open return error instead?

geky, 2025-05-13:
> Should the second open return error instead?

It's debatable! Some users would like the second open to error or assert, others would prefer to not pay the code cost for something that should never happen if the API is being used correctly.

Note there _is_ an assert in this case ([here](https://github.com/littlefs-project/littlefs/blob/8ed63b27be79ab59ee1cd15a950ddd64e7a602f7/lfs.c#L6117)), but it is common to disable asserts to save code.

It's a bit of a hack, but if you want an error, you can define LFS_ASSERT like so:

```
-D'LFS_ASSERT(test)=do {if(!(test)) {return -1;}} while(0)'
```

This is currently tested in CI ([here](https://github.com/littlefs-project/littlefs/blob/8ed63b27be79ab59ee1cd15a950ddd64e7a602f7/.github/workflows/test.yml#L215)), and was introduced ([here](https://github.com/littlefs-project/littlefs/pull/514)). Support for this will not change without a major version bump.

---

Eventually I would like to split LFS_ASSERT into LFS_ASSERT and LFS_USER_ASSERT/LFS_DISK_ASSERT macros, but it would require a breaking change and is low priority.

iXia9, 2025-05-15:
Unfortunately, I have encountered the same phenomenon several times. Five years have passed, and I'm very disappointed that this problem still hasn't been solved.
This should be classified as a bug, as you can't expect a flash chip to always be correct.
I added a little simple code in my local device to terminate this infinite loop, as follows:
  for (struct lfs_mlist *d = lfs->mlist; d; d = d->next) {
    if(d->next==d) {
      d->next=0;
    }
    ...

  for (lfs_file_t *f = (lfs_file_t*)lfs->mlist; f; f = f->next) {
    if(f->next==f) {
      f->next=0;      
    }
    ...
    
This method is simple and effective. Hope this can offer a little help.