littlefs-project/littlefs #367: lfs error:977: Corrupted dir pair at 0 0

fengchun1, 2020-01-17:
const struct lfs_config cfg = 
{
    // block device operations
    .read  = _block_read,
    .prog  = _block_prog,
    .erase = _block_erase,
    .sync  = _block_sync,

    // block device configuration
    .read_size = 256,
    .prog_size = 256,
    .block_size = 32768,
    .block_count = 4,
    .cache_size = 256,
    .lookahead_size = 16,
    .block_cycles = 500,
};
i use stm32f767 Internal flash  test litterfs，when lfs_file_open(&lfs, &file, "boot_count", LFS_O_RDWR | LFS_O_CREAT);   happen lfs error:977: Corrupted dir pair at 0 0，why？

fengchun1, 2020-01-17:
debug ：
The following failed: File:D:TEST flash +litterfs\ Src\lfs.c Line:2745
Expression failed: file->flags LFS_F_OPENED

e107steved, 2020-01-17:
The stm32f767 only has four 32k programmable flash sectors, and these are in the zone where the program is usually loaded. Most of the sectors are 256K in size. Have you configured your boot options etc to make sure nothing else is trying to use this flash memory area?
Also, littleFS needs a minimum number of blocks to work; I'm not certain that four is enough.

fengchun1, 2020-01-17:
thank you  i test it

fengchun1, 2020-01-17:
I have a question， How  does the litterFS to find the address，i use HAL liabrary，please  you check 
my read， prog and erase function：
uint16_t STMFLASH_ReadHalfWord ( uint32_t faddr )
{
   return *(__IO uint16_t*)faddr; 
}

void STMFLASH_Write_NoCheck ( uint32_t WriteAddr, uint16_t * pBuffer, uint16_t NumToWrite )   
{                                           
        uint16_t i;        
         
        for(i=0;i<NumToWrite;i++)
        {
                HAL_FLASH_Program(FLASH_TYPEPROGRAM_HALFWORD,WriteAddr,pBuffer[i]);
          WriteAddr+=2;                                    //地址增加2.
        }  
}

void STMFLASH_Read ( uint32_t ReadAddr, uint16_t *pBuffer, uint16_t NumToRead )           
{
        uint16_t i;
         
        for(i=0;i<NumToRead;i++)
        {
                pBuffer[i]=STMFLASH_ReadHalfWord(ReadAddr);
                ReadAddr+=2;       
        }
}

void STMFLASH_Erase(uint32_t addr)

  {
    
  FLASH_EraseInitTypeDef flash_erase;
    uint32_t flash_error;
   
    
  HAL_FLASH_Unlock();
   
   
  flash_erase.TypeErase = FLASH_TYPEERASE_SECTORS;  
    
  flash_erase.Sector = flash_sector_number(addr);   
   
  flash_erase.NbSectors = 1;  
  
  flash_erase.VoltageRange = FLASH_VOLTAGE_RANGE_3; 
  
  __HAL_FLASH_CLEAR_FLAG(FLASH_FLAG_EOP | FLASH_FLAG_OPERR | FLASH_FLAG_WRPERR |FLASH_FLAG_PGAERR | FLASH_FLAG_PGPERR
   | FLASH_FLAG_ERSERR | FLASH_FLAG_BSY);
   
   
   HAL_FLASHEx_Erase(&flash_erase, &flash_error);
   
   
   FLASH_WaitForLastOperation(50000);
   
   HAL_FLASH_Lock();

}

e107steved, 2020-01-20:
littleFs doesn't know about addresses in flash. Read the interface specs for the read, write, program, and sync routines (in lfs.h); you need to write your own routines which translate littleFs requests with a block number, offset and size into accesses of physical addresses.

fengchun1, 2020-02-28:
block number offset and size need base address to accesses of physical addresses，i didn't find

geky, 2020-02-28:
I think you need this:
```
WriteAddr = block * block_size + off + start_of_flash
```

block address is in block_size units (32768), while off and size are in bytes.

> Also, littleFS needs a minimum number of blocks to work; I'm not certain that four is enough.

4 blocks _should_ work. Though only with files that fit in RAM (file_size <= cache_size).

That being said it isn't the most tested configuration.

fengchun1, 2020-03-02:
Hi  @geky 
Thank you for the detailed reply. It is really useful.
My program prints an error（lfs_error:977: Corrupted dir pair at 0 0 ） when it executes to open file. I do n’t know if there is something wrong with my configuration or something else. Can you refer me? What error might it cause？
Again, Thank you for your time!