littlefs-project/littlefs #369: Metadata Revision Value Incredibly Large

rmollway, 2020-01-20:
As part of an effort to analyze write frequency / understand the efficiency of the directory layout, I extracted the revision from each metadata block. 

While some of them were reasonable, I noticed that there are quite a few that are much larger than I would expect. A common value for these is 0xBE5CC5EF . I have confirmed that this is indeed the value that exists in the Flash chip. It is worth noting that the number of metadata blocks with a value in this range varies from device to device.

When I erase external flash and extract again, the values make sense and increase over time as I would expect. 

I am confident that the files have not been written over 3 billion times (and therefore certainly not the metadata block). Do you have any idea what might be happening here?

geky, 2020-01-28:
Hi @rmollway, excellent observation. But fortunately this isn't unexpected.

Metadata in littlefs is stored in pairs of blocks. The reason is that we need the spare block to be able to erase without losing our metadata if we lose power. By being able to store metadata in both blocks in a metadata-pair, we can build fully atomic operations.

The purpose of the revision count is to tell use which metadata block in the pair is the most recent. But that doesn't necessarily mean the revision count needs to start at zero.

When we create a metadata-pair (lfs_mkdir for example), we need to set both block's revision counts so that the newest metadata has the most recent revision count. We could erase both blocks and write out metadata twice. _Or_, as an optimization, we can assume one block is "old" and read the uninitialized contents of disk to use as the "revision count". This optimization seems minor, needing 1 erase instead of 2, but on some flash parts erases can be very expensive.

This is the reason you see different revision counts based on what was previously on disk.

This does mean our revision count can overflow quite early. But we would still have needed to worry about overflow in a long-running system even if we start at 0. We can handle this with [serial arithmetic](https://en.wikipedia.org/wiki/Serial_number_arithmetic).

The reason you see the counter start at zero after an erase is a funny coincidence. An erase sets all of flash to 0xffffffff, and 0xffffffff+1 happens to be 0.

---

It's also worth noting the revision count plays a role in wear-leveling. But here we don't need a starts-at-zero counter either.

The value of block_cycles determines _roughly_ how many erase cycles before we need to relocate a given metadata-pair. We determine if these cycles have passed by simply taking the modulus of the revision count (`dir->rev % lfs->cfg->block_cycles`).

This means the number of erase cycles before relocation isn't exact, but it doesn't need to be.

---

Hopefully that helps explain a bit what's going on.

Unfortunately littlefs doesn't have a mechanism for getting _exactly_ how many erase cycles has passed on a directory.

Your best bet would be to erase the disk, or maybe store the initial revision count somewhere with something like `lfs_setattr(&lfs, "/dir", MY_ATTR_INITIAL_REV, &(uint32_t){dir->rev}, sizeof(uint32_t))`.

thrasher8390, 2020-01-28:
Great explanation.

In terms of wear-leveling for metadata pairs. Once `dir->rev % lfs->cfg->block_cycles` is true and the pairs are moved. Is the revision re-initialized to some random value or does it continue to increment by 1?

geky, 2020-01-28:
Relocation only moves one block in the pair (this way the old be can be picked up if we lose power). So we increment the old counter for our new block.

```
     oldpair    newpair
     .-' '-.    .-' '-.
    v        v v       v
.-------. .-------. .-------.
| rev 4 | | rev 5 | | rev 6 |
|       | |       | |       |
|       | |       | |       |
'-------' '-------' '-------'
```

Hmm, looking at it now, I'm wondering if that can end up in a cycle where we only ever move one block. Will need to investigate.

thrasher8390, 2020-01-28:
Oh yea, interesting point. It seems like if `block_cycles` was set to be odd then we wouldn't have to worry about it. But if `block_cycles` is even there the case you brought up could definitely be a concern.

geky, 2020-01-29:
The actual math is `dir->rev % (lfs->cfg->block_cycles+1)` to workaround block_cycles=1 (if block_cycles=1 then littlefs would _always_ be relocating). So it should only be a problem if block_cycles is odd, which fortunately should be less common.
https://github.com/ARMmbed/littlefs/blob/ce2c01f098f4d2b9479de5a796c3bb531f1fe14c/lfs.c#L1487

geky, 2020-02-09:
Ok, I've pushed up a fix and test here https://github.com/ARMmbed/littlefs/pull/372/commits/fe957de892edbefd7af54acbe19e9a4e0f0fb1d0 as a part of pr https://github.com/ARMmbed/littlefs/pull/372

This should fix the aliases issue and catch it if it happens again.

This test now checks that there are no obvious outliers with significant wear. It's not full-proof but should be a good sanity check.
https://github.com/ARMmbed/littlefs/blob/fe957de892edbefd7af54acbe19e9a4e0f0fb1d0/tests/test_exhaustion.toml#L433-L443