littlefs-project/littlefs #371: Help with LFS_ERR_CORRUPT, LFS_ERR_INVAL errors in test program

octoknot, 2020-01-26:
Hi littlefs, I'm getting errors when I try run a couple of test programs, including a "boot_count" test program like the one in the README, and I'm hoping someone can help.

First, here's my configuration:

- 8MB of flash, Macronix MX25R6435F
- (chip is a Silicon Labs EFR32)
- .erase_size is 4kb [flash supports 32k and 64k, but we want to use 4k]
- .prog_size is 256 bytes
- .read_size is 256 bytes
- .block_count = (8*1024*1024)/(4*1024) = 2048
- .block_cycles = 1000
- .cache_size = 2048
- .lookahead_size = 256

I can successfully perform these steps for a "boot_count" test program:

- In a separate test program, I erase the flash. Then for littlefs test boot_count...
- lfs_mount fails as expected
- I successfully format the flash with lfs_format
- I successfully re-mount with lfs_mount
- I can open the boot_count file (LFS_O_RDWR | LFS_O_CREAT)
- I update the boot_count variable to 1
- I rewind the file, write to the file, sync, and close
- I try to re-open the file, but it fails with LFS_ERR_CORRUPT

Next, for grins and giggles, I tried a different test program, one where I wrote a simple test struct to flash:

typedef struct {
    int x;
    int y;
    int z;
} read_write_test_t;

I performed the following steps:

- Again, in a separate test program, I erase the flash. Then...
- lfs_mount fails as expected
- I format the flash with lfs_format
- I re-mount with lfs_mount
- I open my test file, "configdata" (LFS_O_RDWR | LFS_O_CREAT)
- I populate my read_write_test_t variable with some values
- I write my read_write_test_t value to the file
- ** note: in this test program I forgot to call lfs_file_rewind, but this might be relevant **
- I sync and close
- I re-open the file, and I do *NOT* get an error (unlike boot_count)
- I re-read my read_write_test_t test data
- re-read is successful, the values match
- I close the file

That is great that it gets this far. But, when I try to re-run the exact same test, it fails at the lfs_mount step with the error LFS_ERR_INVAL. (And no, I do not erase the flash when I re-run that test.)

So, I am hoping someone might know what could be wrong.

Full disclosure -- I've also been working with spiffs and have had problems there, too, so that tells me something very basic is wrong, but I can't figure it out. I *can* run a test program where I erase/write/re-read the flash directly through our flash driver, and that works.

Any help is appreciated! I am new to firmware development and have not worked with flash before, so no suggestion is too simple. For example, at the flash driver level, I *do* check the Write In Progress bit before returning to the caller, but I did not at first, so asking me if I had would have been reasonable!

geky, 2020-01-28:
Hmm, that is strange. Your configuration looks correct.

Is it possible you are trying to read the part at a SPI frequency too fast for either the EFR32 or the MX25R? Sometimes the flash parts can support frequencies past the range of the MCUs themselves.

There's no real lower-bound for SPI, so I would try lowering the frequency and seeing if you get better reads.

It's also possible that capacitance on the SPI bus or neighboring noise could be corrupting the data you're sending. Unfortunately SPI flash doesn't have a built-in checksum for the protocol, so it can't tell if a transaction failed due to noise.

littlefs, however, does have checksums. So if you have LFS_DEBUG enabled and you see a number of "Bad block" messages, that may indicate that you have a bus issue.

octoknot, 2020-01-31:
Hi geky, thanks for replying.

I tried changing some of the SPI params, such as frequency, and unfortunately it didn't help my problem. I was hopeful that it would: On another project, we had issues talking to an SD card over SPI, and it turned out to be a SPI configuration issue.

I'll mention your suggestion of noise or capacitance to the other engineers on my project. That "SD card" project had another problem that only occurred when a particular sensor was enabled, and it turned out to be interference. In the meantime, we're going to try a home-grown logging solution, since the files we need to log are well defined and have specific sizes, so there won't be any surprises.

But, it's concerning that I had problems with both spiffs and littlefs. It makes me wonder if there's a lower level problem that's going to bite us down the road. If so, I will keep your suggestions in mind.

Thanks again.