littlefs-project/littlefs #375: lfs_ctz_extend: Assertion `head >= 2 && head <= lfs->cfg->block_count' failed.

fsulima, 2020-02-04:
There is the following assertion failure under massive write test:
lfs_tester: src/lfs.c:2263: lfs_ctz_extend: Assertion `head >= 2 && head <= lfs->cfg->block_count' failed.
Program received signal SIGABRT, Aborted.
0x000000000041dee7 in raise ()
(gdb) bt
#0  0x000000000041dee7 in raise ()
#1  0x000000000041e071 in abort ()
#2  0x0000000000411ddc in __assert_fail_base ()
#3  0x0000000000411e52 in __assert_fail ()
#4  0x0000000000406366 in lfs_ctz_extend (lfs=0x6cbc00 <_lfs>, 
    pcache=0x6cbea0 <_file_obj_pool+384>, rcache=0x6cbc00 <_lfs>, head=0, size=4096, 
    block=0x6cbe94 <_file_obj_pool+372>, off=0x6cbe98 <_file_obj_pool+376>) at src/lfs.c:2263
#5  0x0000000000407bba in lfs_file_write (lfs=0x6cbc00 <_lfs>, 
    file=0x6cbe58 <_file_obj_pool+312>, buffer=0x7fffffffe210, size=200) at src/lfs.c:2899
#6  0x000000000040e3c0 in _fs_write (pfs_context=0x6c9100 <hb_fs_context>, 
    pfile=0x6cbe58 <_file_obj_pool+312>, pbuf=0x7fffffffe210 "\314;", size=200, 
    pbw=0x7fffffffe120) at src/hb_storage.c:1179
#7  0x000000000040d69f in hb_stor_write_file (phandle=0x7fffffffe1a0, 
    pbuf=0x7fffffffe210 "\314;", size=200, pbw=0x7fffffffe120) at src/hb_storage.c:410
#8  0x000000000040dbd6 in hb_stor_write_buf (phandle=0x7fffffffe1a0, pbuf=0x7fffffffe210, 
    bytes=200) at src/hb_storage.c:612
#9  0x000000000040fac5 in hb_data_log_save_record (pLog=0x6cb6b0 <_test_logs+112>, 
    rec_id=7654, src_buf=0x7fffffffe210) at src/Core/hb_data_log.c:777
#10 0x000000000040feba in hb_data_log_add_record (pLog=0x6cb6b0 <_test_logs+112>, 
    psrc_buf=0x7fffffffe210) at src/Core/hb_data_log.c:866
#11 0x000000000040cd10 in test_data_log_huge_records () at src/data_log_test.c:632
#12 0x000000000040b524 in run_test_case (
    pTestCaseName=0x49f910 "DataLog at big amount of records", 
    pTestCaseFunc=0x40caee <test_data_log_huge_records>) at src/data_log_test.c:75
#13 0x000000000040b625 in run_data_log_unit_test () at src/data_log_test.c:137
#14 0x0000000000400b97 in main () at src/lfs_tester.c:13
(gdb) frame 4
#4  0x0000000000406366 in lfs_ctz_extend (lfs=0x6cbc00 <_lfs>, 
    pcache=0x6cbea0 <_file_obj_pool+384>, rcache=0x6cbc00 <_lfs>, head=0, size=4096, 
    block=0x6cbe94 <_file_obj_pool+372>, off=0x6cbe98 <_file_obj_pool+376>) at src/lfs.c:2263
2263	                LFS_ASSERT(head >= 2 && head <= lfs->cfg->block_count);
(gdb) p i
$1 = 4
(gdb) p skips
$2 = 6
(gdb) p head
$3 = 0
(gdb) 

There is a latest (2.1.4) version of lfs in use. There is a standalone synthetic test on top of emubd to reproduce the situation, the full source is attached:
[lfs_tester_standalone.tar.gz](https://github.com/ARMmbed/littlefs/files/4155602/lfs_tester_standalone.tar.gz)
It needs '$ make' under linux to be built and then '$ gdb ./lfs_tester' to run.
Please help to pin down the problem.

geky, 2020-02-05:
Hi @fsulima, thanks for the reproducible test case! I haven't been able to look at it yet, but it looks like this issue may have been fixed in this branch?
https://github.com/ARMmbed/littlefs/pull/372

fsulima, 2020-02-06:
It is indeed! Impatiently waiting for it to be released!

pjsg, 2020-02-17:
My issue (which has the same symptom) is not fixed in this branch. I've merged my test code into the test-revamp branch -- https://github.com/pjsg/littlefs/tree/afl-fuzzing-revamp -- and there are two different crashes (different line numbers, but the same message) caused by the trigger files in aflresults/. The FUZZING.md explains how to build and run the test cases.