littlefs-project/littlefs #379: Failure due to writing 0x70 into location with value 0x10

pjsg, 2020-02-18:
I made the lfs_rambd.c behave more like a flash -- i.e. initialize with the erase_value and verify, before programming, that the program operation only clears bits. 

My current branch is https://github.com/pjsg/littlefs/tree/afl-fuzzing-revamp -- to exhibit the problem, check out this branch and run the following commands

```
make afl/test
afl/test print < aflresults/prog
```

This should demonstrate the problem and allow debugging.

geky, 2020-02-18:
Interesting! Thanks for creating an issue, I will need to look into this.

Out of curiousity, what's the advantage of this check vs the current check that the page being written to must contain the erase value (1s in flash's case)? It seems the check that all progs are against the "erase value" is a tighter restriction.

pjsg, 2020-02-19:
It is a tighter restriction -- but that check is only performed if the erase value != -1. I was uncertain whether the code actually used single bit writing to zero -- so I implemented that. I doubt that it makes any difference.

I now understand (I think) what is going on in the simplest test case.

```
  open(2, "2file2.xxxxxx", 0x503) -> 0
  Setting prog abort after 1 bytes, bitor = 112
  open(1, "3file3.xxxxxxxx", 0x503)
Powerfail during write of 48 bytes at offset 0x70 in block 1. Wrote 1 bytes.
Byte at offset 0x71 should have been 0x1f, but wrote 0x7f instead.
 prog operation was aborted
powerfail
  mount -> 0
  open(2, "2file2.xxxxxx", 0x502) -> 0
  close(2)
Trying to program 0x70 into location with value 0x10 [at offset 0x70 (in block 1) in a length of 16]
test: bd/lfs_rambd.c:115: lfs_rambd_prog: Assertion `(current & new_value) == new_value' failed.
Aborted
```

The contents of the disk are:
```
000000 01 00 00 00 f0 0f ff f7 6c 69 74 74 6c 65 66 73
000010 2f e0 00 10 00 00 02 00 00 02 00 00 00 04 00 00
000020 ff 00 00 00 ff ff ff 7f fe 03 00 00 70 1f fc 08
000030 98 4e 09 8d ff ff ff ff ff ff ff ff ff ff ff ff
000040 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
*
000200 02 00 00 00 f0 0f ff f7 6c 69 74 74 6c 65 66 73
000210 2f e0 00 10 00 00 02 00 00 02 00 00 00 04 00 00
000220 ff 00 00 00 ff ff ff 7f fe 03 00 00 70 1f fc 08
000230 fc 7b db cb ff ff ff ff ff ff ff ff ff ff ff ff
000240 10 1f f8 10 40 00 00 0d 32 66 69 6c 65 32 2e 78
000250 78 78 78 78 78 20 00 00 0d 70 1f f8 13 f4 86 29
000260 02 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
000270 10 7f ff ff ff ff ff ff ff ff ff ff ff ff ff ff
000280 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
*
080000
```

The readtree (I added printing of the tag offset) output is:

```
$ scripts/readtree.py -a /tmp/littlefs-disk 256
superblock "littlefs" v2.0
  block_size 512
  block_count 1024
  name_max 255
  file_max 2147483647
  attr_max 1022
dir "/"
mdir {0x0, 0x1} rev 1
  0008: tag 0ff00008 superblock id 0 size 8 name "littlefs"
  0014: tag 20100018 inlinestruct id 0 size 24 inline size 24
  0030: tag 500ffc10 crc 0x0 size 16
  0044: tag aff003ef struct 0xff id 0 size 1007
$ scripts/readtree.py -l /tmp/littlefs-disk 256
superblock "littlefs" v2.0
  block_size 512
  block_count 1024
  name_max 255
  file_max 2147483647
  attr_max 1022
dir "/"
mdir {0x0, 0x1} rev 1
  0008: tag 0ff00008 superblock id 0 size 8 name "littlefs"
  0014: tag 20100018 inlinestruct id 0 size 24 inline size 24
  0030: tag 500ffc10 crc 0x0 size 16
$
```

It looks as though there is one corrupt tag which only occurs in block 1. It seems that when the file is created after the powerfail/mount, the free pointer is not set to the right place and it wants to overwrite the partially written tag. It may be complicated by the fact that block 0 has no invalid tags, and block 1 does have the invalid tag.

pjsg, 2020-02-20:
I've updated the code so that it only will overwrite the erase_value. This simplifies the check somewhat.

pjsg, 2020-02-20:
The code (I think) goes off the rails here (around line 800 in lfs.c):

```
            // next commit not yet programmed or we're not in valid range
            if (!lfs_tag_isvalid(tag) ||
                    off + lfs_tag_dsize(tag) > lfs->cfg->block_size) {
                dir->erased = (lfs_tag_type1(ptag) == LFS_TYPE_CRC &&
                        dir->off % lfs->cfg->prog_size == 0);
                break;
            }
```

The tag is invalid, and the dir->off is set to point to the invalid tag. Setting dir->erased to false does seem to cure the problem.

pjsg, 2020-02-23:
There are other problems which I understand less. There appears to be trouble around creating a file, removing that file, and then writing a lot of stuff to it. 

-- Have done a bunch of file operations....
```
  mount{r432,p0,e0} -> 0
  open(3, "1file1.xxxx", 0x503){r1680,p160,e1} -> 0
Hardware watchpoint 2: *(int *) (bd.buffer + 512 * 111)

Old value = 7
New value = -1
```
We are in the process of erasing block 111
Disk looks like
```
dir "/"
mdir {0x0, 0x1} rev 33
| 0008: tag 0ff00008 superblock id 0 size 8 name "littlefs"
| 0014: tag 20100018 inlinestruct id 0 size 24 inline size 24
| 0030: tag 00100409 reg id 1 size 9 name "0file0.xx"
| 003d: tag 20100400 inlinestruct id 1 size 0 inline size 0
| 0041: tag 00100c0d reg id 3 size 13 name "2file2.xxxxxx"
| 0052: tag 20200c08 ctzstruct id 3 size 8 ctz {0x6e} size 2098
| 005e: tag 0010080b reg id 2 size 11 name "1file1.xxxx"
| 006d: tag 20100800 inlinestruct id 2 size 0 inline size 0
| 0071: tag 601ffc08 hardtail size 8 tail {0x6f, 0x70}
| 007d: tag 7ffffc0c movestate size 12 0x000800001e0000001e000000
| 008d: tag 500ffc13 crc 0x0 size 19
v 00a4: tag aff003ec struct 0xff id 0 size 1004
mdir {0x70, 0x6f} rev 8
  0008: tag 00100013 reg id 0 size 19 name "5file5.xxxxxxxxxxxx"
  001f: tag 20200008 ctzstruct id 0 size 8 ctz {0x146} size 142
  002b: tag 7ffffc0c movestate size 12 0x000800001e0000001e000000
  003b: tag 500ffc05 crc 0x0 size 5
  0044: tag aff003fa struct 0xff id 0 size 1018
---
(gdb) c
Continuing.
{d46} [^ 1159 us]  open(1, "4file4.xxxxxxxxxx", 0x503){r880,p96,e1}[^ 98792081 us] -> 0
{d47} [^ 1273 us]  remove("5file5.xxxxxxxxxxxx"){r816,p64,e1} -> 0
{d48} [^ 1189 us]  remove("4file4.xxxxxxxxxx"){r2560,p128,e1} -> 0
{d49} [^ 1221 us]  write(1, , 2007){r2480,p1984,e4} -> 2007
{d50} [^ 1189 us]  write(1, , 2007){r2096,p2048,e4} -> 2007
{d51} [^ 1156 us]  write(1, , 2007){r2112,p2048,e4} -> 2007
{d52} [^ 1173 us]  write(1, , 2007){r2096,p2048,e4} -> 2007
{d53} [^ 1152 us]  write(1, , 2007){r2064,p1984,e4} -> 2007
```
At this point, lfs_alloc returns block 111 to handle data for fd 3
```
#0  lfs_alloc (lfs=0x610420, block=0x7fffffffcd28) at lfs.c:461
#1  0x0000000000407f36 in lfs_ctz_extend (lfs=0x610420, pcache=0x7fffffffe000, rcache=0x610420, head=105, size=2032, block=0x7fffffffdff4, off=0x7fffffffdff8)
    at lfs.c:2191
#2  0x0000000000409a68 in lfs_file_write (lfs=0x610420, file=0x7fffffffdfb8, buffer=0x7fffffffd610, size=2047) at lfs.c:2912
```
DIsk looks like
```
mdir {0x1, 0x0} rev 34
  0008: tag 0ff00008 superblock id 0 size 8 name "littlefs"
  0014: tag 20100018 inlinestruct id 0 size 24 inline size 24
  0030: tag 00100409 reg id 1 size 9 name "0file0.xx"
  003d: tag 20100400 inlinestruct id 1 size 0 inline size 0
  0041: tag 00100c0d reg id 3 size 13 name "2file2.xxxxxx"
  0052: tag 20200c08 ctzstruct id 3 size 8 ctz {0x6e} size 2098
  005e: tag 0010080b reg id 2 size 11 name "1file1.xxxx"
  006d: tag 20100800 inlinestruct id 2 size 0 inline size 0
  0071: tag 500ffc0f crc 0x0 size 15
  0084: tag aff003f0 struct 0xff id 0 size 1008
```
It appears that the pair at 111, 112 has been compacted away. This seems right as there weren't any files in it.
```
Hardware watchpoint 2: *(int *) (bd.buffer + 512 * 111)

Old value = -1
New value = 11
0x00000038a9089754 in memcpy () from /lib64/libc.so.6
(gdb) c
Continuing.
{d55} [^ 1306 us]  finally close(1){r384,p128,e1}[^ 3627687 us] -> 0
```
At this point fd 1 is closed, and this causes a write to block 111 -- which is bad, becuase that block is about to be used to flush the cache for fd 3 when it is closed. Note that block 111 was the block which held the file tag when the file was opened.
```
  finally close(3)
Trying to program 0x69 into location with value 0x0b [at offset 0x0 (in block 111) in a length of 32]
test: bd/lfs_rambd.c:133: lfs_rambd_prog: Assertion `current == bd->cfg->erase_value' failed.
```
When fd 3 is closed, then the assert fires as the block is not ready to be written into.

I'm not sure what the right approach is to this. You could disallow renaming/deleting a file if it is open -- but that seems overly heavy handed. I'm assuming that I ought to be able to open a file, delete it and then continue to read/write to it. This implies that any of its blocks have to remain allocated until the file is closed and vanishes. I wonder what happens if there are two reader/writers to the same file and the file gets deleted. It seems that the file ought to only vanish (and the blocks get freed) when the last reference to the file goes away.

pjsg, 2020-02-23:
It turns out that preventing deletes and renames on open files does *not* make the crashes go away. 

```
format/mount{r576,p128,e2} -> 0
  open(1, "3file3.xxxxxxxx", 0x503){r464,p80,e1} -> 0
{d1}   write(1, , 2007){r2272,p1984,e4} -> 2007
{d2}   write(1, , 2007){r2096,p2048,e4} -> 2007
{d3}   write(1, , 2007){r2112,p2048,e4} -> 2007
{d4}   write(1, , 2007){r2096,p2048,e4} -> 2007
{d5}   write(1, , 2007){r2064,p1984,e4} -> 2007
{d6}   write(1, , 2007){r2096,p2048,e4} -> 2007
{d7}   write(1, , 2007){r2112,p2048,e4} -> 2007
{d8}   write(1, , 2007){r2096,p2048,e4} -> 2007
{d9}   write(1, , 2007){r2144,p2048,e4} -> 2007
{d10}   write(1, , 2007){r2096,p2048,e4} -> 2007
{d11}   write(1, , 2007){r2112,p2048,e4} -> 2007
{d12}   write(1, , 2007){r2032,p1984,e4} -> 2007
{d13}   write(1, , 79){r128,p128,e0} -> 79
{d14}   write(1, , 2007){r2128,p2048,e4} -> 2007
{d15}   write(1, , 158){r160,p128,e1} -> 158
{d16}   write(1, , 126){r128,p128,e0} -> 126
{d17}   write(1, , 2007){r2112,p2048,e4} -> 2007
{d18}   write(1, , 2007){r2096,p2048,e4} -> 2007
{d19}   write(1, , 2007){r2160,p2048,e4} -> 2007
{d20}   write(1, , 2007){r2096,p2048,e4} -> 2007
{d21}   write(1, , 203){r192,p192,e0} -> 203
{d22}   write(1, , 2007){r2112,p2048,e4} -> 2007
{d23}   write(1, , 2007){r2032,p1984,e4} -> 2007
{d24}   close(1){r640,p160,e1} -> 0
{d25}   open(1, "2file2.xxxxxx", 0x2){r160,p0,e0} -> -2
  mount{r368,p0,e0} -> 0
  open(1, "3file3.xxxxxxxx", 0x503){r144,p0,e0} -> 0
  open(0, "3file3.xxxxxxxx", 0x3){r192,p0,e0} -> 0
  write(0, , 63){r896,p0,e1} -> 63
{d26}   sync(1){r560,p80,e1} -> 0
{d27}   write(1, , 2007){r2000,p1984,e4} -> 2007
{d28}   write(1, , 2007){r2096,p2048,e4} -> 2007
{d29}   write(1, , 2007){r2112,p2048,e4} -> 2007
{d30}   write(1, , 2007){r2096,p2048,e4} -> 2007
{d31}   write(1, , 2007){r2064,p1984,e4} -> 2007
{d32}   write(1, , 2007){r2096,p2048,e4} -> 2007
test: lfs.c:2171: lfs_ctz_find: Assertion `head >= 2 && head <= lfs->cfg->block_count' failed.
{d33}   finally close(0)Aborted
```

geky, 2020-02-24:
Hi, sorry about the delayed responses. I'm still working through a couple more known bugs on #372 and will be able to spend more time on this afterwards. (Though as far as I'm aware the remaining bugs all require rename to occur).

Looking at the initial issue, this looks like a bug fixed on the [test-revamp](https://github.com/ARMmbed/littlefs/tree/test-revamp) branch in https://github.com/ARMmbed/littlefs/commit/517d3414c5e04eedb07be2e58107c1f96b8b8684 ([this line](https://github.com/ARMmbed/littlefs/compare/aab6aa0ed939303d7788e21bb547eb0a386636fb...02c84ac5f4da1a2fdb7aa15cc11afcddca2f3ea8#diff-630e6e312bb168dc6903ffc8496ddb9cL804-R815) specifically, there were a number of fixes in that commit). Just wanting to check that the afl-fuzzing branch is up to date with all of the fixes on test-revamp.

pjsg, 2020-02-25:
I've updated to the latest version of test-revamp. Many of the tests now fail with LFS_ERR_CORRUPT -- but the root cause still exists. The simplest case is in my branch in the file `aflresults/nodel3`. This can be run as follows (note that you do *not* need to install afl to do this -- your standard dev environment is fine). 

```
make afl/test
afl/test -p < aflresults/nodel3
```

pjsg, 2020-04-01:
I've now backmerged master and there are two tests that exhibit different problems:

```
make test_corrupt
make test_realistic_powerfail
```

These both fail as of https://github.com/pjsg/littlefs/tree/corrupt-real-pf 

`test_corrupt` just does normal operations and inserts a powerfail between two operations.
`test_realistic_powerfail` simulates a powerfail while programming which corrupts the entire area which is being written.

pjsg, 2020-04-06:
@geky This change to lfs.c solves many of the overwriting problems. It just checks to make sure that if a tag is not valid, then it is also not programmed before it assumes that the rest of the block is erased.

```
diff --git a/lfs.c b/lfs.c
index 97134cc..25f9291 100644
--- a/lfs.c
+++ b/lfs.c
@@ -290,6 +290,10 @@ static inline bool lfs_tag_isvalid(lfs_tag_t tag) {
     return !(tag & 0x80000000);
 }
 
+static inline bool lfs_tag_notprogrammed(lfs_tag_t tag) {
+    return !~tag;
+}
+
 static inline bool lfs_tag_isdelete(lfs_tag_t tag) {
     return ((int32_t)(tag << 22) >> 22) == -1;
 }
@@ -837,7 +841,7 @@ static lfs_stag_t lfs_dir_fetchmatch(lfs_t *lfs,
             // next commit not yet programmed or we're not in valid range
             if (!lfs_tag_isvalid(tag)) {
                 dir->erased = (lfs_tag_type1(ptag) == LFS_TYPE_CRC &&
-                        dir->off % lfs->cfg->prog_size == 0);
+                        dir->off % lfs->cfg->prog_size == 0) && lfs_tag_notprogrammed(tag);
                 break;
             } else if (off + lfs_tag_dsize(tag) > lfs->cfg->block_size) {
                 dir->erased = false;
```

geky, 2020-04-06:
Hi @pjsg, sorry about late responses, currently working on catching up.

The fix you proposed assumes the erase value of storage is 0xffs, which littlefs currently doesn't assume (this is useful for implemented encryption in the block device layer).

And while this offers better protection, I believe littlefs also needs to check the data attached to the tag, as littlefs stored metadata in tag+data pairs.

I just posted in https://github.com/ARMmbed/littlefs/issues/245#issuecomment-609929709 what I think is the correct approach, let me know if it looks reasonable to you.

geky, 2020-04-06:
Also a side-note about the changes to testbd. The setjmp/longjmp addition and bit-level failures look quite useful, but I wasn't sure about the use of mmap.

What advantage over the filebd does mmap offer? It's possible to have a disk that doesn't fit in the available memory, which could cause problems.

Reducing the number of block device implementations isn't necessarily the goal here, since these also offer limited examples users can build off of. As is rambd can be adopted on an MCU which may not have the mmap API. I'd rather see this available as a "mmapbd" implementation.

Also I believe you need to call `msync` for the mmap changes to be reflected in the underlying file. This can be added to the block device's sync function
http://man7.org/linux/man-pages/man2/msync.2.html

pjsg, 2020-04-06:
You can certainly mmap more data than will fit in memory, but it must fit within the address space. 

I take your point about underlying encrypted file systems. 

I also wasn't certain what the roles of the various block device drivers was. I can certainly setup an mmapbd and use that from the testbd by default. The big advantage is that you can see the intermediate state of the disk from another terminal session while using the debugger. Yes, I'll the msync calls as appropriate (they aren't needed on (at least) Linux).

geky, 2020-04-06:
> The big advantage is that you can see the intermediate state of the disk from another terminal session while using the debugger.

Though isn't this available through the filebd? (available with the --persist flag in the test framework, though at a performance cost). 

> Yes, I'll the msync calls as appropriate (they aren't needed on (at least) Linux).

Oh interesting, I only noticed because I wasn't seeing the disk updates also in Linux. I wonder if this is due to a difference in Linux's filesystem configuration?

pjsg, 2020-04-08:
If the underlying flash is encrypted, then it would be much more difficult to tell if the end of a block is erased or not. In particular the tag_isvalid method doesn't seem to work as it tests a bit of encrypted erased memory -- which might have any value. 

On the plus side, I've added an (optional) simple encrypting layer into the test harness -- it currently uses XOR with a key that it reads from the control file. If the key is all zeros, then it is effectively just a straight passthrough.