littlefs-project/littlefs #385: Get pointer to lfs_file_t type from the pathname.

baranovus, 2020-02-26:
Hello All.
Is there a way to get a pointer to "lfs_file_t" structure for a file by the pathname?
The reason I want it:
I keep a list of all open files in a dedicated array of lfs_file structures. When  "lfs_remove(&lfs, pathname);" is called I want to verify that the file with "pathname" is not open i.e there is no correspondent lfs_file in my list.
As far as I could see, Little FS deletes open files without error messaging.
Thanks.

geky, 2020-02-26:
Hi @baranovus, I don't think this is possible with littlefs currently.

One option is wrapping lfs_file_open and lfs_file_close and check the state of the open file list on both.

littlefs does let you delete open files without an error. However it does internally mark the open files such that they won't be written out on close. I don't know if this does what you're looking for though.
https://github.com/ARMmbed/littlefs/blob/master/lfs.c#L1827-L1829

baranovus, 2020-02-26:
Thank. I will try to keep the pathname hash codes in my list of open files.