littlefs-project/littlefs #386: lfs_error:977: Corrupted dir pair at 0 0

fengchun1, 2020-02-28:
My program aborted at （ lfs_file_read  LFS_ASSERT(file->flags & LFS_F_OPENED) ), printf  lfs_error:977: Corrupted dir pair at 0 0 , please help me  tnanks

geky, 2020-02-28:
That assert means you are calling lfs_file_read without lfs_file_open. 

Could it be that lfs_file_open errored?

fengchun1, 2020-02-29:
Thanks in advance for your reply
I copy （https://github.com/ARMmbed/littlefs  README.md) exmple 

   int main(void) {
    // mount the filesystem
    int err = lfs_mount(&lfs, &cfg);

    // reformat if we can't mount the filesystem
    // this should only happen on the first boot
    if (err) {
        lfs_format(&lfs, &cfg);
        lfs_mount(&lfs, &cfg);
    }

    // read current count
    uint32_t boot_count = 0;
    lfs_file_open(&lfs, &file, "boot_count", LFS_O_RDWR | LFS_O_CREAT);
    lfs_file_read(&lfs, &file, &boot_count, sizeof(boot_count));

    // update boot count
    boot_count += 1;
    lfs_file_rewind(&lfs, &file);
    lfs_file_write(&lfs, &file, &boot_count, sizeof(boot_count));

    // remember the storage is not updated until the file is closed successfully
    lfs_file_close(&lfs, &file);

    // release any resources we were using
    lfs_unmount(&lfs);

    // print the boot count
    printf("boot_count: %d\n", boot_count);
}