littlefs-project/littlefs #389: Cache size and prog size

baranovus, 2020-03-04:
I apologize for possibly asking the question, which has been somewhere answered.
Is it necessary to make cache size and prog size values equal?
I tried to make the cache size == 4096 and prog size == 256 and while executing "lfs_file_close"  after "lfs_file_write" I've got some unrelated memory locations corrupted.
It did not happen while both had been equal to 4096.

baranovus, 2020-03-23:
Hello.
I set: cache size 4096;
         prog_size 256;
         read_size 256;
I enter lfs_mount(&lfs, &cfg) and step to: 
// zero to avoid information leaks
    lfs_cache_zero(lfs, &lfs->rcache);
    lfs_cache_zero(lfs, &lfs->pcache);
While stepping I see in my IAR debug watch window valid values for lfs cfg (e.g. valid callback functions for eeprom operations).
As I step into "**lfs_cache_zero**" I see that all my variables are replaced by 0xff and after that in function "**static int lfs_bd_read**"
after call "**int err = lfs->cfg->read(lfs->cfg, rcache->block,    rcache->off, rcache->buffer, rcache->size)**;"
I get to hard fault.

geky, 2020-03-25:
Oh interesting. That sounds like a buffer-overflow issue. Are you allocating the buffers yourself?

It's a bit confusing, but `prog_buffer` and `read_buffer` should both == `cache_size` (the buffer is used for the cache). `prog_size` and `read_size` are only used to inform littlefs what the disk geometry is.

Let us know if that works.

baranovus, 2020-03-25:
Well, if the sizes of cache, read and prog are equal everything is working fine

geky, 2020-03-26:
The size of prog_buffer, read_buffer and cache_size must all be equal. read_size and prog_size are unrelated. I realize this is not the most intuitive.

For example:
``` c
uint8_t read_buffer[256];
uint8_t prog_buffer[256];
struct lfs_config {
    .read_size = 1,
    .prog_size = 1,
    .cache_size = 256,
    .block_size = 4096,
    .read_buffer = read_buffer,
    .prog_buffer = prog_buffer,
}
```