littlefs-project/littlefs #393: I used the STM32F103RCT6 external flash test successfully, but the internal flash will give an error when opening the file. What is the reason?

fengchun1, 2020-03-12:
dear all，
I used the external flash test successfully, but the internal flash will give an error when opening the file. What is the reason?
This is my configuration code.
int _block_read(const struct lfs_config *c, lfs_block_t block, lfs_off_t off, void *buffer, lfs_size_t size)
{
      STMFLASH_Read ( block * c->block_size + off, (uint32_t *)buffer, size );  
    //flash_read_block(block * c->block_size + off, (kal_uint8 *)buffer, (kal_uint32)size);
    return 0;
}
 int _block_prog(const struct lfs_config *c, lfs_block_t block, lfs_off_t off, const void *buffer, lfs_size_t size)
{
   
   STMFLASH_Write_NoCheck ( block * c->block_size + off, (uint32_t * )buffer, size );
    //flash_write_block(block * c->block_size + off, (kal_uint8 *)buffer, (kal_uint32)size);
    return 0;
}
 int _block_erase(const struct lfs_config *c, lfs_block_t block)
{
      STMFLASH_Erase(block * c->block_size);
    //flash_erase_block(block * c->block_size);
    return 0;
}
 int _block_sync(const struct lfs_config *c)
{
    return 0;
}
const struct lfs_config cfg = 
{
   /*block device operations */
    .read  = _block_read,
    .prog  = _block_prog,
    .erase = _block_erase,
    .sync  = _block_sync,

   /*block device configuration */
    .read_size = 256,
    .prog_size = 256,
    .block_size = 2048,
    .block_count =16,
    .cache_size = 256,
    .lookahead_size = 16,

   .lookahead_buffer = lfs_lookahead_buf,
};

uint32_t STMFLASH_ReadWord ( uint32_t faddr )
{
  faddr += STM32_FLASH_BASE ;
  return *(__IO uint32_t*)faddr; 
}

void STMFLASH_Write_NoCheck ( uint32_t WriteAddr, uint32_t * pBuffer, uint32_t NumToWrite )   
{                                           
   uint32_t i;        
   WriteAddr += STM32_FLASH_BASE ;  
   CLEAR_BIT(FLASH->CR, FLASH_CR_PER);	
   for(i=0;i<NumToWrite;i++)
   {
      HAL_FLASH_Program(FLASH_TYPEPROGRAM_WORD,WriteAddr,pBuffer[i]);
      WriteAddr+=4;                               
   }  
}

void STMFLASH_Read(uint32_t ReadAddr,uint32_t *pBuffer,uint32_t NumToRead)   	
{
	uint32_t i;
	for(i=0;i<NumToRead;i++)
	{
		pBuffer[i]=STMFLASH_ReadWord(ReadAddr);
		ReadAddr+=4;	
	}
}

void STMFLASH_Erase(uint32_t start_Add)
{
  HAL_FLASH_Unlock();
  CLEAR_BIT(FLASH->CR, FLASH_CR_PER);	
 // addr += STM32_FLASH_BASE;
  FLASH_PageErase(start_Add);
  FLASH_WaitForLastOperation(FLASH_WAITETIME);           

  HAL_FLASH_Lock();
}

e107steved, 2020-03-19:
What error are you getting?
Does your block_size of 2048 match the size of erasable blocks on the device?

fengchun1, 2020-03-19:
@e107steved 
dear
Glad you could answer my question。
After executing the mounted file system, I get an error when opening the file, the file cannot be created, and the error message: lfs_error:977: Corrupted dir pair at 0 0

geky, 2020-03-20:
Hmmm, it looks like you're not adding the base address of the internal flash to your address? Do you need to add the base address?