littlefs-project/littlefs #399: Use littlefs on flash , get LFS_F_ERRED in function lfs_file_write

lijg0511, 2020-03-18:
I use w25n01 nandflash ,mcu use spi comm with flash. 
my requirement is get a page data (2k) writed in frequence 16khz(actual it's i2s data,worked 16k frequence ), it will write 64kb data one sencond . when I write 512kb data,errhappened  in function lfs_file_write ->lfs_ctz_extend ,it will reurn err. so what happened ?thankyou for answer

thrasher8390, 2020-03-18:
Can you post your `lfs_config` settings? What is the size of your external memory?
What have you done to prove that you aren't having and IO error with your serial flash?

lijg0511, 2020-03-19:
> Can you post your lfs_config settings? What is the size of your external memory?

thanku very muchfor my reply !!!! I'm newer,so forgive my superficial understanding.
w25n01(128M 1204blocks) use page(2k)read and write,erase  block(64page) everytime,  so in my lfs config, I set read_size prog_size lookahead_size 2k szie，block_size = PAGE_SIZE*64,block_count = 1024 (128M /128k). multi file usage isnt required,so I use static buffer, so the read_buffer prog_buffer lookahead_buffer is static buffer ,buffer size is 2k. 
another motify is in function lfs_file_opencfg ,file->cache.buffer = lfs_malloc(lfs->cfg->cache_size);     ---->>> file->cache.buffer  = cfg_buffer;  //use static buffer static uint8_t cfg_buffer[2048] = {0};
here give my read write function 
static int _block_read(const struct lfs_config *c, lfs_block_t block, lfs_off_t off, void *buffer, lfs_size_t size)
{
		uint8_t err ;
		uint8_t *ptr = (uint8_t *)buffer;
		uint16_t count =size/2048;
		uint32_t page = block * 64 + off;
		uint16_t i;
		for(i=0;i<count;i++){
			err += serial_Nand_readPage(ptr,page+i); // read one page 
			ptr+=2048;
		}
    return err;  //if success ,always 0
}
static int _block_prog(const struct lfs_config *c, lfs_block_t block, lfs_off_t off, const void *buffer, lfs_size_t size)
{
		uint8_t err;
		uint8_t *ptr = (uint8_t *)buffer;
		uint16_t count =size/2048;
		uint16_t i;
		uint32_t page = block * 64 + off;
		for(i=0;i<count;i++){
			err += serial_Nand_writePage((uint8_t *)ptr,page+i); //write one page
			ptr+=2048;
		}
    return err;
}  
//actuall I want to asure thing .before write page,page must be erased? but as know ,my earse size is one block ,so I just erase the block which page include .is that allright ???
 static int _block_erase(const struct lfs_config *c, lfs_block_t block)
{
		uint8_t ph=0,pl=0;
		ph = (block*64)&0xff;
		pl = (block*64) &0xff;
		if(block ==0)	ph = pl =0;
		WB_Serial_NAND_BlockErase(ph, pl); 
		return 0;
}

> What have you done to prove that you aren't having and IO error with your serial flash?
I use one block to write  read ,verify data ,and all data is ok.
actually, I check other block ,and there is problem happened, block may be bad, so I think I lost bad block manager, I'm trying to using its bad block manager(BBM) to get bad block. the question is ,how littlefs include the bad block function?  can you show me the way to add the BBM in littlefs ?
here give the BBM info :
Due to large NAND memory density size and the technology limitation, NAND memory devices are allowed to be shipped to the end customers with certain amount of “Bad Blocks” found in the factory testing. Up to 2% of the memory blocks can be marked as “Bad Blocks” upon shipment, which is a maximum of 20 blocks for W25N01GV. In order to identify these bad blocks, it is recommended to scan the entire memory array for bad block markers set in the factory. A “Bad Block Marker” is a non-FFh data byte stored at Byte 0 of Page 0 for each bad block. An additional marker is also stored in the first byte of the 64-Byte spare area. 
W25N01GV offers a convenient method to manage the bad blocks typically found in NAND flash memory after extensive use. The “Bad Block Management” command is initiated by shifting the instruction code “A1h” into the DI pin and followed by the 16-bit “Logical Block Address” and 16-bit “Physical Block Address” as illustrated in Figure 11. A Write Enable instruction must be executed before the device will accept the Bad Block Management Instructions (Status Register bit WEL= 1). The logical block address is the address for the “bad” block that will be replaced by the “good” block indicated by the physical block address. 
Once a Bad Block Management command is successfully executed, the specified LBA-PBA link will be added to the internal Look Up Table (LUT). Up to 20 links can be established in the non-volatile LUT. If all 20 links have been written, the LUT-F bit in the Status Register will become a 1, and no more LBA-PBA links can be established. Therefore, prior to issuing the Bad Block Management command, the LUT-F bit value can be checked or a “Read BBM Look Up Table” command can be issued to confirm if spare links are still available in the LUT. 
To guarantee a continuous read operation on the first 1,000 blocks, the manufacturer may have used some of the BBM LUT entrees. It is advisable for the user to scan all blocks and keep a table of all manufacturer bad blocks prior to first erase/program operation.

thanku again and ecpect for your reply !

lijg0511, 2020-03-19:
It seems happened in reglulaty, after 512 pages data write in file, lfs_file_write will return that err. so I guess,is there any limit ?

e107steved, 2020-03-19:
1. You don't set err = 0 in your routines.
2. If the routines that you list are the ones called by littleFs, you need to return a specific error code (as listed in lfs.h, although others can be valid). Zero for OK is valid, but note that all the other error codes have negative values.
And you would probably do better terminating your loop as soon as an error occurs, rather than struggling on - partial failure can often cause far more problems than total failure.

lijg0511, 2020-03-20:
> You don't set err = 0 in your routines.
> If the routines that you list are the ones called by littleFs, you need to return a specific error code (as listed in lfs.h, although others can be valid). Zero for OK is valid, but note that all the other error codes have negative values.
> And you would probably do better terminating your loop as soon as an error occurs, rather than struggling on - partial failure can often cause far more problems than total failure.

thanku for reply, I check what you said,it may cause some problems, and I make up.but it seems not the reason that case the err mentioned. Tryed serveral times, I found some regularity that , multiply of datas (512page * 2k)write over, this err appear .It seems that the file's max writable size is 1Mbyte ?

closed 2020-03-20 by lijg0511