littlefs-project/littlefs #400: lfs_stag_t lfs_dir_fetchmatch

fengchun1, 2020-03-18:
dear all
  in  lfs_stag_t lfs_dir_fetchmatch

  if (crc != dcrc) {
                    dir->erased = false;
                    break;
                }

  // failed, try the other block?
        lfs_pair_swap(dir->pair);
        dir->rev = revs[(r+1)%2];

e107steved, 2020-03-19:
What is your question?
(And note that, in general, littleFs "just works", although there are some bugs which are in the process of being fixed. So you shouldn't usually need to worry about the littleFs code).

fengchun1, 2020-03-19:
My program aborted at （ lfs_file_read LFS_ASSERT(file->flags & LFS_F_OPENED) ),

geky, 2020-03-20:
Hi @fengchun1, that assert indicates that the file has not been opened.

You need to open a file before you can read from it:
``` c
lfs_file_open(&lfs, &file, "boot_count", LFS_O_RDWR | LFS_O_CREAT);
lfs_file_read(&lfs, &file, &boot_count, sizeof(boot_count));
lfs_file_close(&lfs, &file);
```

thrasher8390, 2020-05-04:
@geky, do you have the ability to close issues that aren't bugs or feature requests?