littlefs-project/littlefs #404: lfs_dir_open fails with nested directories

MarkusK92, 2020-03-26:
Hi guys!

I am struggeling with this problem for a couple of days now.. I really really hope you guys can help me out :)

First of all, this is my LittleFS setup
```
static struct lfs_config cfg = {
    // block device operations
    .read  = read_callback,
    .prog  = prog_callback,
    .erase = erase_callback,
    .sync  = sync_callback,

    // block device configuration
    .read_size = 512,
    .prog_size = 512,
    .block_size = 32768,
    .block_count = 466944,
    .cache_size = 1024,
    .lookahead_size = 512,
    .block_cycles = -1
};
```
I am using an ARM A53 64-bit processor with a 16GB eMMC on which I mounted LittleFS. I don't use an OS, it's baremetal right now. I know read and write sizes are uncommonly big, however I cannot reduce this size as I am using a DMA with 512 bytes a transfer and all of this already setup in the drivers. I have changed the block_size, the block_count, the cache_size and the lookahead_size a million times, however it didn't solve my problem. It seems like the problem lies somewhere else.

My problem is that opening nested directories after they were created successfully fails!
I attached a log file of my running program, which should explain a lot:
[lfs_dir_problem_log.txt](https://github.com/ARMmbed/littlefs/files/4388234/lfs_dir_problem_log.txt)

IMPORTANT, I changed the code in lfs.c for debugging as follows
(this change is inside the function static lfs_stag_t lfs_dir_find(lfs_t *lfs, lfs_mdir_t *dir, const char **path, uint16_t *id))
```
// find entry matching name
while (true) {
    printf("name before lfs_dir_fetchmatch: %s\r\n", name);
    tag = lfs_dir_fetchmatch(lfs, dir, dir->tail,
            LFS_MKTAG(0x780, 0, 0),
            LFS_MKTAG(LFS_TYPE_NAME, 0, namelen),
             // are we last name?
            (strchr(name, '/') == NULL) ? id : NULL,
            lfs_dir_find_match, &(struct lfs_dir_find_match){
                lfs, name, namelen});
    printf("name after lfs_dir_fetchmatch: %s\r\n", name);
    if (tag < 0) {
        return tag;
    }

    if (tag) {
        printf("Dir Entry found! tag = %d \r\n", tag);
        break;
    }

    if (!dir->split) {
        printf("No dir entry found! \r\n");
        return LFS_ERR_NOENT;
    }
}
```

Now I recognized looking at the logfile that the "name" gets corrupted inside lfs_dir_fetchmatch and by that also "path". Do you have any ideas what the problem is or any suggestions what to do next?

I already checked if writing to and reading from the eMMC works. It works perfectly.

geky, 2020-03-26:
This will need a bit of time to look into, though I can say that LittleFS should work with the larger prog/read size without issue (it can get larger on RAM).

There's (unfortunately) a long-lived branch here that has a number of bug-fixes. It may be worth checking if this bug has been fixed on that branch:
https://github.com/ARMmbed/littlefs/pull/372

MarkusK92, 2020-03-27:
Thanks for your fast reply! I was very grateful for it and also tested it with the branch you told me. The error was still there, because it was not LittleFS's fault.

Thanks to a great colleague of mine who spent 2 hours debugging with me we found the problem. I created a global variable called **mydir** in main.c, wich was a self created struct with a char array of 256 bytes which held the directory-path. Then, in a different c-File I created another global variable called **mydir**, but of the struct type lfs_dir_t. This object I then gave to **lfs_dir_open**. Then, inside the function **lfs_dir_find** the line
`    dir->tail[0] = lfs->root[0];`
corrupted the path inside my **mydir** object in main.c.

Renaming the global variable **mydir** in main.c to a different name and calling it static solved my problem. I am very surprised that my compiler didn't give me at least a warning, but oh well, that's just how it is. Thanks anyway and have a good weekend!

geky, 2020-03-29:
Oh interesting, that's a strange bug. Yeah I would have expected the compiler to complain about multiple declarations unless one of the "mydirs" was declared extern.

MarkusK92, 2020-04-01:
They were not declared extern, but as I learned now, it's up to the compiler to choose if the global variable is extern or static if it is not specified. GCC chooses extern if it is not specified. Fun fact: A C++ linker would have given me a "multiple definition" error, whereas a C-Linker obviously doesn't.
Now LittleFS works like a charm and it is so much fun! Thanks a lot!

closed 2020-04-01 by MarkusK92