littlefs-project/littlefs #406: In one case, lfs will keep reading flash

Guozhanxin, 2020-04-05:
When I use lfs, the program is stuck. I debugged and found that lfs has been reading flash, and finally found that there is a problem with the following line of code.

https://github.com/ARMmbed/littlefs/blob/master/lfs.c#L49
https://github.com/ARMmbed/littlefs/blob/master/lfs.c#L66
[image]

When diff> size, size will become a very large number, so it will always read flash.

geky, 2020-04-06:
Hi @Guozhanxin, thanks for raising an issue.

What path causes diff > size? I don't think this is possible (but would be happy to be proven wrong).

At the beginning of each pass diff is set `= size`, and only modified `= min(diff, ...)`. The size is only changed before a continue, which marks the end of a pass. So I don't believe it's possible for diff to be > size.

C47D, 2020-04-06:
> What path causes diff > size? I don't think this is possible (but would be happy to be proven wrong).

Not the OP but, can diff underflow?

geky, 2020-04-09:
@C47D That's a good point, but I don't think that's possible either.

`size` is only ever decremented by `diff`. Assuming `diff <= size` (from previous comment) at all points in the program, `size` will never undeflow.

Is this a 32-bit device? Is it possible there is a buffer overflow somewhere corrupting the system. rcache's and pcache's buffers maybe? Does sizeof(cfg->read_buffer) and sizeof(cfg->prog_buffer) == cache_size? (it should)

e107steved, 2020-04-09:
From lfs.h:
typedef uint32_t lfs_size_t;

So size -= diff; can turn size into a large positive number.

Should typedef int32_t  lfs_ssize_t; be used?

X-Ryl669, 2020-04-13:
No, it's safe. Since `diff < size`, `size - diff` is always positive and can not turn "negative" (what you call a large positive number). Since both `diff` and `size` are the same type, no promotion to `int` is done in computing the subtraction so the (negative) sign is never used (that is, `diff` can not be negative and as such, will never cause `size` to increase either).

X-Ryl669, 2020-04-13:
If I were you, I would put a conditional watchpoint on the `size` variable and check where it's modified to become large. It'll pinpoint immediately where the issue is.