littlefs-project/littlefs #408: Implementing littlefs port for nor spi flash

Houe2005, 2020-04-07:
Should the port's prog and erase functions wait until those operations are complete on the SPI flash before the function returns or can the function return while the operation is still in progress? Said another way should the prog and erase functions verify (either wait the appropriate time or better yet poll the flash chip) for operation completion before returning?

When does littlefs call sync? Will sync always be called in between a prog (or erase) and a read? Can littlefs call multiple progs back to back without calling sync in between?

I'm looking for best practices on where this verification needs to be placed. Thanks!

geky, 2020-04-09:
Hi @Houe2005, interesting question.

The sync callback marks the point in time where littlefs needs pending write operations to be reflected on disk. The reason for sync is to allow the hardware to best take advantage of buffers and caching that is available.

So it should be fine to delay progs and erases until sync. littlefs will make sure to call sync before reading from the page (though it may read from other pages while building the commit, [line 1311](https://github.com/ARMmbed/littlefs/blob/a049f1318eecbe502549f9d74a41951985fb956f/lfs.c#L1311) for example).

> Said another way should the prog and erase functions verify (either wait the appropriate time or better yet poll the flash chip) for operation completion before returning

You only need to poll in sync.

Though it's worth noting that polling during prog/erase may be easier to implement. So unless you're writing large files or go a while without calling lfs_file_close/sync, it may not be worth the extra logic to delay a poll until sync.

> Will sync always be called in between a prog (or erase) and a read?

Yep!

> Can littlefs call multiple progs back to back without calling sync in between?

Yes, littlefs may call erase/prog any number of times before calling sync. If you write a file, littlefs will erase+prog enough blocks to store the file data and the file's metadata, only calling sync when you close/sync the file.

Hopefully that helps answer some questions.

Houe2005, 2020-04-10:
Thanks for the reply geky it was very helpful.