littlefs-project/littlefs #409: No space left on device, but write data success

yplwrt, 2020-04-11:
DEBUG:
littlefs/lfs.c:497:error: No more free space 441
LINE 116: lfs write data length: 256
-----------------------------------------------------------------------------------------------------
Code:
		err = lfs_file_open(&g_lfs, &g_lfs_file[0], lfs_read_buf1, LFS_O_RDWR | LFS_O_CREAT);		
		if(err) 
			lynx_prt("lfs open error:%d  LINE %d\n", err, __LINE__);			

		err = lfs_file_write(&g_lfs, &g_lfs_file[0], lfs_write_buf, sizeof(lfs_write_buf));
		if (err > 0)
			lynx_prt("LINE %d: lfs write data length:%d\n", __LINE__,  err);
		
		lfs_file_close(&g_lfs, &g_lfs_file[0]);
--------------------------------------------------------------------------------------------------------
QUESTION:
When the flash space is exhausted, an error should be reported when writing data again, but now the lfs_file_write() function does not report an error. why?

Johnxjj, 2020-04-17:
Inline file？

w366813640, 2021-02-07:
> ## DEBUG:
> littlefs/lfs.c:497:error: No more free space 441
> LINE 116: lfs write data length: 256
> Code:
> const struct lfs_config g_lfs_cfg = {
> 
> ```
> // block device operations
>   .read  = user_provided_block_device_read,
>   .prog  = user_provided_block_device_prog,
>   .erase = user_provided_block_device_erase,
>   .sync  = user_provided_block_device_sync,
> 
>   // block device configuration
>   .read_size = 256,
>   .prog_size = 256,
>   .block_size = 4096,
>   .block_count = 256,
>   .cache_size = 256,
>   .lookahead_size = 32,
>   .block_cycles = 500,
> ```
> 
> };
> err = lfs_file_open(&g_lfs, &g_lfs_file[0], lfs_read_buf1, LFS_O_RDWR | LFS_O_CREAT);
> if(err)
> printf("lfs open error:%d LINE %d\n", err, **LINE**);
> 
> ```
> 	err = lfs_file_write(&g_lfs, &g_lfs_file[0], lfs_write_buf, sizeof(lfs_write_buf));
> 	if (err > 0)
> 		printf("LINE %d: lfs write data length:%d\n", __LINE__,  err);
> 	
> 	lfs_file_close(&g_lfs, &g_lfs_file[0]);
> ```
> 
> QUESTION:
> When the flash space is exhausted, an error should be reported when writing data again, but now the lfs_file_write() function does not report an error. why?

I have met the same issue, want to know the reason.

geky, 2021-02-08:
If I understand correctly, you are calling write after open returns LFS_ERR_NOSPC?

littlefs doesn't handle this, the file is left uninitialized if lfs_file_open errors.

So the correct code would be something like this:

``` c
err = lfs_file_open(&g_lfs, &g_lfs_file[0], lfs_read_buf1, LFS_O_RDWR | LFS_O_CREAT);       
if(err) 
    printf("lfs open error:%d  LINE %d\n", err,  __LINE__);         

if (!err) {
    err = lfs_file_write(&g_lfs, &g_lfs_file[0], lfs_write_buf, sizeof(lfs_write_buf));
    if (err > 0)
        printf("LINE %d: lfs write data length:%d\n", __LINE__,  err);
    
    err = lfs_file_close(&g_lfs, &g_lfs_file[0]);
    if(err) 
        printf("lfs close error:%d  LINE %d\n", err,  __LINE__);     
}
```

We recently added some checks that assert in this case, but they require that asserts are enabled (https://github.com/littlefs-project/littlefs/pull/443)

JonJust, 2021-05-11:
Hey there,

You wouldn't happen to be developing for the winbond w25, would you? What does your disk io look like?