littlefs-project/littlefs #410: Close the file , open and read again, Assertion failed, (file->flags & LFS_F_OPENED)

WangRui-wnlbs, 2020-04-13:
My flash type:Micron Serial NOR Flash Memory(N25Q00AA), 
My question is below:
Test case main() result is OK;but test case LittleFS_main(), after this step: lfs_file_close(&lfs, &file), open and read file again.when read again there are errors as :E:/Code/A9/Project/BSP/am335x_sysbios_ind_sdk_1.1.0.4/sdk/os_drivers/src/LittleFS/lfs.c:1006:error: Corrupted dir pair at {0x1, 0x0}
Assertion failed, (file->flags & LFS_F_OPENED), file E:/Code/A9/Project/BSP/am335x_sysbios_ind_sdk_1.1.0.4/sdk/os_drivers/src/LittleFS/lfs.c, line 2818

I can't find the reason，can you give me some tips? thank you.

Micron Serial NOR Flash Memory(N25Q00AA)
```
#define SPIFLASH_PAGESIZE         (256U)
#define SPIFLASH_SUBSECTORSIZE       4096
#define SPIFLASH_SECTORSIZE       65536

// configuration of the filesystem is provided by this struct
const struct lfs_config cfg =
{
    // block device operations
    .read  = user_provided_block_device_read,
    .prog  = user_provided_block_device_prog,
    .erase = user_provided_block_device_erase,
    .sync  = user_provided_block_device_sync,

    // block device configuration
    .read_size = SPIFLASH_PAGESIZE,
    .prog_size = SPIFLASH_PAGESIZE,
    .block_size = SPIFLASH_SUBSECTORSIZE,
    .block_count = 12*1024*1024 / SPIFLASH_SUBSECTORSIZE,
    .cache_size = SPIFLASH_SUBSECTORSIZE,
    .lookahead_size = 24,
    .block_cycles = 500,
};

int main(void) {
    // mount the filesystem
    int err = lfs_mount(&lfs, &cfg);

    // reformat if we can't mount the filesystem
    // this should only happen on the first boot
    if (err) {
        lfs_format(&lfs, &cfg);
        lfs_mount(&lfs, &cfg);
    }

    // read current count
    uint32_t boot_count = 0;
    lfs_file_open(&lfs, &file, "boot_count", LFS_O_RDWR | LFS_O_CREAT);
    lfs_file_read(&lfs, &file, &boot_count, sizeof(boot_count));

    // update boot count
    boot_count += 1;
    lfs_file_rewind(&lfs, &file);
    lfs_file_write(&lfs, &file, &boot_count, sizeof(boot_count));

    // remember the storage is not updated until the file is closed successfully
    lfs_file_close(&lfs, &file);

    // release any resources we were using
    lfs_unmount(&lfs);

    // print the boot count
    printf("boot_count: %d\n", boot_count);
}

int LittleFS_main(void)
{
    // mount the filesystem
    int err = lfs_mount(&lfs, &cfg);

    // reformat if we can't mount the filesystem
    // this should only happen on the first boot
    if (err)
    {
    	printf("lfs_mount_EER\r\n");
        lfs_format(&lfs, &cfg);
        lfs_mount(&lfs, &cfg);
    }
   
    // read current count
    uint32_t boot_count = 0;
    lfs_file_open(&lfs, &file, "boot_count", LFS_O_RDWR | LFS_O_CREAT);
    lfs_file_read(&lfs, &file, &boot_count, sizeof(boot_count));

    // update boot count
    boot_count += 1;
    lfs_file_rewind(&lfs, &file);
    lfs_file_write(&lfs, &file, &boot_count, sizeof(boot_count));

    // remember the storage is not updated until the file is closed successfully
    lfs_file_close(&lfs, &file);

    lfs_file_open(&lfs, &file, "boot_count", LFS_O_RDWR | LFS_O_CREAT);
	lfs_file_read(&lfs, &file, &boot_count, sizeof(boot_count));

	// update boot count
	boot_count += 1;
	lfs_file_rewind(&lfs, &file);
	lfs_file_write(&lfs, &file, &boot_count, sizeof(boot_count));

    // release any resources we were using
    lfs_unmount(&lfs);
	
    // print the boot count
    printf("boot_count: %d\n", boot_count);
}
```

e107steved, 2020-04-16:
Firstly, return values are often negative for error, zero or positive for success. It's a good habit to check for (err < 0) rather then (err != 0).
Secondly, check the return codes for lfs_file_open(), lfs_file_read() and so on. It might be that an early call is failing, and it is therefore invalid to keep going - hence the assertion.

WangRui-wnlbs, 2020-06-12:
The interface user_provided_block_device_erase of Adaptive layer is't correct.

closed 2020-06-12 by WangRui-wnlbs