littlefs-project/littlefs #413: Easy replacing FatFS by littlefs

pjanco, 2020-04-14:
I want to use littlefs on SD card for my existing project where ElmChan FatFS library was used before (NXP SDHC interface).

I am not sure if callback funcions used by FatFS library are compatible with littlefs.
FatFS library for example do not use erase and sync funcions.
Functions for read/write in FatFS use number of sectors as "size" parameter.
It looks like littlefs use numer of bytes as "size" parameter.

Can anyone provide a example of callback functions compatible with FatFS interface?

pjanco, 2020-04-15:
Now I am not sure if littlefs is good choise for SD card or eMMC chip because both are using internal on-chip implementation for wear leveling, erasing and copy-on-write. With littlefs all these funcion will be done twice.

Littlefs is designed only for RAW memory. It is my opinion correct?

e107steved, 2020-04-16:
I don't think it's practical to use LFS through your FatFS interface routines - things are organised differently! (And the LFS callbacks are slightly simpler). The FatFS hardware interface shim should provide you with a good start, since the underlying operations end up being pretty much the same - read a block of memory, write a block of memory, erase a block..... It's just how you join it all together that's different.

I've not looked at the use of SD/eMMC, but my memory of comments here is that:
a) littlefs doesn't scale well for bigger memories
b) You can disable wear levelling, but the internal 'intelligence' of SD cards causes some problems.

This may have changed with the very latest release. There's a lot of useful information and discussion in the replies to some of these issues.

Incidentally, for my own application I kept FatFS for the memory card, since that's a standard format for interchange with the 'outside world'. I'm using littlefs for internal memories for two reasons:
a) FatFs has a substantial overhead for the root directory - about 20% of one of my memories! LFS takes very much less.
b) FatFS requires that sector size is the same as erasable block size, and also at least one memory buffer equal to the sector size. Even with an erasable sector size of 4K, that's a lot of RAM on an embedded processor (especially when many of the files are 1kb or less)