littlefs-project/littlefs #414: calling `lfs_fs_size` get stuck in a loop

ShaharHD, 2020-04-15:
I have managed to get the littlefs instance I have into a weird state.

It was probably writing while I turned off the device, which if I remember correctly was connected via an active SWI (JTAG) interface.

I downloaded the whole QSPI ram (16MB) - so I have it as a hex file
[lfs_errors.zip](https://github.com/ARMmbed/littlefs/files/4483746/lfs_errors.zip)

Any other way I can assist with debugging this issue?
(after I've downloaded the hexfile, I erased the qspi device, no issue afterwards)

Here's are the parameters for this storage device (nothing secret about the data in the files):
```c
#define LFS_BLOCK_COUNT (4096)
#define LFS_BLOCK_OP_SIZE (128)
#define LFS_CACHE_SIZE (LFS_BLOCK_OP_SIZE * 4)
#define LFS_LOOKAHEAD_SIZE (LFS_BLOCK_COUNT / 8) // INFO "Waste" the ram to have full bitmap of the blocks
#define LFS_BLOCK_SIZE (4096)

static uint8_t __ALIGN(4) m_lfs_read_buffer[LFS_CACHE_SIZE];
static uint8_t __ALIGN(4) m_lfs_prog_buffer[LFS_CACHE_SIZE];
static uint8_t __ALIGN(4) m_lfs_lookahead_buffer[LFS_LOOKAHEAD_SIZE];

static const struct lfs_config m_lfs_config = {
    .context = NULL,

    // block device operations
    .read  = &lfs_block_dev_read,
    .prog  = &lfs_block_dev_prog,
    .erase = &lfs_block_dev_erase,
    .sync  = &lfs_block_dev_sync,

    // block device configuration
    .read_size      = LFS_BLOCK_OP_SIZE,
    .prog_size      = LFS_BLOCK_OP_SIZE,
    .block_size     = LFS_BLOCK_SIZE,
    .block_count    = LFS_BLOCK_COUNT,
    .cache_size     = LFS_CACHE_SIZE,
    .lookahead_size = LFS_LOOKAHEAD_SIZE,
    .block_cycles   = 500,

    // statically allocated buffers
    .read_buffer      = m_lfs_read_buffer,
    .prog_buffer      = m_lfs_prog_buffer,
    .lookahead_buffer = m_lfs_lookahead_buffer,
};
```

ShaharHD, 2020-04-15:
Could this be related to #394 ?

e107steved, 2020-04-16:
> 
> 
> Could this be related to #394 ?

Are you using the very latest littleFS? If so, I think #394 might have been fixed.

ShaharHD, 2020-04-16:
I will pull the latest release, re-test, and will update.

The version currently in the code is using 2.1.

ShaharHD, 2020-04-16:
Updates:

`2.2.1` was successful in mounting the storage - which is awesome!

Does it interests you to investigate it?
It reports error for `lfs_fs_size failed: -84` (`LFS_ERR_CORRUPT`)

But other than that ... it looks like valid files (but there's no way for me of knowing which files might be corrupt)

Should I simply format the storage?
Or is there a way to have like a "check-disk" to re-scan the storage and evaluate the actual used pages?

closed 2020-04-17 by ShaharHD