littlefs-project/littlefs #415: This working NAND configuration, is it the best?

kazola, 2020-04-17:
Dear all,
I have been using littlefs v2.0 with the NAND I specify next and seems to work more or less fine. v2.2 instead, seems a bit more slower. However, this is not the point for this issue thread :)

From the datasheet: The W25N01GW1G-bit memory array is organized into 65,536 programmable pages of 2,048-bytes each. The entire page can be programmed at one time using the data from the 2,048-Byte internal buffer. Pages can be erased in groups of 64 (128KB block erase). The W25N01GW has 1,024 erasable blocks

Now, after reading a lot of posts about performance on NAND, I thought could be nice to share my configuration, as well as to ask if this is the best one you could recommend. Consider plenty of RAM at this point, since, as I said, it works.

```
// -------------------------------
// memory hardware
// -------------------------------
#define MEM_BYTES_PER_PAGE          (2048)
#define MEM_PAGES_PER_BLOCK         (64)
#define MEM_BLOCK_SIZE              (MEM_BYTES_PER_PAGE * MEM_PAGES_PER_BLOCK)
#define MEM_BLOCK_COUNT             (1024)

// -------------------------
// LittleFS memory bridging
// -------------------------
#define LFS_TO_MEM_READ_SIZE        MEM_BYTES_PER_PAGE
#define LFS_TO_MEM_PROG_SIZE        MEM_BYTES_PER_PAGE
#define LFS_TO_MEM_LOOK_SIZE        (MEM_BLOCK_COUNT / 8)
#define LFS_TO_MEM_CACHE_SIZE       MEM_BYTES_PER_PAGE
#define LFS_TO_MEM_BLOCK_SIZE       MEM_BLOCK_SIZE
#define LFS_TO_MEM_BLOCK_COUNT      MEM_BLOCK_COUNT

const struct lfs_config lfs_cfg =
{
    .read  = user_provided_block_device_read,
    .prog  = user_provided_block_device_prog,
    .erase = user_provided_block_device_erase,
    .sync = user_provided_block_device_sync,

    // whenever these values change, reformat memory
    .read_size = LFS_TO_MEM_READ_SIZE,
    .prog_size = LFS_TO_MEM_PROG_SIZE,
    .block_size = MEM_BLOCK_SIZE,
    .block_count = MEM_BLOCK_COUNT,
    .cache_size = LFS_TO_MEM_CACHE_SIZE,
    .lookahead_size = LFS_TO_MEM_LOOK_SIZE,
};
```

I am concernetd about the block count, but please feel free to suggest anything. Again, this is NAND and I have plenty of RAM. This system has been working more or less ok but I am not very demanding in terms of search of files, for example.

Afterwards, maybe I can do some performance tests on littlefs v2.2. I even thought like a small performance test suite useful to compare tests on this hardware since lots of people seems to use it but seems to be doing different tests which are difficult to extrapolate to compare.

Please have a nice day and take care!

ShaharHD, 2020-04-19:
Is there a reason you use the big `64K` pages and not the `2K` pages?

Seems to me you might be more efficient if you have a lots of small writes maybe?
As it won't need to erase a whole 64K block

kazola, 2020-04-20:
Thanks for your input @ShaharHD :)
Let's see if anyone can correct me further

ShaharHD, 2020-04-20:
@kazola I would try to increase the `read_size` and `prog_size` - as maybe having a bigger buffers (as you have no memory shortage) would give you better throughput

githubwilliamx, 2020-04-21:
Hi kazola, I am also using the same flash memory chip as you.But it doesn't work,could you  share your code？plz.

kazola, 2020-04-21:
Hi @githubwilliamx ,
at this point I cannot share the code, it is not my property.
However, littlefs checks the written stuff so you do not need the check part of yours. Also, be sure when using the same sizes as me, you have enough RAM!

kazola, 2020-04-21:
@ShaharHD will try! thanks! even if as I said this more or less works, I tried the lower tests, but not increasing beyond 2048, do you see the rest of configuration more or less logical? I am afraid of block count since I read around memory hardware block count does not need to match littlefs block count

e107steved, 2020-04-21:
@githubwilliamx
Without fully understanding your code, it assumes that Page_Addr is always a multiple of 2048. Is this always the case? If not, I would guess that the '0' parameter to WB_Serial_NAND_Pageprogram_Pattern() is actually the offset within the page. (Even if Page_Addr is a multiple of 2048, I would suggest taking account of that offset, to avoid problems later)

kazola, 2020-04-26:
I did some benchmarking tests with real numbers so I will open a new issue related to this.

closed 2020-04-26 by kazola