littlefs-project/littlefs #416: littlefs v2.0 speed test

kazola, 2020-04-26:
Hi,

I wanted to test about littlefs getting slower after some files populate the system.
I indicate hardware used, littlefs version, littlefs configuration and tests results. Questions are at the bottom.
Feel free to comment, suggest, ask for more (simple) tests.

### hardware
- SPI freq 6 MHz, 12 MHz does not change much. Does not do 2 x speed. 
- NAND memory details: Winbond W25N01GW1G-bit 65536 pages, 2048-bytes.
- Page can be programmed at one time the 2,048-Byte internal buffer. 
- Pages can be erased in groups of 64 (128KB block erase). 
- The W25N01GW has 1,024 erasable blocks.

### littlefs version used:
```
#define LFS_VERSION 0x00020000
#define LFS_DISK_VERSION 0x00020000
```

### littlefs configuration used:
```
// -------------------------------
// memory hardware
// -------------------------------
#define MEM_BYTES_PER_PAGE          (2048)
#define MEM_PAGES_PER_BLOCK         (64)
#define MEM_BLOCK_SIZE              (MEM_BYTES_PER_PAGE * MEM_PAGES_PER_BLOCK)
#define MEM_BLOCK_COUNT             (1024)

// -------------------------
// LittleFS memory bridging
// -------------------------
#define LFS_TO_MEM_READ_SIZE        MEM_BYTES_PER_PAGE
#define LFS_TO_MEM_PROG_SIZE        MEM_BYTES_PER_PAGE
#define LFS_TO_MEM_LOOK_SIZE        (MEM_BLOCK_COUNT / 8)
#define LFS_TO_MEM_CACHE_SIZE       MEM_BYTES_PER_PAGE
#define LFS_TO_MEM_BLOCK_SIZE       MEM_BLOCK_SIZE
#define LFS_TO_MEM_BLOCK_COUNT      MEM_BLOCK_COUNT

const struct lfs_config lfs_cfg =
{
    .read  = user_provided_block_device_read,
    .prog  = user_provided_block_device_prog,
    .erase = user_provided_block_device_erase,
    .sync = user_provided_block_device_sync,

    // whenever these values change, reformat memory
    .read_size = LFS_TO_MEM_READ_SIZE,
    .prog_size = LFS_TO_MEM_PROG_SIZE,
    .block_size = MEM_BLOCK_SIZE,
    .block_count = MEM_BLOCK_COUNT,
    .cache_size = LFS_TO_MEM_CACHE_SIZE,
    .lookahead_size = LFS_TO_MEM_LOOK_SIZE,
};
```

### Tests
We test speed for 100 files created, length 20480 bytes (10 blocks long), file content all 0xAA bytes. Test results are:

000046 [ 25.937 ] INFO: (test_fs.c:207) # time:        w_file AA.txt   0:   20480 bytes,  1229 ms
000047 [ 27.268 ] INFO: (test_fs.c:207) # time:        w_file AB.txt   1:   20480 bytes,  1231 ms
000048 [ 28.603 ] INFO: (test_fs.c:207) # time:        w_file AC.txt   2:   20480 bytes,  1233 ms
000049 [ 29.942 ] INFO: (test_fs.c:207) # time:        w_file AD.txt   3:   20480 bytes,  1238 ms
000050 [ 31.284 ] INFO: (test_fs.c:207) # time:        w_file AE.txt   4:   20480 bytes,  1242 ms
000051 [ 32.631 ] INFO: (test_fs.c:207) # time:        w_file AF.txt   5:   20480 bytes,  1246 ms
000052 [ 33.983 ] INFO: (test_fs.c:207) # time:        w_file AG.txt   6:   20480 bytes,  1251 ms
000053 [ 35.342 ] INFO: (test_fs.c:207) # time:        w_file AH.txt   7:   20480 bytes,  1258 ms
000054 [ 36.707 ] INFO: (test_fs.c:207) # time:        w_file AI.txt   8:   20480 bytes,  1264 ms
000055 [ 38.076 ] INFO: (test_fs.c:207) # time:        w_file AJ.txt   9:   20480 bytes,  1269 ms
000056 [ 39.454 ] INFO: (test_fs.c:207) # time:        w_file BA.txt  10:   20480 bytes,  1277 ms
[... 72 lines cut ...]
000129 [ 1085.724 ] INFO: (test_fs.c:207) # time:      w_file ID.txt  83:   20480 bytes, 53265 ms
000130 [ 1139.133 ] INFO: (test_fs.c:207) # time:      w_file IE.txt  84:   20480 bytes, 53308 ms
000131 [ 1192.584 ] INFO: (test_fs.c:207) # time:      w_file IF.txt  85:   20480 bytes, 53351 ms
000132 [ 1246.080 ] INFO: (test_fs.c:207) # time:      w_file IG.txt  86:   20480 bytes, 53396 ms
000133 [ 1299.619 ] INFO: (test_fs.c:207) # time:      w_file IH.txt  87:   20480 bytes, 53440 ms
000134 [ 1353.204 ] INFO: (test_fs.c:207) # time:      w_file II.txt  88:   20480 bytes, 53485 ms
000135 [ 1406.834 ] INFO: (test_fs.c:207) # time:      w_file IJ.txt  89:   20480 bytes, 53530 ms
000136 [ 1460.510 ] INFO: (test_fs.c:207) # time:      w_file JA.txt  90:   20480 bytes, 53576 ms
000137 [ 1514.233 ] INFO: (test_fs.c:207) # time:      w_file JB.txt  91:   20480 bytes, 53622 ms
000138 [ 1568.002 ] INFO: (test_fs.c:207) # time:      w_file JC.txt  92:   20480 bytes, 53670 ms
000139 [ 1621.819 ] INFO: (test_fs.c:207) # time:      w_file JD.txt  93:   20480 bytes, 53716 ms
000140 [ 1675.684 ] INFO: (test_fs.c:207) # time:      w_file JE.txt  94:   20480 bytes, 53765 ms
000141 [ 1729.597 ] INFO: (test_fs.c:207) # time:      w_file JF.txt  95:   20480 bytes, 53813 ms
000142 [ 1783.558 ] INFO: (test_fs.c:207) # time:      w_file JG.txt  96:   20480 bytes, 53862 ms
000143 [ 1837.570 ] INFO: (test_fs.c:207) # time:      w_file JH.txt  97:   20480 bytes, 53911 ms
000144 [ 1891.631 ] INFO: (test_fs.c:207) # time:      w_file JI.txt  98:   20480 bytes, 53961 ms
000145 [ 1945.742 ] INFO: (test_fs.c:207) # time:      w_file JJ.txt  99:   20480 bytes, 54011 ms

### Questions
My questions (Q1, Q2, Q3) are as follow. Maybe Q2 has to be done after adjusting Q1.
    Q1) I have plenty of RAM, is my current littlefs configuration optimal?
    Q2) Any hint if latest littlefs solves this?
    Q3) Do you recommend switching to NOR technology? Any particular model / manufacturer?

Have a really nice day and hope you are all well.

kazola, 2020-04-27:
Hi guys,

just done the same for latest littlefs commit 4c9146e and is worse. Any clue is welcome.

001047 [ 53.824 ] INFO: (test_fs.c:207) # time:        w_file AA.txt   0:   20480 bytes,  1271 ms
001048 [ 55.535 ] INFO: (test_fs.c:207) # time:        w_file AB.txt   1:   20480 bytes,  1610 ms
001049 [ 57.609 ] INFO: (test_fs.c:207) # time:        w_file AC.txt   2:   20480 bytes,  1973 ms
001050 [ 60.089 ] INFO: (test_fs.c:207) # time:        w_file AD.txt   3:   20480 bytes,  2380 ms
001051 [ 63.023 ] INFO: (test_fs.c:207) # time:        w_file AE.txt   4:   20480 bytes,  2833 ms
001052 [ 66.454 ] INFO: (test_fs.c:207) # time:        w_file AF.txt   5:   20480 bytes,  3331 ms
001053 [ 70.429 ] INFO: (test_fs.c:207) # time:        w_file AG.txt   6:   20480 bytes,  3874 ms
001054 [ 74.991 ] INFO: (test_fs.c:207) # time:        w_file AH.txt   7:   20480 bytes,  4462 ms
001055 [ 80.188 ] INFO: (test_fs.c:207) # time:        w_file AI.txt   8:   20480 bytes,  5096 ms
001056 [ 86.062 ] INFO: (test_fs.c:207) # time:        w_file AJ.txt   9:   20480 bytes,  5774 ms
001057 [ 92.662 ] INFO: (test_fs.c:207) # time:        w_file BA.txt  10:   20480 bytes,  6499 ms
001058 [ 100.030 ] INFO: (test_fs.c:207) # time:       w_file BB.txt  11:   20480 bytes,  7268 ms
001059 [ 108.213 ] INFO: (test_fs.c:207) # time:       w_file BC.txt  12:   20480 bytes,  8083 ms
001060 [ 117.255 ] INFO: (test_fs.c:207) # time:       w_file BD.txt  13:   20480 bytes,  8942 ms
001061 [ 127.202 ] INFO: (test_fs.c:207) # time:       w_file BE.txt  14:   20480 bytes,  9846 ms
001062 [ 138.099 ] INFO: (test_fs.c:207) # time:       w_file BF.txt  15:   20480 bytes, 10796 ms
001063 [ 149.993 ] INFO: (test_fs.c:207) # time:       w_file BG.txt  16:   20480 bytes, 11793 ms
001064 [ 162.927 ] INFO: (test_fs.c:207) # time:       w_file BH.txt  17:   20480 bytes, 12834 ms
001065 [ 176.948 ] INFO: (test_fs.c:207) # time:       w_file BI.txt  18:   20480 bytes, 13920 ms
001066 [ 192.099 ] INFO: (test_fs.c:207) # time:       w_file BJ.txt  19:   20480 bytes, 15051 ms
001067 [ 208.427 ] INFO: (test_fs.c:207) # time:       w_file CA.txt  20:   20480 bytes, 16227 ms
001068 [ 225.976 ] INFO: (test_fs.c:207) # time:       w_file CB.txt  21:   20480 bytes, 17449 ms
001069 [ 244.792 ] INFO: (test_fs.c:207) # time:       w_file CC.txt  22:   20480 bytes, 18716 ms
001070 [ 264.921 ] INFO: (test_fs.c:207) # time:       w_file CD.txt  23:   20480 bytes, 20028 ms
001071 [ 286.406 ] INFO: (test_fs.c:207) # time:       w_file CE.txt  24:   20480 bytes, 21385 ms
001072 [ 309.292 ] INFO: (test_fs.c:207) # time:       w_file CF.txt  25:   20480 bytes, 22786 ms
001073 [ 333.627 ] INFO: (test_fs.c:207) # time:       w_file CG.txt  26:   20480 bytes, 24235 ms
001074 [ 359.454 ] INFO: (test_fs.c:207) # time:       w_file CH.txt  27:   20480 bytes, 25726 ms
001075 [ 386.818 ] INFO: (test_fs.c:207) # time:       w_file CI.txt  28:   20480 bytes, 27263 ms
001076 [ 415.765 ] INFO: (test_fs.c:207) # time:       w_file CJ.txt  29:   20480 bytes, 28847 ms
001077 [ 446.340 ] INFO: (test_fs.c:207) # time:       w_file DA.txt  30:   20480 bytes, 30475 ms
001078 [ 582.661 ] INFO: (test_fs.c:207) # time:       w_file DB.txt  31:   20480 bytes, 136223 ms
001079 [ 585.551 ] INFO: (test_fs.c:207) # time:       w_file DC.txt  32:   20480 bytes,  2789 ms
001080 [ 590.226 ] INFO: (test_fs.c:207) # time:       w_file DD.txt  33:   20480 bytes,  4575 ms
001081 [ 596.712 ] INFO: (test_fs.c:207) # time:       w_file DE.txt  34:   20480 bytes,  6384 ms
001082 [ 605.051 ] INFO: (test_fs.c:207) # time:       w_file DF.txt  35:   20480 bytes,  8239 ms
001083 [ 615.290 ] INFO: (test_fs.c:207) # time:       w_file DG.txt  36:   20480 bytes, 10139 ms
001084 [ 627.474 ] INFO: (test_fs.c:207) # time:       w_file DH.txt  37:   20480 bytes, 12084 ms
001085 [ 641.649 ] INFO: (test_fs.c:207) # time:       w_file DI.txt  38:   20480 bytes, 14074 ms
001086 [ 657.857 ] INFO: (test_fs.c:207) # time:       w_file DJ.txt  39:   20480 bytes, 16108 ms
001087 [ 676.146 ] INFO: (test_fs.c:207) # time:       w_file EA.txt  40:   20480 bytes, 18188 ms
001088 [ 696.559 ] INFO: (test_fs.c:207) # time:       w_file EB.txt  41:   20480 bytes, 20313 ms
001089 [ 719.146 ] INFO: (test_fs.c:207) # time:       w_file EC.txt  42:   20480 bytes, 22486 ms
001090 [ 743.949 ] INFO: (test_fs.c:207) # time:       w_file ED.txt  43:   20480 bytes, 24703 ms
001091 [ 771.014 ] INFO: (test_fs.c:207) # time:       w_file EE.txt  44:   20480 bytes, 26965 ms
001092 [ 800.385 ] INFO: (test_fs.c:207) # time:       w_file EF.txt  45:   20480 bytes, 29270 ms
001093 [ 832.107 ] INFO: (test_fs.c:207) # time:       w_file EG.txt  46:   20480 bytes, 31621 ms
001094 [ 866.226 ] INFO: (test_fs.c:207) # time:       w_file EH.txt  47:   20480 bytes, 34019 ms
001095 [ 902.787 ] INFO: (test_fs.c:207) # time:       w_file EI.txt  48:   20480 bytes, 36461 ms
001096 [ 941.838 ] INFO: (test_fs.c:207) # time:       w_file EJ.txt  49:   20480 bytes, 38951 ms
001097 [ 983.421 ] INFO: (test_fs.c:207) # time:       w_file FA.txt  50:   20480 bytes, 41483 ms
001098 [ 1027.579 ] INFO: (test_fs.c:207) # time:      w_file FB.txt  51:   20480 bytes, 44058 ms
001099 [ 1074.360 ] INFO: (test_fs.c:207) # time:      w_file FC.txt  52:   20480 bytes, 46681 ms
001100 [ 1123.809 ] INFO: (test_fs.c:207) # time:      w_file FD.txt  53:   20480 bytes, 49349 ms
001101 [ 1175.969 ] INFO: (test_fs.c:207) # time:      w_file FE.txt  54:   20480 bytes, 52060 ms
[... cut ...]
001108 [ 1620.892 ] INFO: (test_fs.c:207) # time:      w_file GB.txt  61:   20480 bytes, 72325 ms
001109 [ 1696.392 ] INFO: (test_fs.c:207) # time:      w_file GC.txt  62:   20480 bytes, 75400 ms
001110 [ 2064.475 ] INFO: (test_fs.c:207) # time:      w_file GD.txt  63:   20480 bytes, 367988 ms
001111 [ 2068.809 ] INFO: (test_fs.c:207) # time:      w_file GE.txt  64:   20480 bytes,  4234 ms
001112 [ 2076.377 ] INFO: (test_fs.c:207) # time:      w_file GF.txt  65:   20480 bytes,  7467 ms

..... (stopped here but curious about 367s and then 4s)....

kazola, 2020-04-28:
Dear,
I keep tracking this and adding info as I get it. I include traces to detect the time it takes to do some of littlefs instructions.

Around file 40, and always around this time, the time for lfs_open(), indicated as "o" in the next trace, increases drastically. "w" stands for the first of the 10 writes of every file, "c" for close.

000405 [ 160.661 ] INFO: (lfs.c:2278) b - a = 3
000406 [ 160.731 ] INFO: (lfs.c:2296) c - b = 7062
000407 [ 160.842 ] INFO: (lfs.c:2350) d - c = 11032
000408 [ 160.842 ] INFO: (lfs.c:2381) e - d = 0
000409 [ 160.842 ] INFO: (lfs.c:2397) f - e = 6
000410 [ 160.842 ] INFO: (lfs.c:2429) g - f = 3
000411 [ 160.842 ] INFO: (test_fs.c:177) # time: o,   181 ms
000412 [ 161.102 ] INFO: (test_fs.c:183) # time: w,   159 ms
000413 [ 162.300 ] INFO: (test_fs.c:189) # time: c,   110 ms
000414 [ 162.400 ] INFO: (test_fs.c:218) # time:       w_file DI.txt  38:   20480 bytes,  1739 ms
000415 [ 162.500 ] INFO: (lfs.c:2278) b - a = 3
000416 [ 162.614 ] INFO: (lfs.c:2296) c - b = 11338
000417 [ 162.724 ] INFO: (lfs.c:2350) d - c = 11032
000418 [ 162.724 ] INFO: (lfs.c:2381) e - d = 0
000419 [ 162.724 ] INFO: (lfs.c:2397) f - e = 3
000420 [ 162.724 ] INFO: (lfs.c:2429) g - f = 6
000421 [ 162.724 ] INFO: (test_fs.c:177) # time: o,   223 ms
000422 [ 162.984 ] INFO: (test_fs.c:183) # time: w,   159 ms
000423 [ 164.182 ] INFO: (test_fs.c:189) # time: c,   110 ms
000424 [ 164.282 ] INFO: (test_fs.c:218) # time:       w_file DJ.txt  39:   20480 bytes,  1782 ms
000425 [ 164.383 ] INFO: (lfs.c:2278) b - a = 3
000426 [ 164.496 ] INFO: (lfs.c:2296) c - b = 11344            
000427 [ 174.831 ] INFO: (lfs.c:2350) d - c = 1033506
000428 [ 174.831 ] INFO: (lfs.c:2381) e - d = 3            
000429 [ 174.831 ] INFO: (lfs.c:2397) f - e = 3             
000430 [ 174.831 ] INFO: (lfs.c:2429) g - f = 3             
000431 [ 174.831 ] INFO: (test_fs.c:177) # time: o, 10448 ms
000432 [ 175.091 ] INFO: (test_fs.c:183) # time: w,   159 ms
000433 [ 176.289 ] INFO: (test_fs.c:189) # time: c,   110 ms
000434 [ 176.389 ] INFO: (test_fs.c:218) # time:       w_file EA.txt  40:   20480 bytes, 12006 ms
000435 [ 176.490 ] INFO: (lfs.c:2278) b - a = 3
000436 [ 176.561 ] INFO: (lfs.c:2296) c - b = 7096
000437 [ 187.324 ] INFO: (lfs.c:2350) d - c = 1076375
000438 [ 187.324 ] INFO: (lfs.c:2381) e - d = 0
000439 [ 187.324 ] INFO: (lfs.c:2397) f - e = 6
000440 [ 187.324 ] INFO: (lfs.c:2429) g - f = 3
000441 [ 187.324 ] INFO: (test_fs.c:177) # time: o, 10834 ms
000442 [ 187.584 ] INFO: (test_fs.c:183) # time: w,   159 ms
000443 [ 188.782 ] INFO: (test_fs.c:189) # time: c,   110 ms
000444 [ 188.882 ] INFO: (test_fs.c:218) # time:       w_file EB.txt  41:   20480 bytes, 12393 ms
000445 [ 188.983 ] INFO: (lfs.c:2278) b - a = 3

Before every "o", I have further subdivided the time inside lfs_open() as in the following code. It is just lfs_opencfg() from littlefs with some modifications to keep track of time.

```
/// Top level file operations ///
int lfs_file_opencfg(lfs_t *lfs, lfs_file_t *file,
        const char *path, int flags,
        const struct lfs_file_config *cfg) {
    // deorphan if we haven't yet, needed at most once after poweron

    uint32_t a = Clock_getTicks();

    if ((flags & 3) != LFS_O_RDONLY) {
        int err = lfs_fs_forceconsistency(lfs);
        if (err) {
            return err;
        }
    }

    uint32_t b = Clock_getTicks();
    LY1("b - a = %d", b - a);

    // setup simple file details
    int err;
[... 129 lines cut ...]
            if (res < 0) {
                err = res;
                goto cleanup;
            }
        }
    }

    uint32_t g = Clock_getTicks();
    LY1("g - f = %d", g - f);

    return 0;

cleanup:
    // clean up lingering resources
    file->flags |= LFS_F_ERRED;
    lfs_file_close(lfs, file);
    return err;
}
```

So, it would seem what happens between points "d" and "c" at iteration number 39-40. At this point, my RTOS task stack peak increases from 1400 to 1440. Still very far away of its limit, but it means something is happening.

So, any idea about what can be happening? Maybe someone identifies this 40-bytes step? 

Please let me know and have a nice and healthy day.

closed 2020-04-29 by kazola

mundodisco8, 2020-04-30:
I had a similar issue (I think) on #124 (excuse me but I've only read your issue briefly)
It's been a long time since I left the company I was working on using littleFS, so I'm a bit rusty, so forget if something I write here is not accurate. The short answer is that you found something that is in the design of litttleFS so there's not much you can do about it.

If I am correct, you keep a bunch of blocks to use as index for your files. Every file you create needs an entry on the index. But not only that: opening an existing file,writing something, and closing it, will change its metadata (length and position, for example), so that goes to the index too. . On a hard drive, you can rewrite the journal. On a flash, given that you can only erase per sector, and it's quite costly in time (and each erase you are one erase closer to the flash death), you can't simply do that. The index in littleFS is a journal: instead of erasing the index of 'a.txt', you write a new one at the end. You end up with 20 references to 'a.txt', but you know that only the last one is up to date.
So if you want to access 'a.txt', you have to read the whole index in order to grab the latest reference.

Reading is not particularly costly in time, and if you have enough RAM you can read a whole sector at once and do the op in almost no time. The problem is when you run out of space in your index (too many entries, for example). Then you need to grab another block to write more entries. The problem here is that littleFS can't rely on a block being empty, so the first thing it does is erase it, and that TAKES TIME!

I think there's a setting where you can keep a map of the state of sectors in memory as a bitmap (so you need one bit per sector). That could help with access times, because littleFS knows straight away which sectors are in use and which aren't, but there's no setting to mark sectors "as empty" so littleFS can safely write to them straight away. If you had a way to erase sectors when you know you are idle, paired with that "empty sectors" map, maybe you could avoid the speed hit that way? I don't know, I feel like Trump suggesting that a fine mist of bleach paired with a good UV LED in your lungs could kill Coronavirus...

Sorry for the long message. Maybe you could change your approach and try to open and closing files when you know time is not critical. In my case, I decided that missing a couple of minutes in my logging was not ideal, but not a deal-breaker (I was logging data for days, skipping a couple of minutes wouldn't be the end of the world), but it might be to you. In that case, make sure you always have two files open at the same time, and enough RAM buffers to store the logged data while a heavy file open/close operation monopolizes the flash. Just a note, I never had time to implement this, so I can't assure it will work.