littlefs-project/littlefs #417: What is the maximum stack depth of LFS?

Johnxjj, 2020-04-27:
I see a recursive function in the file system.

e107steved, 2020-06-18:
From readme.md (the front page): 

> The filesystem contains no unbounded recursion and dynamic memory is limited to configurable buffers that can be provided statically.

Roger-Waters, 2020-06-18:
Memory consuption depends on which API function you use. In my project on AVR8 avarage stack depth is about 500 bytes. But it is really important for me to know calculated (is it possible?) amount of memory required. May be author knows something?

e107steved, 2020-06-25:
Stack usage is going to depend not only on the chosen processor, but on the compiler used and the compile options chosen; so I don't see any alternative to measuring it. 
The usual technique is to fill the stack area with a known value, exercise your program, and see how much of the stack area has been overwritten.
Possibly @geky can indicate which code paths are likely to use the most stack, when he gets a chance

zacwbond, 2020-10-16:
I would also like to know about the worst-case call chain along with a process by which it could be triggered.

We've seen filesystem corruption errors occasionally in the field, and I suspect it's due to some unusual lfs code path blowing out the stack.  After rebooting, the filesystem (usually a single file) is in a corrupt state and can't be used without either deleting the corrupt file or, in some cases, reformatting the filesystem.

blomnik, 2020-10-17:
@zacwbond, hello!
I had a similar problem in my project (avr-gcc 5.4.0 + avr-libc 2.0.0 + Atmega2560) and, you are right, it is a stack overflow problem. I noticed, that most intensive stack usage is on block relocation while writing. At least, I had no other "heavy" cases. Hope it helps you.

geky, 2020-10-19:
Ah, sorry for the late response. Unfortunately littlefs's stack usage is an area that needs quite a bit of work. There are quite a few corner cases that can lead to different stack uses, and other than best practices such as no on-stack arrays, I don't have more in-depth info.

The relocation logic is definitely the path to be the most concerned about, as it _is_ written recursively. It should be tail-recursive, but whether or not that gets optimized away is at the whim of the compiler. (It only recurses if bad blocks are detected while relocating another bad block, this is rare but I realize it is still an important concern for small stacks).

What should be done is this piece of logic should be rewritten as a proper loop.

After that it may be interesting to explore more the stack usage in-depth. It will just be a bit tricky given all the routes littlefs can take.

closed 2023-11-08 by Johnxjj