littlefs-project/littlefs #418: Can't understand how fs write data....

Spider84, 2020-04-28:
```
const struct lfs_config cfg = {
    .....
    .read_size = 16,
    .prog_size = 16,
    .block_size = 4096,
    .block_count = (8000000/4096),
    .cache_size = 256,
    .lookahead_size = 246,
    ......
};
```
As result I can't save file with size 700k at 8Mb fs. Why? What I do wrong?

e107steved, 2020-04-28:
What error returns do you get?
Have you fully debugged and checked your low level drivers that actually access the memory?

Spider84, 2020-04-28:
Thx for answer!
I make simple test application which work with file by read()/write() functions as block device.
lfs_format() seems to work.
"boot_count example" work fine.
But if I trying to write more than 1k data - lfs will write data infinity.

lfs_fs_size() - always return -84. I has created lfs image by lfs.fuse and load in this program. Dir listing is work file, but lfs_fs_size() return -84.
```bash
dd if=/dev/zero bs=512 count=15625 | tr "\000" "\377" >image
losetup /dev/loop1 image
lfs --read_size=256 --prog_size=256 --cache_size=256 --lookahead_size=256 --block_cycles=100000 --block_size=4096 --format /dev/loop1
lfs <same options> mount /dev/loop1/mnt
cp alert1.mp3 /mnt/
dd if=/dev/zero of=/mnt/boot_count  bs=4 count=1
umount /mnt
```

[littlefsmk.c.zip](https://github.com/ARMmbed/littlefs/files/4544148/littlefsmk.c.zip)
[trace.log.zip](https://github.com/ARMmbed/littlefs/files/4544151/trace.log.zip)